OpenCPN/OpenCPN #1827: Plugin Manager: Fails Download

rgleason, 2020-04-06:
Master catalog.  Ocpn_Draw.   Upgrade to 1.6.46 fails.
[image]

Using fresh newly compiled  O v5.0.522

leamas, 2020-04-06:
> Could not resolve host: dl.cloudsmith.io

This is most likely some kind of network and/or DNS problem in your local environment, unrelated to OpenCPN

rgleason, 2020-04-06:
How do I find the URL if I am just a normal user?  Is there an easy way?
I need to check that it is available on the internet.  Can we have a button access to that or something?

leamas, 2020-04-06:
The url is in the log snippet in your initial post.

leamas, 2020-04-06:
BTW: It *is* your internet: Resolving dl.cloudsmith.io/:

```
$ host dl.cloudsmith.io
dl.cloudsmith.io has address 52.85.241.84
dl.cloudsmith.io has address 52.85.241.70
dl.cloudsmith.io has address 52.85.241.114
dl.cloudsmith.io has address 52.85.241.101
dl.cloudsmith.io has IPv6 address 2600:9000:21c3:200:e:f4d2:20c0:93a1
dl.cloudsmith.io has IPv6 address 2600:9000:21c3:3800:e:f4d2:20c0:93a1
dl.cloudsmith.io has IPv6 address 2600:9000:21c3:400:e:f4d2:20c0:93a1
dl.cloudsmith.io has IPv6 address 2600:9000:21c3:a200:e:f4d2:20c0:93a1
dl.cloudsmith.io has IPv6 address 2600:9000:21c3:9400:e:f4d2:20c0:93a1
dl.cloudsmith.io has IPv6 address 2600:9000:21c3:7c00:e:f4d2:20c0:93a1
dl.cloudsmith.io has IPv6 address 2600:9000:21c3:d200:e:f4d2:20c0:93a1

```

rgleason, 2020-04-06:
Yes, but when this happens, users are going to want to check the url. Having the URL in the pop up is the solution.

From the plugins/ master branch  ocpn_draw
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.46-ubuntu-16.04-xenial-tarball/versions/1.6.46.0+129.371ea5d/ocpn_draw_pi-1.6.46.0_ubuntu-16.04-xenial.tar.gz

This is "Not Found"  

Why?

leamas, 2020-04-06:
Your log messages states  `Could not resolve host: dl.cloudsmith.io`

Which basically boils down to a broken and/or DNS setup;  As I showed dl.cloudsmith.io is a perfectly valid dns hostname.

rgleason, 2020-04-06:
Alec, my internet setup is fine and has worked reliably on other cloudsmith repos.
Maybe the program should try longer.

rgleason, 2020-04-06:
This newer link works properly, 
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.51-msvc-10.0.14393-MSVC-tarball/versions/1.6.51.1+352.4755917/ocpn_draw_pi-1.6.51.1_msvc-10.0.14393-MSVC.tar.gz

**This url is wrong and  results in file not found.**

https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.46-msvc-10.0.14393-MSVC-tarball/versions/1.6.46.0+342.371ea5d/ocpn_draw_pi-1.6.46.0_msvc-10.0.14393-MSVC.tar.gz

Why? 
**Because it does not exist!**
https://cloudsmith.io/~jon-gough/repos/ocpn_draw_pi-alpha/packages/?page=5

Maybe Jon removed it?

jongough, 2020-04-06:
For space reasons I have got cloudsmith only maintaining the last 100 uploaded files. If you are too far behind you may not find the file. This is my alpha test repository so it changes quite fast, or has recently. I will increase the number of files maintained to 200.

bdbcat, 2020-04-06:
Simple solution:  The file did not exist.
And so this issue is closed?

bdbcat, 2020-04-06:
Jon...
We could use the PR to plugins/master, so that everyone is on the same page...

rgleason, 2020-04-07:
It might also be easier if Jon had a meta-url in the Opencpn/plugins/Alpha branch (similar to my meta-url)  to his more private plugin branch, call it, say "jg-alpha"  where he can upload his own ocpn-plugins.xml and we can all access his latest plugin, without Dave having to accept the PR.

jongough, 2020-04-07:
Not sure how the alpha meta repository works, have not been following that discussion. Any hints?

jongough, 2020-04-07:
I am building OD now and will create new ocpn-plugins.xml file when it is done.

rgleason, 2020-04-07:
meta-urls  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installer_opencpn_plugins

On Alpha branch, git pull it  and then add to the ocpn-plugins.xml this

```
</plugin version="1">
  <name> Jon Gough Alpha Plugins meta-url </name>
  <version> 1.0 </version>
  <summary> meta-url to Jon's Alpha Plugins </summary>
    <meta-url> https://raw.githubusercontent.com/jongough/plugins/jg-alpha/ocpn-plugins.xml </meta-url>
</plugin>
```

Then make a new jongough/plugins branch called jg-alpha
and put your plugin xml's in the metadata for that branch and build the ocpn-plugins.xml.
Then push it up to your remote.

Now others can try those plugins out.

jongough, 2020-04-07:
I am trying this now

Hakansv, 2020-04-07:
On my Win10, for info..
Built now. Last commit: "Correct plugin panel layout ........."
A try to upgrade OCPN Draw: Installation error as of below shot.
I checked "Beta" and "Alfa" catalogs. Then again updated "master". The same error. 
A O restart makes no difference.

[image]

```
08:57:44.915 MESSAGE pluginmanager.cpp:642 Installing managed plugin: OCPN Draw
08:57:45.204 MESSAGE Downloader.cpp:132 filesize https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.46-msvc-10.0.14393-MSVC-tarball/versions/1.6.46.0+342.371ea5d/ocpn_draw_pi-1.6.46.0_msvc-10.0.14393-MSVC.tar.gz: 0 bytes

08:57:46.084 WARNING Downloader.cpp:84 Failed to get 'https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.46-msvc-10.0.14393-MSVC-tarball/versions/1.6.46.0+342.371ea5d/ocpn_draw_pi-1.6.46.0_msvc-10.0.14393-MSVC.tar.gz' [The requested URL returned error: 404 Not Found]

08:57:50.382 MESSAGE PluginHandler.cpp:691 Cleaning up failed install of OCPN Draw
08:58:05.516 MESSAGE pluginmanager.cpp:995 PlugInManager: loading plugins from C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins
08:58:05.516 MESSAGE pluginmanager.cpp:958 Using PATH: C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Users\admin\AppData\Local\opencpn\plugins;C:\Program Files (x86)\OpenCPN\plugins;C:\Python38\Scripts\;C:\Python38\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\System32;C:\Windows;C:\Windows\System32\wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\OpenCL SDK\2.0\bin\x86;C:\Program Files (x86)\Intel\OpenCL SDK\2.0\bin\x64;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\Poedit\GettextTools\bin;C:\wxWidgets-3.1.2;C:\Builds\OCPN\OpenCPN\buildwin;C:\GE2KAP\ooRexx;C:\Builds\OCPN\OpenCPN\buildwin\expat-2.2.5;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\Process Lasso;C:\Program Files (x86)\GDAL;C:\Windows\System32\OpenSSH\;C:\wxWidgets-3.1.2\lib\vc_dll;C:\Program Files\CMake\bin;C:\Program Files (x86)\ooRexx;C:\Program Files (x86)\Schneider Electric\SoMachine Software\Tools\ConfigurationManager\;C:\Program Files (x86)\GnuPG\bin;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;C:\Builds\Ruby22\bin;C:\Users\admin\AppData\Local\Programs\Python\Python35\Scripts\;C:\Users\admin\AppData\Local\Programs\Python\Python35\;C:\Builds\OCPN;C:\Users\admin\AppData\Local\Microsoft\WindowsApps;C:\MinGW\bin;C:\MinGW\msys\1.0\bin;C:\Builds\OCPN\OpenCPN\buildwin;C:\Builds\OCPN\OpenCPN\buildwin\expat-2.2.5;C:\Builds\OCPN\OpenCPN\buildwin\gtk;C:\Builds\OCPN\OpenCPN\buildwin\crashrpt;C:\Builds\OCPN\OpenCPN\buildwin\wxWidgets;C:\Users\admin\Documents\TV_film\uwtd\ffmpeg\bin;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\
08:58:05.516 MESSAGE pluginmanager.cpp:1001 Loading plugins from dir: C:\Users\admin\AppData\Local\opencpn\plugins

```

leamas, 2020-04-07:
The log says it all: 

`https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.46-msvc-10.0.14393-MSVC-tarball/versions/1.6.46.0+342.371ea5d/ocpn_draw_pi-1.6.46.0_msvc-10.0.14393-MSVC.tar.gz' [The requested URL returned error: 404 Not Found]`

It is indeed non-existing, probably for reasons explained  in https://github.com/OpenCPN/OpenCPN/issues/1827#issuecomment-610069684.

Hakansv, 2020-04-07:
Alec..
Ye,s I agree the reason is clear. A rather relevant user question at this situation may be: "Why is there an option to upgrade when it's not available?"
Thanks/ Håkan

jongough, 2020-04-07:
If you want to try out the latest version copy this version to the location where opencpn.ini/conf is and you should be able to download the latest version of OD 1.6.52. I have submitted a patch which should be there in the next 24 hours.
[ocpn-plugins.xml.txt](https://github.com/OpenCPN/OpenCPN/files/4442679/ocpn-plugins.xml.txt)
(rename to remove the '.txt.). If you want the windows version you select 'Custom' and place the following in the URL location:
```
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-msvc-10.0.14393-MSVC-metadata/versions/1.6.52.0+353.feb1882/ocpn_draw_pi-1.6.52.0_msvc-10.0.14393-MSVC.xml
```

jongough, 2020-04-07:
ODraw has been in alpha testing whilst doing this and has yet to move to beta, but currently the master version is being used for testing (not sure why). I am trying now to setup an 'alpha' version which will let you use the 'Alpha' repository setting to more easily find a version.

leamas, 2020-04-07:
> Why is there an option to upgrade when it's not available?"

Which is a bug that should be fixed, agreed.  Still, it's part ot the game that alpha is unstable, it will not be part of upcoming beta or final release.

rgleason, 2020-04-07:
Lots of open questions. This should remain open.

Right now I am wondering why there are not separate xml files stored in programdata/opencpn, for example  master.xml, beta.xml, alpha.xml and custom.xml.  Then the metadata for downloaded and installed plugins would be available.

rgleason, 2020-04-07:
Btw Alec you are right, that my inet was not working at that time. I was testing the way things worked when I had no internet available too. So we are probably looking at a wrong segment of the log file.

leamas, 2020-04-07:
> Right now I am wondering why there are not separate xml files stored in programdata/opencpn, for example master.xml, beta.xml, alpha.xml and custom.xm

etc... This was the kind of flexibility we dropped when we went for something which should be as simple as possible for end users. Also known as _There Ain't no thing such as a Free Lunch_..

jongough, 2020-04-07:
Also remember *Don't reinvent the wheel*.

rgleason, 2020-04-07:
Jon you wry devil.  Hakan, can you open a more targeted Issue with your question, which I think it a good point. 

"...user question at this situation may be: "Why is there an option to upgrade when it's not available?"

and I will close this one?

bdbcat, 2020-04-08:
Rick...
Don't really need a new issue.
The master catalog was broken, due the the deletion of a target tarball.  Simple enough.
These thing happen in rapid development cycles.
We will not release a master catalog that has broken links.

closed 2020-04-08 by rgleason