OpenCPN/OpenCPN #1829: Plugin Manager - tarball cache clean-up

jongough, 2020-04-06:
Using OCPN master (b45153f1f) on Ubuntun 18.04  when installing plugins the tarballs are downloaded to ~/.opencpn/plugins/cache/tarballs and these are used for reinstalling plugins without the need to have an internet connection. How are old tarballs cleaned up and when? There does not appear to be any way the user would know to do this and the Plugin Manager does not appear to do this automatically.

rgleason, 2020-04-07:
Windows cache is here: C:\ProgramData\opencpn\plugins\cache\tarballs
It has nine tarballs, there are no metadata files.
There is no metadata xml file for the plugin stored inside the tarball.
Are there backup master, Beta and alpha ocpn-plugin.xml files stored somewhere?

I notice that download of a Custom URL also adds a tarball too the cache too, fortunately.

rgleason, 2020-04-07:
This cache could be very useful as backup storage of current versions of plugins that are used frequently (at sea). 

I hope that the user is given control over the cleanup. 

Users may want to keep previous versions around for awhile, other users might be space constrained and want to set a use limit with automatic removal of older versions.

Downloading and having the metadata available for each tarball would also be useful in building a simple direct installer that is accessible from within the cache initially.

rgleason, 2020-04-07:
I think the future Cache Installer paths for Windows and Linux would be built in to expand the tarball to these paths:

From Alec's [PI Manager Wiki](https://github.com/leamas/OpenCPN/wiki[)
Installation Paths](https://github.com/leamas/OpenCPN/wiki/Installation-paths)

Windows 
%localappdata%\opencpn\plugins\<plugin.dll>
%localappdata%\share\locale\opencpn\share\locale\<each language>\LC_MESSAGES\<plugin.mo>

Linux
~/.local/lib/opencpn/LIBRARIES/lib<plugin>.so
~/.local/share/opencpn/plugins/<plugin>/DATA FILES
~/.local/bin/BINARY HELPERS

rgleason, 2020-05-09:
Jon asks at the start:
> How are old tarballs cleaned up and when? There does not appear to be any way the user would know to do this and the Plugin Manager does not appear to do this automatically.

This is a consideration when a plugin tarball is installed, and then when it is "enabled" and it fails or opencpn crashes.  It may make good sense to remove the plugin tarball from the cache at this point?  Manually or Automatically?

bdbcat, 2020-05-09:
Rick...
Based on your recent tests, it is a good idea.
But difficult to accomplish.  In general, one cannot know which plugin has crashed in the last OCPN run.  So the alternative is to completely clear the cache.  Or clear the cache of enabled plugins, since presumably the crashing plugin was enabled. But this is maybe not what is wanted, since potentially good plugin cache records are nuked.
Needs some thought....

jongough, 2020-05-09:
Perhaps if OCPN crashes the window that is displayed could offer the ability to 'uninstall' and/or delete the cache items for the selected plugin. So you could disable all plugins, get OCPN working, enable one plugin at a time to find one that goes 'bang' and then uninstall it and remove the cache item.
Or the Plugin Manager could off the option for each plugin to delete its cached copies of the tarballs.

rgleason, 2020-05-09:
The choice to "disable" or "remove cache" of the plugins, upon crash or subsequent startup as part of the "safe" choice  would be great.

If Plugin Manager can offer removal of cached copies of tarballs in one of the secondary menus that would be good.

PR #2263 (2021-05-06): Add  cache feedback and a Clear Cache  button (#1829)
Add some info on cache size and a Clear Cache button, the bottom panel in screenshot:

[image]

Closes: #1829
closed 2021-05-06 by bdbcat