OpenCPN/OpenCPN #1830: Plugin Manger - Ubuntu 18.04 click on plugin causes install dialog to show

jongough, 2020-04-07:
OCPN master (eda25b76) OD 1.6.52 & oeRNC Charts 1.2.0 install plugin, then reinstall plugin, then uninstall plugin, now click on plugin and the install dialog comes up automatically
[image]
[image]

bdbcat, 2020-04-07:
Would that not be the expected action for an un-installed, managed plugin? (i.e. the left icon is a "package" box)

jongough, 2020-04-07:
I would have thought the default action would have been to expand the plugin area to show the install button. I would have expected it to look more like this:
[image]
with the Preferences and Reinstall button greyed out and the Uninstall button to say Install (or some combination. This then gives the user a bit more information and the option of going to the 'Website' to further investigate before installing. Here is what I would expect to find on clicking the plugin:
[image]
I just haven't gone to as much effort yet to put that information into this window (I presume it would come from the `<description>` tag in the xml).

jongough, 2021-04-26:
Latest version of OCPN 5.5.0 displays the install dialog when you left click on any uninstalled plugin. The expansion to show what the plugin does does not occur. This now does not require the plugin to have been installed and uninstalled first. With the current way it works there is no way for the user to better understand the plugin from the Plugin Manager.

bdbcat, 2021-04-27:
Jon...
The "more" button in the install dialog expands to show the available information about the plugin, as extracted from the metadata file.
This should be sufficient.

jongough, 2021-04-27:
The supplied plugins behave differently to the non-supplied plugins. The supplied plugin on clicking it gives information about the plugin: https://github.com/OpenCPN/OpenCPN/issues/1830#issuecomment-610183447, whereas a non-supplied plugin that is got from the catalog or by the import button opens the install dialog on which the "More" button is display. The two should react the same.

bdbcat, 2021-04-29:
Not necessarily.  They are different beasts.
Closing, not a bug.

closed 2021-04-29 by bdbcat