OpenCPN/OpenCPN #1832: Plugin Manager - Ubuntu 18.04 reporting uninstalled plugin as downgradeable

jongough, 2020-04-07:
Using OCPN master (6171e1f2) open options/Plugins and select uninstalled  plugin testplugin_pi. It is in the ocpn-pluginx.xml file twice so it shows two versions that can be installed. Both have a button with 'Downgrade' on it rather than install.
[image]
Here is the associated log, but it does not show anything about testplugin_pi.
[log of ocpn showing uninstalled plugin as a downgrade.txt](https://github.com/OpenCPN/OpenCPN/files/4447199/log.of.ocpn.showing.uninstalled.plugin.as.a.downgrade.txt)

bdbcat, 2020-04-07:
Is this still happening with updated master containing only testplugin 1.0.58?
Looks OK to me.
FYI, we will never intentionally release a master catalog with more than one installation choice.  Very confusing to users, will prompt immediate support event.
I'm sorry I did not catch this before.

jongough, 2020-04-08:
Yep. Currently on OCPN master (a71ed122), have created ocpn-plugins.xml with only testplugin_pi
[ocpn-plugins.xml.txt](https://github.com/OpenCPN/OpenCPN/files/4447743/ocpn-plugins.xml.txt)
and get the following result:
[image]

bdbcat, 2020-05-06:
Ready to close after https://github.com/OpenCPN/OpenCPN/commit/294b28b31770ff39d746dea8ecd8be15c4edf3cb

closed 2021-04-26 by jongough