OpenCPN/OpenCPN #1834: Weather-Routing install

rgleason, 2020-04-08:
Currently the weather-routing installation is not initializing properly by copying data directories to ProgramData/opencpn/plugins  (for Windows)  including the Boats and Polars directories and files.  

This needs to be fixed.  Until then please copy the files over manually. 
I've tested weather-routing after doing that and it is working fine with the files located in the proper location.

[image]

rgleason, 2020-05-01:
WeatherRouting.cpp

```
Line165     /* if the boats or polars directories did not previously exist, populate them */
    if (forceCopyBoats)
        CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/boats"), boatsdir);
    if (forceCopyPolars)
        CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/polars"), polarsdir);
```

Perhaps the plugin has to be completely removed, for the bool forcecopyboats and forcecopypolars to be set to true.

rgleason, 2020-05-01:
Found this from Jon Gough for copying data files for macos
[Try copying directory structure](https://github.com/rgleason/weather_routing_pi/commit/e03dc26dddb47fe70eb1fa91902d6a5f84957c76)   -might work for other env?

```
 foreach(_currentDataFile ${PACKAGE_DATA_FILES})
    message(STATUS "copying: ${_currentDataFile}")
    configure_file(${_currentDataFile} ${CMAKE_CURRENT_BINARY_DIR}/data COPYONLY)
    #configure_file(${_currentDataFile} ${CMAKE_CURRENT_BINARY_DIR}/data COPYONLY)
    file(COPY ${_currentDataFile} DESTINATION ${CMAKE_CURRENT_BINARY_DIR}/data)
  endforeach(_currentDataFile)

```

rgleason, 2020-05-01:
Cleaned up all plugin directories, 
%localappdata%/opencpn/plugins 

After installation of very recent weather_routing_pi 1.13.6 from "import plugin"
the all the files in 
C:\Users\Rick\AppData\Local\opencpn\plugins\weather_routing_pi\data\boats
C:\Users\Rick\AppData\Local\opencpn\plugins\weather_routing_pi\data\polars
must be copied to
C:\ProgramData\opencpn\plugins\weather_routing\boats
C:\ProgramData\opencpn\plugins\weather_routing\polars 
respectively, because they are found to be empty.
It appears the the weatherroutingconfiguration.xml file is no longer present, so those configurations are gone.  The file will be recreated when configurations are saved.

rgleason, 2020-05-01:
Why isn't line 165 working  (above)?
/* if the boats or polars directories did not previously exist, populate them */

rgleason, 2020-05-01:
Sean has programmed the initialization of weather_routing quite carefully, however the change of plugin location has screwed up the initialization programming.  I am thinking  that "StandardPath() is not working, so none of the initialization works.

**WeatherRouting.cpp**   starting at  Line 147
```
    m_default_configuration_path = weather_routing_pi::StandardPath()
        + _T("WeatherRoutingConfiguration.xml");

    bool forceCopyBoats = false;
    bool forceCopyPolars = false;
```

bdbcat, 2020-05-02:
Rick...
Where are the data files supposed to be copied to, in your understanding?

bdbcat, 2020-05-02:
Rick...
What branch are you using to build managed version of weather routing from your github fork?

rgleason, 2020-05-02:
Dave, we need the data files to be read/write so they need to be copied to the user directory.
In the case of Windows that is 
 C:\ProgramData\opencpn\plugins\weather_routing

We need all the files copied to these two directories so that the plugin has access to read and write
boat.xml files and polar.csv or. txt files

C:\ProgramData\opencpn\plugins\weather_routing\boats
C:\ProgramData\opencpn\plugins\weather_routing\polars
Right now they are empty!
For other env it would be to the comparable user directories for read-write
I believe there are variables for this.

Sean was going to fix it, but I have not heard from him for 2 days. Sent him a note about using Cloudsmith last night.

rgleason, 2020-05-02:
Oh, the branch to use now is 
https://github.com/rgleason/weather_routing_pi/commits/master
I just updated it and all env are working, but Jon has one set of additional changes for MacOS. but we will be working on another branch for that.  Thanks so much!!

rgleason, 2020-05-02:
Dave, to be clear, they are normally first copied to the program directory or in the case of PI Manager to the normal tarball destinations.  Then the plugin is initialized at some point and if the user data directories don't exist, the boats and polars files should be copied there.

bdbcat, 2020-05-02:
OK, I'll look at it tonight.
Is the Mac CI build fixed?

rgleason, 2020-05-02:
It builds, not tested, but Jon has other changes I need to add. I will try to do that today, using frontend2 first then PR & merge it to "master" once working, all before 5 pm.

bdbcat, 2020-05-02:
OK, thanks.

rgleason, 2020-05-02:
It done and building now
https://github.com/rgleason/weather_routing_pi/commits/master

right now they are landing in cloudsmith alpha
https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-alpha/packages/

bdbcat, 2020-05-03:
Rick...
I have a solution to the install problem:
```
Everywhere you use "GetPluginDataDir()", change from:

for example:
GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/boats"), boatsdir);

to:
GetPluginDataDir("weather_routing_pi") + _T("/data/boats"), boatsdir);
```

Looks like there are about 5 or 6 places in WeatherRouting.cpp

If you make me a collaborator on you github project, I'll fix it directly.
Dave

rgleason, 2020-05-03:
Dave, 
You, Jon and Sean are collaborators for weather_routing. You are all set to go. Thanks!

bdbcat, 2020-05-03:
OK, should be fixed now.  Please let me know how this goes, and I'll look for a PR to the master catalog.
Dave

rgleason, 2020-05-03:
Will do. Thnk you again!

rgleason, 2020-05-05:
Thanks Dave, done

closed 2020-05-05 by rgleason