OpenCPN/OpenCPN #1835: Plugin Manager - Ubuntu 18.04 only Master and Beta remote catalogs available

jongough, 2020-04-08:
OCPN master (a71ed1225) open options/Plugins, select 'Choose Remote Catalog' button and only Master and Beta are being show, should also show Alpha and Custom:
[image]

jongough, 2020-04-08:
Even with the new hidden 'CatalogExpert=1' set Alpha and Custom don't show.

I can 'sort of' understand removing Alpha, but Custom seems a bit strange. Perhaps there should be a tick box in the Options dialog to allow a user to select 'Expert' or not rather then putting another hidden variable in the Config file.

rgleason, 2020-04-08:
Did you hit update catalog Jon?

jongough, 2020-04-08:
Huh? That still doesn't show the Alpha and Custom catalog selections I should be able to make. Do you see the ability to select these on this version (you need to have at least the version of OCPN shown at the top of this issue as it was the last commit)

bdbcat, 2020-04-08:
Jon...
If you edit manually the config file, it must be while OCPN is stopped.  Otherwise your changes are overwritten.

bdbcat, 2020-04-08:
We require a manual edit because, well, Experts should be able to do this kind of thing.
Allowing the loading of arbitrary, uncurated XMLs can be dangerous.  Malware, anyone?

jongough, 2020-04-08:
I have done the edits and OCPN does not take any notice. The line added at 5045 does not return the value in the config file, rather it defaults to 0. Don't know why, but perhaps a global variable may work better.

bdbcat, 2020-04-08:
What system are you running?
If this does not work, nothing else will work.  That is the normal way to access the config file....

bdbcat, 2020-04-08:
Jon...
Please try yhis substitution:

     pConfig->SetPath( _T("/PlugIns") );

Drop the ending '/'

jongough, 2020-04-08:
My main testing is now on Ubuntu 18.04, I can try it on 16.04 if you want. I have also tried it under debugging on my normal desktop, Linux Mint 19.3 (Ubuntu 18.04 based).

bdbcat, 2020-04-08:
I tried it on 18.04 .  No problem.

bdbcat, 2020-04-08:
For ref, my config:

```
[PlugIns]
PluginOrder=WeatherRouting;testplugin;oeRNC Charts;Radar;Squiddio;Dashboard;S63;GRIB;WMM;ChartDownloader;VDR;oeSENC;OCPN Draw
bEnabled=0
LatestCatalogDownloaded=Master
CatalogExpert=1

```

jongough, 2020-04-08:
Ahhh, I put the line in with CatalogCustomURL and CatalogChannel which are both in the [Settings] section. It would seem that we are spreading the information around a bit.

jongough, 2020-04-08:
I have tried it in the Plugins section and it works.

closed 2020-04-09 by jongough