OpenCPN/OpenCPN #1836: RFE: Simplify testing of new plugins

leamas, 2020-04-09:
The process to test a new plugin is somewhat complicated, involving the generation of a new, complete catalog. This process is described in https://github.com/opencpn/plugins (the overall README.md).

An alternative is the ocpn-install.sh script which is able to install a plugin directly. However, these kind of tools has man rough corners for new developers.

We need to simplify this process to lower the threshold for new developers.

The basic usecase is a developer which has developed a new plugin, or just patched an existing one, and needs to test it.

rgleason, 2020-04-09:
Bcn

> how to deal with the following scenario:
> - system installed on boat with no Internet access
> - user needs to install an additional managed plug-in offline
> 

> Just to add one of our "daily" proceedings for illustration:
> "can you please test this patched version of the plug-in?"
> 
> Download on the Mac, copy the PI onto a USB stick, search the W10
> tablet, install, next the Dell XPS13 (hiRes, W8.1) and then the boat PC
> with another W8.1 Dell XPS18. Using the same stick for screenshots,
> logs, etc.
> 

> ..the overhead to get into a catalog is pretty heavy in my opinion, and will result is less plugin development"
> A script file during beta testing would be very useful, but that is not the long term answer, in my opinion. I think it needs to have an interface, even if it is another plugin that becomes Standard. "
> The user doesn't know there is a cache or where it is  and does not understand what an xml file is or it's purpose.  So I think you guys are talking at development level here.
> 

As a part of that plugin, you could also backup and restore the cache and ocpn-plugins.xml

rgleason, 2020-04-09:
Use Cases
1 . Restoration at Sea, no internet. Computer has had a hard failure and  installation needs to be restored
2.  User need to install/update an additional managed plugin off-line (fetching it on an internet café or something like that).
3. Testing case: User want to test various plugins and versions while disconnected.

This item only deals with #3

leamas, 2020-04-09:
> This item only deals with 3

Yes, on purpose. We cannot handle all use-cases  everywhere. Specifically, 1. is to be discussed in the mailing thread.  2 is probably the same as 3 from an implementation view, to be discussed here.

bdbcat, 2020-04-09:
This is, in general, not an easy problem to automate. Lets walk through the steps to do this fully manually, just to exercise our collective understanding.

While in the internet cafe:
1.  Look online at github repo "plugins".  Download and inspect the current catalog.  Parse (manually) that catalog to find the tarball URL for the platform of interest (Win, Mac, etc.).  
2.  Download that tarball from the discovered URL.
3.  Save catalog and tarball on a thumbdrive.

Travel to the target system.
On the target system, OCPN not running:
4.  Replace {config dir}/ocpn-plugins.xml with the catalog on the thumbdrive.
5.  Copy to tarball from the thumbdrive to the {config dir}/plugins/cache/tarballs directory
6.  Start OCPN.  New plugin should now be available as an upgrade or new install, as appropriate.

All following so far?

bdbcat, 2020-04-09:
If we are OK so far....

I can imagine some PHP code on the opencpn.org site that would accomplish the cafe part.  A simple dialog where the user specifies the plugin name, and the desired target machine.  The PHP code parses the catalog, creates a zip file containing the catalog and tarball, and downloads automatically for the user.

leamas, 2020-04-09:
hm... the usecase here is a developer which has created a new plugin (new version of existing plugin). 

Does the assumptions here hold? After all, the new plugin is not part of the catalog...

bdbcat, 2020-04-09:
Alec...
Sorry, was thinking of Hubert's specific case, testing an update to an existing plugin.

bdbcat, 2020-04-09:
Right....
Slightly different problem.  More complicated, since the catalog will need to be merged with the plugin metadata section associated with the newly patched plugin and target platform.
Could be done by the imagined PHP module, effectively creating a Beta catalog containing one entry...

rgleason, 2020-04-09:
I like the description of the process Dave (thumbs up).  I also like the idea of using the website download page to help, in whatever form, as this can be very direct.

This is not a completely thought out idea, but the germ of one. Assume the metadata and tarball get to the cache somehow (Dave's idea).  Then the user goes to Custom URL (or it's counterpart for installing from a hard drive) and browses to pick the meta.url for the plugin (it is assumed the tarball will be in the same directory, or they will both be compressed together). This gets installed and and the metadata gets added to the Beta catalog, so that it looks like a managed plugin.
If the Beta catalog is ever updated, the individual plugins will disappear, but they will be available to be installed from the cache.
If you are not connected to the internet, the Beta Catalog will not update, so that is good, the individually added plugins to Beta will stay there until they uninstalled.

This arrangement would be more flexible.

leamas, 2020-04-09:
> Could be done by the imagined PHP module, effectively creating a Beta catalog containing one entry...

But then again, for our new developer using server-side code might be complicated.

Just thinking aloud: what if

  - We bundle the metadata in the tarball (with a fixed filename like metadata.xml)
  - Add an "import plugin" dialog to ocpn which:
      - Let user select the plugin tarball (standard filepicker).
      - Extracts metadata from tarball.
      - Merges it with current catalog
      - installs the tarball same way as the downloaded stuff
      - If everything is OK, installs the updated catalog.

Note: these are big changes, not going into the beta.

BTW: In hindsight I think not bundling the metadata in the tarball was a major mistake.

rgleason, 2020-04-09:
Is it too late to add a line to copy the xml into the tarball now?  Would the xml being in the tarball create any trouble when expanding it to various locations?

There are not that many plugins at this point.

leamas, 2020-04-09:
> Is it too late to add a line to copy the xml into the tarball now? Would the xml being in the tarball create any trouble when expanding it to various locations?

I think so. More important would be to add code which accepts a metadata.xml file in the tarball for now,  just ignoring it. That would make the transition easier. But the rest should IMHO wait to after this release.

rgleason, 2020-04-09:
Thanks for thinking about it.

bdbcat, 2020-04-10:
OK.  Alec's idea has merit.  We shall proceed in that way.
I have done the following:
1.  Modified Frontend#1 (oesenc_pi) to include the metadata.xml file in the tarball.
2.  Modified OCPN core to ignore any file called "metadata.xml" in the tarball encountered during normal installation.

@ Rick, @Jon, @Kees:
Please adjust your plugin CI build scripts to emulate the behavior of oesenc_pi.  Source changes are very minimal.  See github.

@Alec:
metadata not yet included in oesenc_pi magic Flatpak build scripts.  Please take a look.  Thanks.
Also might like to adjust the tarball spec to reflect the inclusion of the metadata.

I am a bit more sanquine than Alec about the possibility of having a "tarball install dialog" for the Beta.  Give me 24 hrs or so to take a detailed look.

Thanks to Alec...

bdbcat, 2020-04-10:
If the outlined "bare tarball installer" works out, it still leaves the potential problem of requiring users to wander through various cloudsmith repos to find a useful tarball, without specific direction.
I think the PHP code on the opencpn.org site could be sufficiently generic that the developer need not be involved, at least for the case where the desired plugin has been released to "some" catalog, either Beta or master.
So now we need a skilled and interested PHP coder....

leamas, 2020-04-10:
> If the outlined "bare tarball installer" works out, it still leaves the potential problem of requiring users to wander through various cloudsmith repos to find a useful tarball, without specific direction.

Yes. But this is not a problem for the specific use case here, the developer knows where to find the tarball to test -- after all, she has just built it.

The idea is that most users should use the plugins available in the catalog. The developer use case  is a corner-case, but an important one.  Let's assume that devs after initial testing takes the next step and makes PRs against the plugin project i. e., gets the plugin included in the master catalog.

I think discussions has revealed that testing plugins not in the catalog is considered  too complicated with the current scheme

leamas, 2020-04-10:
Implementations notes out of the top of my head:

  - We should mark imported plugins as such, in the plugin data and in the GUI.
  - This will mean that the the effective catalog is created by merging  the downloaded and the imported metadata. This will affect catalog updates besides importing plugins.

lskillen, 2020-04-10:
Lee from @cloudsmith-io here (so take my suggestions with a pinch of salt):

It would seem like a great idea to use the centralised OpenCPN organisation on Cloudsmith for the "catalog" repository. If you invite developers into the organisation, and create staging repositories for them, you could then designate one (or multiple) people who have the permission to "release" those assets to the production catalog repository (by copying/moving from the staging repositories).

In that way, management would be centralised, and you would have all of the fantastic benefits that Cloudsmith offers, with worldwide distribution. Plus it would be one point-of-contact, one place to collaborate, and one way of retrieving the assets. If you lean on using the repositories correctly, by using them as Debian/RedHat/etc. repos with native tooling, you would get a lot for free. E.g. Running `apt install <package>` to get the latest version, without having to find out what the latest version is.

The actual choice of implementation is of course, up to you as a group; but I think the hub and spoke model, where each developer has their own account and different ways of doing things, is quite uncommon to how most people use Cloudsmith.

YMMV! Good luck.

rgleason, 2020-04-10:
@bdbcat - have done[squiddio_pi frontend1](https://github.com/rgleason/squiddio_pi/tree/frontend1) but need to pull mauro's changes.  Will do Vdr_pi frontend1 next.

bdbcat, 2020-04-10:
Alec...
I am thinking of Hubert's "internet cafe sneaker-net" scenario.  The plugin is in production, and a user wants to install from scratch for the first time from a memory stick.  An average user would have a hard time finding the correct tarball on cloudsmith, since each developer has their own repo.

rgleason, 2020-04-10:
See Lee Skillen's gentle suggestion above.

rgleason, 2020-04-10:
https://github.com/rgleason/vdr_pi/runs/576905107
Flatpak and Mingw failures, trying to rerun Flatpak now.     

Mingw notice
```
CPack: Create package using NSIS
CPack: Install projects
CPack: - Run preinstall target for: vdr_pi
CPack: - Install project: vdr_pi
CMake Error at /root/project/build/cmake_install.cmake:188 (message):
  ABSOLUTE path INSTALL DESTINATION forbidden (by caller): //metadata.xml
CPack Error: Error when generating package: vdr_pi
make: *** [Makefile:133: package] Error 1

```

rgleason, 2020-04-10:
Looking to see if Vdr_pi creates a metadata.xml in one of the tarballs.
https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/?page=2
Yes I found one, but it is malformed becsause of an ampersand. I have to remove that from the xml file!

rgleason, 2020-04-10:
Squiddio also fails on flatpak and mingw
https://github.com/rgleason/squiddio_pi/commits/frontend1

leamas, 2020-04-10:
>  am thinking of Hubert's "internet cafe sneaker-net" scenario. 

OK, I see what you mean. And it makes a lot of sense in that perspective.

Is PHP the only option? Modern backends often uses Python or Java. 

Speaking about PHP: is  any framework such as Laravel used?

bdbcat, 2020-04-11:
Alec...
Any language will do, probably.  Our server is know to reliably support javascript and PHP.  Others maybe.  But I'm not a web-guy.  So we will need some help...

bdbcat, 2020-04-11:
Status:
"Import Plugin" functionality is now 90% complete.  Will end up at about 400 LOC.  Should have some test article in github tomorrow.

Dave

leamas, 2020-04-11:
Filed #1839  which is Hubert's "Internet café" use case so we can focus on the developer use case here. As it turns out, they are really different.

rgleason, 2020-04-11:
I wrote Mauro, with Dave's request for some web expert help with PHP code for the Downloads page.

Enable the Plugins Download page to allow direct download from cloudsmith using the  plugin master, Beta and Alpha catalogs as a basis for creating weblinks in the opencpn downloads page.  Each of the catalogs have plugin records with the tarball urls. For each plugin, there will often be 8-10 different build environments each with a tarball-url to cloudsmith.  What is needed is the basic code to read the catalog, find the URL and make the active link.

bdbcat, 2020-04-12:
All...
"Import Plugin" is now available in github, building in normal repos.
Notes:
1.  You may test using oesenc_pi plugins (tarballs) available at 
https://cloudsmith.io/~david-register/repos/ocpn-plugins-unstable/packages/

2.  When you successfully import a managed plugin, the local catalog is updated to reflect the imported plugin's existence, version, etc.  If you subsequently download the Production master catalog, the just-imported plugin will be purged from the catalog, and replaced by any previous version of that plugin that may be in the official master catalog.  However, the imported plugin will remain installed and available.  There are two scenarios to think about:
a.  Testing an update to an in-production plugin:  In this case, if you download a new master catalog, the imported plugin is still installed and available.  The plugin manager will offer a "downgrade" to the previous version, if you like.
b.  Sneaker-net from internet cafe:  Not a problem, since presumably the on-board computer will not be connected to internet, so unable to download a fresh catalog.

3.  Other plugins with embedded metadata will become available as the CI workflows are updated.  I'll let the specific plugin authors communicate the rep URLs.

Good Luck
Dave

leamas, 2020-04-12:
Not tested, but: Great work!

tai-fun, 2020-04-12:
Dave
I "tested" a little bit outside the box.
On an Ubuntu 19.10 box, with an oesenc deb package already installed, I
downloaded and "installed" oesenc_pi-4.2.4-0_ubuntu-18.04.tar.gz
And saw this dialog .....
Nothing showed up in the Pluginmanager, apart from the existing oesenc
entry created by the deb package.
Now, imagine that I'm a "complaining" user on the forum saying that I
installed the plugin "perfectly", but it does not work.
It seems a few more checks or warnings are called for.

Also, please don't forget to update the oesenc plugin to use the new fonts
for soundings.

Thanks
Thomas

rgleason, 2020-04-12:
Thomas, you could also try one of the tarballs from today, April 12 which should have metadata added to the tarball.  These are essentially done by Jon Gough.

https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/

bdbcat, 2020-04-12:
Thomas...
What dialog/warning was shown, please?

tai-fun, 2020-04-13:
Dave

There was an attached screenshot.
The message was:
"Plugin imported successfully
Active catalog updated.
Plugin may be installed or updated now."
Thomas

bdbcat, 2020-04-13:
Thomas...
Try your test again with current github build.
I added a simple test for compatibility, and related error message.

Thanks
Dave

tai-fun, 2020-04-13:
Dave
Much Better!
Probably all that is needed.
Thanks
Thomas

mauroc, 2020-04-13:
Since we agreed that squidd.io will pitch in with the implementation , I spent some time reading this conversation and looking at the https://github.com/OpenCPN/plugins repo. I am mostly focusing on the Internet Cafe use case, but I think the implementation could be made to cover other use cases with some tweaks. 

I want to make sure I am on the same page as you guys, so here's my current thinking...

1- at an Internet cafe, user navigates to squidd.io/ocpn_plugins (or similar) from the opencpn.org download pages (or wherever else)

2- a squidd.io page renders, with perhaps the Opencpn logo added to the top bar for navigating back to opencpn.org. Possibly using some of the ocpn css for buttons, labels etc. to make it more consistent with opencpn.org (luckily the two sites actually have a similar look and feel (blue top bar, white background, blue button )

3- squidd.io retrieves the raw ocpn-plugins.xml catalogs from the relevant https://github.com/OpenCPN/plugins  branches and parses them and builds some kind of structured in-memory object containing all the relevant information. (if the real time retrieval of the .xml files with each user request makes it too slow, we could have a cron job running every 15 mins or so that caches the .xml files, but I suspect it won't be necessary, as I don't expect volumes to be that high)

4- there may be a number of horizontal tabs at the top:  Master (or Production), Beta, Alpha, Experimental 

5- below, is a set of cascading context-sensitive pulldowns (meaning the values in each pull down will depend on the values previously selected), possibly in this sequence:
```
| - Plugin name 
    |-> Target Platform      
          |-> Target Version
                 |-> Plugin Version
                     |-> Release 

```
Let me know id I am over-engineering this. Perhaps some defaults could be used to reduce the number of steps, such as latest version, latest release etc

6- as the pull downs are selected, some of the relevant information (such as description, author etc. ) may be displayed in a separate box, for verification purposes

7- when all the values are selected, a (very prominent) link is displayed at the bottom and user can start the download 

It seems from the later part of the discussion, that the actual tar file and the installation .xml will be bundled , so the link could be made directly to the cloudsmith object , with no need for squiddio to intermediate and create a zip file. Am I understanding correctly?

Let me know if I am heading in the right direction.

rgleason, 2020-04-13:
Mauro, this seems to me to be right on track, but then Dave probably has some preferences and ideas.

leamas, 2020-04-14:
Indeed. Also, there is  #1839 for the "Internet Cafe" use case, trying to keep this bug focused on the developer one.

bdbcat, 2020-04-14:
Mauro...
I am in general agreement with the plan.  One point is that I would like to eventually migrate this functionality to the opencpn.org server.  But that can be done after the Proof-of-Concept is verified.

Meanwhile, lets move this discussion to https://github.com/OpenCPN/OpenCPN/issues/1839
Dave

rgleason, 2020-04-14:
Alec, I know you have been busy, but please don't forget to update for metadata.xml
https://github.com/leamas/OpenCPN/wiki/Installation-paths

Others will be using this a lot I think.  Thanks.

jongough, 2020-04-14:
I have updated both ODraw and testplugin_pi to the new metadata and have tested both on my Ubuntu system using the 'import plugin...' process and it appears to work OK. I did have to make sure that previous installs for debugging had been removed but this is not surprising. I did hand download the correct 'gz' file from cloudsmith, so when the webpage stuff is working it should be 'simple'.

bdbcat, 2020-04-15:
jon...
Great! Ready for commit to master catalog?

jongough, 2020-04-15:
I have put up a pull request to the plugins project. All is still running out of my Alpha cloudsmith as I have not made a 'tag' of the changes yet, been busy fixing an OD issue. With the new installation method OD is testable using the embeded metadata.xml file ( it works for me). The current OD is 1.6.61 and the current testplugin_pi is 1.0.68 .

jongough, 2020-04-15:
OD had built for all platforms apart from stretch-armhf which did a normal timeout (it is currently building). My OD is still on branch cmake_process_update as I have not yet commited it to the OD 1.6 main branch. So my builds are still going into the ocpn_draw_pi-alpha. The same holds true of the testplugin_pi. Are you happy for me to put a change into plugins with stuff still pointing at my

leamas, 2020-04-15:
Why not close this party while we are still in good mood?

That is, as I understand both @jongough  and @tai-fun (and implicitly also @bdbcat ) has reported "works for me" for the initial description. Unless anyone objects I'll  close this bug. If there are specific problems they could then be filed as such

leamas, 2020-04-16:
No objections, closing. Thanks to all of you!

closed 2020-04-16 by leamas