OpenCPN/OpenCPN #1839: RFE: Support downloading of specific plugin(s)

leamas, 2020-04-11:
This RFE depends on #1836. Once this is resolved it is possible for a developer to import a new plugin tarball which is not listed in any catalog.

A related use case is a user in a harbour which needs to install or update a plugin using an internet café  in the vicinity.  It is assumed that the plugin is in production and listed in a catalog. The steps are basically.
  - Take laptop to internet café or so.
  - Locate the plugin to use.
  - Download plugin and store it on an USB stick
  - Carry lapto + stick back to boat
  - Insert stick in target system and use new "Import plugin" functionality to use it. 

The problem for a regular user is to locate the plugin. The only reliable way to find it is to walk through thousand of lines of XML which is non-trivial. This RFE is about supporting this in a user-friendly way.

leamas, 2020-04-11:
This is probably best done using  a web page where user selects

  - Catalog (alpha, beta, or master)
  - Plugin
  - Platform
  - Version (often just one available).

After this, the appropriate URL is opened and downloaded by the browser.

The current hosting is known to support PHP and client-side javascript. Other languages might work.

Looking at the requirements, which does not involve any server state, the simplest solution might a single-page javascript application. 

See: initlal discussion in  #1836

rgleason, 2020-04-11:
I wrote Mauro, with Dave's request for some web expert help with PHP code for the Downloads page.

Enable the Plugins Download page to allow direct download from cloudsmith using the  plugin master, Beta and Alpha catalogs as a basis for creating weblinks in the opencpn downloads page.  Each of the catalogs have plugin records with the tarball urls. For each plugin, there will often be 8-10 different build environments each with a tarball-url to cloudsmith.  What is needed is the basic code to read the catalog, find the URL and make the active link.

leamas, 2020-04-11:
As I said, this is probably best done without PHP, using client-side javasciript instead.

rgleason, 2020-04-14:
Mauro's Response/Ideas

> Since we agreed that squidd.io will pitch in with the implementation , I spent some time reading this conversation and looking at the https://github.com/OpenCPN/plugins repo. I am mostly focusing on the Internet Cafe use case, but I think the implementation could be made to cover other use cases with some tweaks.
> 
> I want to make sure I am on the same page as you guys, so here's my current thinking...
> 
> 1- at an Internet cafe, user navigates to squidd.io/ocpn_plugins (or similar) from the opencpn.org download pages (or wherever else)
> 
> 2- a squidd.io page renders, with perhaps the Opencpn logo added to the top bar for navigating back to opencpn.org. Possibly using some of the ocpn css for buttons, labels etc. to make it more consistent with opencpn.org (luckily the two sites actually have a similar look and feel (blue top bar, white background, blue button )
> 
> 3- squidd.io retrieves the raw ocpn-plugins.xml catalogs from the relevant https://github.com/OpenCPN/plugins  branches and parses them and builds some kind of structured in-memory object containing all the relevant information. (if the real time retrieval of the .xml files with each user request makes it too slow, we could have a cron job running every 15 mins or so that caches the .xml files, but I suspect it won't be necessary, as I don't expect volumes to be that high)
> 
> 4- there may be a number of horizontal tabs at the top:  Master (or Production), Beta, Alpha, Experimental 
> 
> 5- below, is a set of cascading context-sensitive pulldowns (meaning the values in each pull down will depend on the values previously selected), possibly in this sequence:
> 
> | - Plugin name 
>     |-> Target Platform      
>           |-> Target Version
>                  |-> Plugin Version
>                      |-> Release 
> 
> Let me know id I am over-engineering this. Perhaps some defaults could be used to reduce the number of steps, such as latest version, latest release etc
> 
> 6- as the pull downs are selected, some of the relevant information (such as description, author etc. ) may be displayed in a separate box, for verification purposes
> 
> 7- when all the values are selected, a (very prominent) link is displayed at the bottom and user can start the download 
> 
> It seems from the later part of the discussion, that the actual tar file and the installation .xml will be bundled , so the link could point directly to the cloudsmith object , with no need for squiddio to intermediate and create a zip file. Am I understanding correctly?
> 
> Let me know if I am heading in the right direction.

bdbcat, 2020-04-14:
Mauro...
Your last para correct.
The tarball and metadata are bundled into one .gx file, and should be treated by your code as an opaque "blob".
Dave

mauroc, 2020-04-14:
@leamas : you are correct: this could end up being mostly a client side javascript implementation, so more easily portable. I just find it easier to use Ruby to parse XML files so there may be a few Ruby statements that eventually render a html/javascript page.

Also, you suggest selecting based on:

> - Catalog (alpha, beta, or master)
> - Plugin
> - Platform
> - Version (often just one available).

I was also including:

- target version
- release 

are these really necessary?  Fewer  steps would obviously be easier for the user

leamas, 2020-04-14:
Those are a bit different:

  - release is part of the version. 
  - target-version is *not* part of the version. However, the easiest for all is probably to present for example ubuntu-18.04 and ubuntu-16.04 as separate targets  selected by user.

leamas, 2020-04-14:
I need to brush up my frontend coding for other reasons. So, there is a mockup at http://opencpn.duckdns.org:3000/. Feel free to check.

Here are multiple rough corners, and no XML parsing at all (using dummy data). But should give a feeling of the ui.

Done using javascript, react, bootstrap and npm. Should be easy to host.

leamas, 2020-04-14:
@mauroc : If we should  make a full-fledged, layered solution one alternative is to parse the XML server-side and publish a REST API. Looking at my code it should be rather straight-forward -- as usual, the devil is in the error handling details.

Connecting the client js to such an API is trivial. I don't think that parsing the XML in js actuallly is that hard either. However, if there is a REST API it could stimulate other developments.

@bdbcat: What OS runs opencpn.org? Can we install software on it, or is it offered as-is w r t available tools?

Thoughts?

Hakansv, 2020-04-14:
> So, there is a mockup at http://opencpn.duckdns.org:3000/. Feel free to check.

Very nice! Simple and understandable for most users I suppose.

rgleason, 2020-04-14:
I also like this.

bdbcat, 2020-04-14:
Alec...
opencpn.org is hosted on a virtual linux-based server.  We have the capability of adding software to the host matrix.  Presently we are running SQL and PHP.
I personally would prefer that all the hard work be done on client side.  Maintenance of the server is at arms length for us.  When things go bad, they can take some time to identify and fix.  Meanwhile, users are stuck....

bcn2, 2020-04-14:
Alex..
for the mock-up:
please "macOS" instead of "darwin". Too much insider...

Hubert

leamas, 2020-04-14:
I have moved the mockup to http://opencpn.duckdns.org/ocpn-download/ (a "real"! deployment this time."

mauroc, 2020-04-14:
this looks like an excellent implementation. Since it appears the work has pretty much already been done, I am not sure my contribution is really needed at this point. If there is anything  you guys want me to help with, happy to pitch in!

bcn2, 2020-04-14:
:-))
[image]

leamas, 2020-04-14:
@mauroc:
It's actually not an implementation at all at this point -- the whole backend part is missing. Don't know when I have time to complete it. For now. it's a mockup.

@bcn2: This is the downside of client-side js. Initiating downloads is hard, constrained by security measures in the browser. To make things more interesting, the official download API (which does exist) is not  available in react.

I'm aware of this, assuming that our users can make the right choice here.

mauroc, 2020-04-15:
@leamas 
>  publish a REST API. Looking at my code it should be rather straight-forward

this makes sense. Let me take a stab at a simple REST API interface, based on the client you put together. Will prob. be a couple of days

leamas, 2020-04-15:
> Let me take a stab at a simple REST API interface, 

Appreciated! Will do.

The frontend sources are at https://github.com/leamas/ocpn-download. The actual file is src/App.js. Here you can find the class PluginList which is the database mockup. The methods here should  reflect the API.

It might make sense to make the API agnostic about the master, Beta and Alpha urls etc.  The client could just send the "raw" url.

leamas, 2020-04-15:
@mauroc: hm... an alternative to implementing all the methods in PluginList is that the API is a single method which given an url returns the parsed results as a JSON blob. The idea is that the backend makes life easier for the frontend by returning

  - version 
  - date
  - [plugin, plugin, ...]

Each plugin is a list of platforms as defined by *PluginList.getPlatforms()*.
Each platform is a list of versions.
Each version is a dict with a single item *download-url*.  (Using a dict to make i open for expansion with more data). 

This would make the API simpler to use (and perhaps to test), but will probably cause you more work. 

And, in the end, this is your show. Just thinking aloud.

mauroc, 2020-04-15:
> methods in PluginList is that the API is a single method which given an url returns the parsed results as a JSON blob.

yep - this is kind of what I had in mind...

mauroc, 2020-04-15:
PS - I was thinking of just adding an endpoint to the current [squiddio API set](https://squidd.io/api_docs) . I assume there is no need for authentication tokens or user identification etc.
BTW my Javascript skills are pretty basic, so if you want to run with the client when the API is implemented...be my guest :-)

leamas, 2020-04-15:
> yep - this is kind of what I had in mind...

OK. When you come to the platform part, please reach out -- there are horrible heuristics involved in that. The rest should be straight-forward.

> I was thinking of just adding an endpoint to the current squiddio API set . 

Hm... I  (and perhaps Dave) would like that the API eventually is hosted on opencpn's server. So  the best is probably to make it a separate, deployable  service.

> so if you want to run with the client when the API is implemented...be my guest :-)

This is what I have envisioned. Really happy if you take care of this...

mauroc, 2020-04-15:
moving the endpoint to a different server should be really straight-forward for a PHP developer. We are talking about some basic XML parsing and Json object formatting in response. The endpoint doesn't really belong in the squiddio API anyway.  I'll share the API code on github of course

leamas, 2020-04-16:
OK.  As I said: this is your show. 

Let's get things working using you existing setup. How to proceed is a later question.

mauroc, 2020-04-16:
ok, @leamas [give it a go ](https://squidd.io/api_docs)when you have a minute. Open up the category "ocpn_plugins". There are 3 endpoints, you need to start from the bottom one (I need to reverse the order when I have a chance). It should be fairly intuitive: you retrieve a plugin id for a specific catalog, then enter it in the end point above. I think this is what your react client is going to do....I am sure there are quirks and kinks , particularly error handling, but wanted to make sure the approach works for your React client.

I have not implemented the 'custom url' option but it should not be too complicated.

When it's all tested, I'll  just post a gist to github with the code. It's a simple Ruby class and DSL instructions to rails on how to configure endpoints etc. I am sure PHP has similar classes.

mauroc, 2020-04-16:
also forgot to mention: right now I do not return info like author, etc...specific to the plugin. I can add that to the response JSON blob for the plugins request, or open a separate endpoint. Whatever is easier ...

mauroc, 2020-04-16:
...finally, here's the discovery url for when you need the complete [endpoint urls](https://squidd.io/apps/v1/api/swagger_doc)

lskillen, 2020-04-16:
Folks - Cloudsmith also has an API that let's you enumerate all packages (and retrieve their specific download URL):
http://api.cloudsmith.io
https://help.cloudsmith.io/v1.0/reference

mauroc, 2020-04-16:
@leamas - hang in there ...I am having problems after moving the app to production. I'll let you know when it's working again

rgleason, 2020-04-17:
Thank you Lee.
That might be very useful for moving metadata files into the github plugin repos.
I think the plan is to use the plugin repository ocpn_plugins.xml  in  Master, Beta and Alpha Branches, but the API might be useful.    

I think Dave would like to have good control over the Master Plugins and Beta.

leamas, 2020-04-17:
@lskillen: While gentle, your proposals are a bit off-topic here. If you open a separate issue we can discuss if/how we could use the cloudsmith capabilities there. In short, enumerating cloudsmith repo(s) isn't what this is about.

leamas, 2020-04-17:
@mauroc: I don't have time right now to test. However, I have reviewed the API and have some remarks:

  - The client needs the catalog version and date (visible when expanding the catalog in the mockup). I don't see how they can be obtained.
  - The dict returned when looking for a version does not contain the url but instead what looks like test data `{pippo: 'azz'}`.
  - We need to define how errors such as "cannot access catalog url" and "cannot parse" are handled. (The responses looks sound, but docs seems missing).
 - The "experimental" catalog does not exist.
 - As per Hubert's remark we should present darwin as MacOS. Here we should also strip the minor version nums (always compatible) i. e., darwin-10.13.6 => macOS-10
  - Likewise  we should present msvc-10.0.14393 as windows-10
  - As you said, the Custom url is missing.

I also have some ideas:
  - I think it would be better if the client provided the url instead of the string "master", "beta" , etc. This will  push the work to translate for example master to an url to the client. 
  - What if we replace the three operations with a single one as outlined in https://github.com/OpenCPN/OpenCPN/issues/1839#issuecomment-613885509?  

To me, changes in this direction would create a simpler, more well defined API?  

BTW: TBH, I'm not convinced that the master/Beta/Alpha setup will survive in a longer perspective...

lskillen, 2020-04-17:
> @lskillen: While gentle, your proposals are a bit off-topic here. If you open a separate issue we can discuss if/how we could use the cloudsmith capabilities there. In short, enumerating cloudsmith repo(s) isn't what this is about.

Understood, although it sounded like you might have needed to get the specific download links for packages, and Cloudsmith is the source of truth for that. :) I am completely happy to defer until the next discussion. Just wanted to make sure that users get the most from the service. Cheers and good luck!

leamas, 2020-04-17:
@lskillen: 

> Cheers and good luck!

Thanks, and the same to you! 

I think I speak for all of us when I say that we are very happy using the cloudsmith service. 

Personally, I have an overall experience that the service is stable, flexible and reasonably easy to use. In particular, the cloudsmith-cli script is shining: it is python which actually is portable (ruby isn't...) and it also handles the required deployment secrets is a nice way.

PS: This is a monotheistic project: we believe in a single Catalog. Thus, we cannot worship other Gods such as the Cloudsmith API when we carry out our download rites. DS

lskillen, 2020-04-17:
> Personally, I have an overall experience that the service is stable, flexible and reasonably easy to use. In particular, the cloudsmith-cli script is shining: it is python which actually is portable (ruby isn't...) and it also handles the required deployment secrets is a nice way.

Much appreciated. Thank you. :-)

> PS: This is a monotheistic project: we believe in a single Catalog. Thus, we cannot worship other Gods such as the Cloudsmith API when we carry out our download rites. DS

😂 I'll chat about it in a different context elsewhere!

rgleason, 2020-04-17:
Being open source here is a discussion about a [duotheistic project](https://github.com/OpenCPN/OpenCPN/issues/1845)  where we can discuss the possibilities.

rgleason, 2020-04-17:
Look at this
https://api.cloudsmith.io/#/packages

mauroc, 2020-04-17:
> @mauroc: I don't have time right now to test. However, I have reviewed the API and have some remarks:
> 
> * The client needs the catalog version and date (visible when expanding the catalog in the mockup). I don't see how they can be obtained.

I hear you. Seems like both date and version are present at the head of the XML file. Perhaps I can structure the response to the `/ocpn_plugins/catalogs/{catalog} ` request as a dict, with 2 keys: date, version and 'plugins' (=> basically the array you are getting now.)

> * The dict returned when looking for a version does not contain the url but instead what looks like test data `{pippo: 'azz'}`.

sorry - my bad. Left some last-minute testing code in there. It displays the URL now
> * We need to define how errors such as "cannot access catalog url" and "cannot parse" are handled. (The responses looks sound, but docs seems missing).

> * The "experimental" catalog does not exist.

I am just going off of the branches on https://github.com/OpenCPN/plugins. I will remove it from the docs and the code

> * As per Hubert's remark we should present darwin as MacOS. Here we should also strip the minor version nums (always compatible) i. e., darwin-10.13.6 => macOS-10

everything I show is , of course, parsed from the catalog .xml files. If we want to change the nomenclature or version formatting, it will have to be in the catalog files.

> * Likewise  we should present msvc-10.0.14393 as windows-10

ditto
> * As you said, the Custom url is missing.

I'll get around to that.

> I also have some ideas:
> * I think it would be better if the client provided the url instead of the string "master", "beta" , etc. This will  push the work to translate for example master to an url to the client.

fine by me.
> * What if we replace the three operations with a single one as outlined in [#1839 (comment)]

 In general, fewer endpoints is better, I agree. I tried to implement a more REST-ful approach by using a cascading url but I'll give it some thought after looking at that discussion again

> 
> To me, changes in this direction would create a simpler, more well defined API?
> 
> BTW: TBH, I'm not convinced that the master/Beta/Alpha setup will survive in a longer perspective...

wanted to mention: I am getting some really odd 404 responses from the nginx server in production (not in development) for certain requests. Really weird - anyway I'll take a look at this first and deal with the rest later

leamas, 2020-04-18:
> everything I show is , of course, parsed from the catalog .xml files. If we want to change the nomenclature or version formatting, it will have to be in the catalog files.

Not really, this is a presentation thing. But no problems, I can make it in the frontend. 

My original thinking was that since you are tinkering with these anyway (merging os + os version) you could make this as well. Far from crystal-clear thinking, that was.

leamas, 2020-04-18:
>>   I   think it would be better if the client provided the url instead of the string "master", "beta" , etc. This will push the work to translate for example master to an url to the client.

> fine by me.

custom url resolved!

mauroc, 2020-04-18:
have a look at the [latest rev.](https://squidd.io/api_docs#!/ocpn_plugins/getOcpnPluginsCatalogs) It's down to one endpoint, with parameters passed as query params. Less rest-ful but easier to test. It takes the URL of a catalog file (this is the regex I use to validate the url `/\A[Hh]ttps?:\/\/.+\.(?:xml|XML)\z/` , it throughs exceptions if the catalog has not children named plugins, if it is inaccessible and a couple of others.

It still supports catalogs as names (master, beta etc.) but that can be removed. 

Finally, I added the catalog date and version to the plugins response, and the plugin info (authro etc.) to the platform response. 

Good luck with the client and let me know if you need any help

rgleason, 2020-04-19:
Tried:  
Master: https://github.com/OpenCPN/plugins/master/ocpn-plugins.xml
Beta: https://github.com/OpenCPN/plugins/Beta/ocpn-plugins.xml
Alpha: https://github.com/OpenCPN/plugins/blob/Alpha/ocpn-plugins.xml
Appear to fail. What am I doing wrong?  Raw does not seem to work either.

These return errors too.
https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-beta/raw/names/vdr-ubuntu-18.04-tarball/versions/1.2.0.4.4c0a196/vdr_pi-0.5.1-4_ubuntu-18.04.tar.gz

Please expain.

mauroc, 2020-04-19:
Rick
the urls must point to raw xml file, like:

https://raw.githubusercontent.com/OpenCPN/plugins/master/ocpn-plugins.xml

I should add this to the doc

also - the app only parses xml catalogs. You are feeding it tarball files...doesn't really know what to do with them!

rgleason, 2020-04-19:
**https://raw.githubusercontent.com/OpenCPN/plugins/master/ocpn-plugins.xml**
RESPONSE
{
  "attibs": {
    "date": "2020-04-18 20:39",
    "version": "0.0.0"
  },
  "plugins": [
    "ocpn draw",
    "weatherrouting",
    "oesenc",
    "radar",
    "oernc charts",
    "vdr"
  ]
}

**Type in "ocpn draw"**
RESPONSE
{
  "attibs": {
    "summary": "drawing plugin to allow extra objects to be drawn and used within ocpn",
    "author": "jon gough",
    "source": "https://github.com/jongough/ocpn_draw_pi",
    "descr": "ocpn_draw (od) is a plugin that extends the capability of ocpn to include non-standard object: boundaries, boundary points, text points, dead reckoning paths, electronic bearing lines (ebl) and guard zones. the plugin also has an api to allow other plugins to directly interact with it.",
    "info_url": "https://opencpn.org/opencpn/plugins/ocpn_draw.html"
  },
  "platforms": [
    "debian-10",
    "raspbian-10",
    "darwin-10.13.6",
    "fedora-29",
    "mingw-10",
    "flatpak-18.08",
    "ubuntu-18.04",
    "msvc-10.0.14393",
    "ubuntu-gtk3-18.04",
    "ubuntu-14.04",
    "ubuntu-16.04"
  ]
}

**Enter  msvc-10.0.14393**
RESPONSE
[
  {
    "1.6.59": [
  {
    "1.6.59": "https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.59-msvc-10.0.14393-msvc-tarball/versions/1.6.59.0+368.5a81c45/ocpn_draw_pi-1.6.59.0_msvc-10.0.14393-msvc.tar.gz"
  }
]
  }
]

From above, this link fails. I wonder if the catalog is right.
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.59-msvc-10.0.14393-msvc-tarball/versions/1.6.59.0+368.5a81c45/ocpn_draw_pi-1.6.59.0_msvc-10.0.14393-msvc.tar.gz

From Cloudsmith https://cloudsmith.io/~jon-gough/repos/ocpn_draw_pi-alpha/packages/?page=2  I  found this: 
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.63-msvc-10.0.14393-MSVC-tarball/versions/1.6.63.0+375.a2dd6fa/ocpn_draw_pi-1.6.63.0_msvc-10.0.14393-MSVC.tar.gz
Which works, but is a different version.

I found this one has the right version number and it works! : 
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.59-msvc-10.0.14393-MSVC-tarball/versions/1.6.59.0+368.5a81c45/ocpn_draw_pi-1.6.59.0_msvc-10.0.14393-MSVC.tar.gz

The difference is the "MSVC" caps.  This is a bit of a pain.

rgleason, 2020-04-19:
Sorry, I was thinking the metadata inside the tarball would work.

mauroc, 2020-04-19:
Rick
good catch. Try the urls now. It should work

leamas, 2020-04-19:
@mauroc: I suppose you have considered the brain-dead approach to just return everything. That is, if the query parameters are missing you just send data for all plugins, all platforms and all versions in one single json blob?

I'm interested because this would be the simplest for me (the client). And, I presume this would be simple also on the server side. 

The size of that thing should not be a problem, < 100 kb uncompressed.

rgleason, 2020-04-19:
https://dl.cloudsmith.io/public/jon-gough/ocpn_draw_pi-alpha/raw/names/ocpn_draw_pi-1.6.59-msvc-10.0.14393-MSVC-tarball/versions/1.6.59.0+368.5a81c45/ocpn_draw_pi-1.6.59.0_msvc-10.0.14393-MSVC.tar.gz

Works now

rgleason, 2020-04-19:
This is like a giant filter, that presents options which refine the target download. I realize that we are using the underlying structure and the actual interface will be different. 

As each step is selected, will the options be then presented in a dropdown in the next field down?

Will we be able to select from known things like Master, Beta, Alpha  or type that in?
Can we put in the url of any xml catalog?   Even the url to a single metadata.xml catalog?

leamas, 2020-04-19:
The interface is as defined in #1839. This discussion with mario has nothing to do with this

rgleason, 2020-04-19:
This is going to be a dynamite system! Thanks guys.

mauroc, 2020-04-19:
@leamas 

> I'm interested because this would be the simplest for me (the client). And, I presume this would be simple also on the server side.

funny - I thought I was making your life easier by breaking up the responses in smaller chunks :-).

In any event, if you now specify 'all' in the plugin id field, you get a big ol' json object of the entire catalog.

leamas, 2020-04-19:
> In any event, if you now specify 'all' in the plugin id field, you get a big ol' json object of the entire catalog.

> It takes the URL of a catalog file (this is the regex I use to validate the url /\A[Hh]ttps?:\/\/.+\.(?:xml|XML)\z/ , it throws exceptions if the catalog has no children named plugins, if it is inaccessible and a couple of others.

Thanks! 

The thing is that all fetches from the API are asyncronous, so each of them adds some complexity. And, navigating in a json structure is after all trivial. The only reason to retrieve it piecemeal is if it's too big -- which it isn't.

I see no need to remove the possibility to use 'master', 'beta' etc. as url. They are certainly useful when testing.

leamas, 2020-04-19:
Minor remark: Specifiying 'all' for plugins seems to work just fine, but perhaps there should be a note in the API docs?!

rgleason, 2020-04-19:
I might not know the full environment, just the ubuntu or msvc  but that produces invalid.
I might not spell the plugin exactly right it could be WeatherRouting, Weather Routing, WeatherRouting_pi, weatherrouting, weatherrouting_pi.  

It could even be more incomplete than that,  "weather"  or Weather" 
The could it return  WeatherRouting_pi  https and Weatherfax https if they were both available in the catalog?

Do we need to use the full and exact "plugin name" and "platform"  we humans are very incomplete and lazy and expect things to just work.

leamas, 2020-04-19:
It doesn't really matter. The frontend will use 'all' to retrieve all data, and will then be able to present the alternatives for the user.

You must, really must, stop trying to understand the user interface by reading this internal API. They are just not related. The user interface is as of http://opencpn.duckdns.org/ocpn-download, the rest is implementation details.

rgleason, 2020-04-19:
Ok I will stop bothering you.

mauroc, 2020-04-19:
@rgleason 
there is, however, a somewhat related consideration: I don't how how the catalog files are created, but if there is any manual input in the process, we will have to be careful that identifiers for plugins and platforms be spelled consistently. For instance, I found an instance of plugin `Radar` and one of `radar`, which caused my app to see them as two separate plugins. 
I remedied by standardizing on lower case, but there may by other similar situations we'll have to look out for.

rgleason, 2020-04-19:
I think the capitalization issue should be solved by simply forcing all use lower case, for a completely consistent rule.  Just use it everywhere. How do we enforce this?

Also if there is a space or gap it should always be an "_"

[54 more replies]

closed 2020-04-23 by leamas