OpenCPN/OpenCPN #1840: Lock file folder not available on virgin system

transmitterdan, 2020-04-11:
On a virgin Ubuntu system when starting O for the first time a popup box displays with message:
```
Failed to inspect the lock file '/home/user/.opencpn/_OpenCPN_SILock' (error 2: No such file or directory)
```
I think this suggests there should be some test for existence of the .opencpn folder before trying to stat the lock file.

leamas, 2020-04-16:
Indeed. At a glance, by adding that test to `OCPNPlatform::Initialize_2()`...

closed 2020-04-17 by bdbcat

asy972, 2021-02-07:
I see this message when running 5.2.4 in a similar situation. Commit c0ad3f3 exists in source tree. OpenCPN compiled by me.

leamas, 2021-02-07:
Hi,

> I see this message when running 5.2.4 in a similar situation.

Could you please expand a little on "similar situation"?

asy972, 2021-02-07:
> Could you please expand a little on "similar situation"?

rm -rf ~/.opencpn && opencpn

asy972, 2021-02-07:
btw. it would be good if error message can be copied from this window.

leamas, 2021-02-07:
It's hard to copy text from a screendump...

That said, this looks like a plain bug. It's about the (too)  large function MyApp::OnInit in chart1.cpp. The code which uses the _OCPN_Silock file is around  line 1740 while the part which creates the config dir is in the function Initialize_2(), called from around line 2225 in OnInit().

Which means that the config directory is used (assumed existing) before it actually is created. Needless to say, this requires a fix.

leamas, 2021-02-08:
@transmitterdan: Can you reopen this bug?

transmitterdan, 2021-02-08:
It appears I cannot re-open.  Or at least cannot see a way.

leamas, 2021-02-08:
@transmitterdan: OK
@asy972: Could you please file a new bug (keep it simple, just link to this one).

asy972, 2021-02-08:
> Could you please file a new bug (keep it simple, just link to this one).

I'm late. :-)

PR #1847 (2020-04-17): OCPNPlatform: Ensure private data dir exists (#1840)
Create the GetPrivateDataDir() dir if it doesn't exist at startup.

Closes: #1840