OpenCPN/OpenCPN #1841: ENC charts not displaying correctly

transmitterdan, 2020-04-11:
Latest github master [...304e6](https://github.com/transmitterdan/OpenCPN/commit/5a50812fa0594f012ca03094fd44c6af9d8304e6) does not display ENC's correctly.  When no chart groups are selected only raster charts display even if ENC piano key is clicked.  On Raspberry Pi the raster chart is always displayed.  On Windows the chart region is dark but routes and waypoints display.

If the selected chart group contains only ENCs then the ENC charts will display correctly.

bdbcat, 2020-04-12:
Dan...
Sorry, cannot reproduce on linux/AMD64.
Anyone else?

transmitterdan, 2020-04-12:
Dave,
All I have to do is select chart group "All Active Charts" and then ENC's do not display.  On Window the chart outlines display but the chart displays black.  On RPi the raster chart continues to display but no ENC will display.
If I select a chart group consisting only of ENC then the ENC's display correctly.

bdbcat, 2020-04-12:
Dan...
Something funny with your chart database or groups, I guess.  No problem here on rPI4.
I'll check W10 next.

Hakansv, 2020-04-12:
On Win10, today's build.
I can really see some strange behaviors on chart group "All"  even though not exactly the same as Dan describe it.
Quilting between different ENC types is a mess. Hard to describe but see the shot and a movie (28M): https://www.dropbox.com/s/gfs5y3778tlna03/W10_ENC_hakan.mp4?dl=1
As seen from the chart bar also dongle files are listed. But no dongle is mounted! It's though hard to tell is O really tries to view the dongle files. Probably not?
The shot below is another example.
Using a chart group with pure ENC and rasters works fine though. 
How can we test this in more details to trace it?

[image]

transmitterdan, 2020-04-12:
Dave,
I deleted the files in SENC folder and rebuilt database.  Same problem.  These are the latest ENC files from NOAA.  Location is San Francisco Bay.  On left side only ENC charts are in the group.  On the right all charts are in the group and I clicked on a green piano key.  I should see the same charts on both sides.

[image]

bdbcat, 2020-04-12:
Sorry, guys.
I am not seeing anything like this at all.
What we need to do is go back in history to find a version that works correctly, and then do some git bisect to identify the offending change.
@Hakan...
Can you disable oeSENC plugin, and reproduce Dan's results in SFO?

bdbcat, 2020-04-12:
I would try to checkout and build:
e827039e5f717c954869678c3fa0f5844fa18df5

Long time ago....

hakan, 2020-04-12:
I can’t @bdbcat but maybe @Hakansv can.

bdbcat, 2020-04-12:
Dan...
Now I can see something similar.
Setup:
Split screen, California NOAA ENCs, ENC group set in both windows.
Result:  Right window display problems.
Investigating further...
[image]

bdbcat, 2020-04-12:
https://github.com/OpenCPN/OpenCPN/commit/e827039e5f717c954869678c3fa0f5844fa18df5 works fine.
Bisecting now, takes a while...

Hakansv, 2020-04-12:
Dave..
Just for info is there's for me no difference when head is e827039. Still the same display errors. But didn't tried wo oeSENC

Hakansv, 2020-04-12:
Dave..
Head is still e827039. oeSENC/oeRNC plugin disabled.
Chart group "All". Here both canvases are affected. When using a chart group where ENC and raster are included, no problem

[image]

bdbcat, 2020-04-12:
Hakan...
Doe this still happen in single-canvas mode?

bdbcat, 2020-04-12:
Hakan...
Does it happen on rPI build?

Hakansv, 2020-04-12:
Dave..
Single canvas --  Yes.

Will check on RPi3-stretch. I'll be back
[image]

Hakansv, 2020-04-12:
Dave..
RPi3, yesterday's master built:
No, I couldn't reproduce so far on the RPi3. Neither in single nor double canvas. I'm using the same chart sets so in principle the same test scenario. All cells of the ENC set are shown at the same zoom level all times. A difference may be the zoom speed is considerable more slow due the system capacity?
Håkan

Hakansv, 2020-04-12:
Dave..
Without OpenGL still disturbed but on another way. The picture below is Win10 wo OpenGl. Not rendered areas instead of a mismatch. Still; Using a chart group (1,2,3...) no problem. "All"(0) and the effect is there.
[image]

transmitterdan, 2020-04-12:
Dave,
I think bisecting will not help.  I went back to the current release version (v5.0.1_MAC) and still the problem remains.  I believe my original issue was triggered by something that changed in the actual MBTiles files produced by NOAA.  If I remove the folder containing MBTiles from the list of chart folders then the problem goes away for me.  So I think this is some kind of new data issue that has triggered a really old bug.

I updated MBTiles using chart downloader yesterday.  That coincides with the introduction of this issue.

This is what is in my MBTiles chart folder:
[image]

Hakansv, 2020-04-12:
But I'm on Danish Pure ENC charts from about 2018. And why on "All" but not any chart group.

Hakansv, 2020-04-12:
Now I've built down to V5.0 and still the same issue. It may be I haven't used "All" in a frequent way since very long??

bdbcat, 2020-04-12:
I agree that bisecting will not show us anything.  My bisect trial was inconclusive.
I think there is some un-initialized variable somewhere, that only occasionally causes the trouble.
I will snoop around, and try to reproduce again.

transmitterdan, 2020-04-12:
Dave,
This is a link to my MBTiles folder which I confirmed triggers the problem.  It's a big file (3.5GB).
[MBTiles.zip](https://www.icloud.com/iclouddrive/0RBvZcGU6xxIYJv2M6Yft5H0Q#MBTiles)
Dan

bdbcat, 2020-04-13:
Hakan...
Maybe I figured this out.
1.  Dan's problem seems to relate to new MBTiles set.  I will download overnight, and test tomorrow.
2.  Your problem:
Trying to quilt standard S57 ENC with oeSENC charts, but oeSENC is not available (either due to missing dongle, or disabled plugin.)  I can see a potential logic "hole" in quilt code.  But this scenario is not often seen, since one would rarely have S57 overlapping oeSENC,  At least for me, I am having a hard time finding such a place in the world, among my chartsets.

I could probably fix this, but I need to reproduce it first....
Tell me again what the ENC and oeSENC situation at your test location is, please.

Hakansv, 2020-04-13:
Dave..
Yes, I can confirm your conclusion that the quilt problem is related to a mix of oeSENC files and pure ENC.
If I use a group with only ENC - no problem. If I add geographically overlapped oeSENC files to that chart group I see the quilt problem. Either if oeSENC_pi is activated or not. 
Probably a rare user situation. I agree to that as well. But if you'd like to spend time on it my set up is the Danish oeSENC and rare pure Danish-ENC. You may have them but I'll send you a link.

transmitterdan, 2020-04-29:
I discovered another possibly related phenomenon on Rpi 4 (Buster).  This is ENC charts from NOAA with OpenGL turned off:

[image]

With OpenGL turned on no issue with just ENC in a group.
I have been beating up the Rpi a bit with lots of AIS and VDR playout to simulate a real voyage.  I have seen a few random Segfaults when running in gdb.  They don't seem to be repeatable.

transmitterdan, 2020-04-30:
Update: I have been debugging a bit more on this issue.  I have found evidence there is heap corruption when I run on the Rpi 4. Each platform has different heap management so it is likely that some platforms behave differently.  But I am convinced something is wrong.

I have installed only latest ENC files from NOAA.  No mbTiles, or RNC or CM93.

I believe that the SENC data is corrupted before it is written to disk.  The code that reads SENC files does not do much sanity checking which can lead to heap corruption. I can easily get O into a state where it will never start because it cannot display the cached SENC data without crashing.  If I delete the SENC folder then O will start once only.  After that it will not start again without crashing.
Turning OpenGL on helps but does not completely fix the issues.

bdbcat, 2020-04-30:
Dan...
Cached SENC files are platform independent.
So, if you can send me a corrupted SENC file, maybe I can dig into it and find some crumbs to point to the root cause of corruption.
Dave

transmitterdan, 2020-04-30:
Dave,
I can send files. But the “bad” files are created by latest git master build. Will post file tonight.

bdbcat, 2020-04-30:
Dan...
Never seen bad SENCs or corruption on my builds.  So I look forward to seeing your built SENCs.

transmitterdan, 2020-05-01:
Dave,
Here are the files.  The SENC files were created on RPi 4 with OpenGL turned off.  This seems to make a difference in my testing.

https://www.icloud.com/iclouddrive/0ZGj4O02faiuNdXe0rVGO6Cag#DANSENC

https://www.icloud.com/iclouddrive/0rzSVgUzAR3_LXztKMxCjj42w#ENCCHARTS

bdbcat, 2020-05-01:
OK, got'em.

bdbcat, 2020-05-01:
So far, the three SENCs in DANSENC work fine on full size linux machine.
Do I understand that they cause a crash on RPi4, no OpenGL?

transmitterdan, 2020-05-04:
Dave
Here are steps to reproduce on Rpi 4 using NOAA ENCs I posted.

1. Delete SENC folder
2. Start O with OpenGL=1, LastChartScale=20019.5 & OwnShipLatLon="   37.8941,  -122.7966" in opencpn.conf
3. Wait for SENC builder threads to finish.
4. Close O
5. Change opencpn.conf OpenGL=0
6. Start O
Now O starts slowly and eventually I get this at the command prompt:
```
pi@raspberrypi4:~ $ opencpn
 SENC: nCPU: 4    m_max_jobs :3
Segmentation fault
```

transmitterdan, 2020-05-04:
Forgot to mention my cmake line is this:
```
cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=CRUDE -DCMAKE_BUILD_TYPE=RelWithDebInfo -DOCPN_USE_CRASHREPORT=OFF -DOCPN_ENABLE_PORTAUDIO=OFF -DOCPN_ENABLE_SYSTEM_CMD_SOUND=ON -DOCPN_FORCE_GTK3=OFF ..
```

transmitterdan, 2020-05-10:
@bdbcat 
Still testing this issue on RPi 4.  It is triggered by display size.  My display is huge as you know (3840X2160).  If I make O smaller on the screen in height then the display problem goes away.  So looks like it requires a huge Y size to trigger.

transmitterdan, 2020-05-11:
@bdbcat: update on testing: I recently upgraded the kernel of my RPi 4 and after that no more heap corruption. The only issue I can easily reproduce is the bottom part of the ENC chart area is not displayed right (greyed and jagged lines) in non-GL mode only. No issues with OpenGL turned on. I consider this a minor issue as it only affects huge screens (>~1800 vertical lines) in non GL mode. Most RPi users must use OpenGL.

If no objection, I will close this and I’ll keep looking for the display issue in the background.

transmitterdan, 2020-06-23:
This issue no longer happens with latest RC.  Closing....

closed 2020-06-23 by transmitterdan