OpenCPN/OpenCPN #1842: Current status of github OpenCPN for AMD64?

hreuver0183, 2020-04-12:
I was cleaning up my old laptop and backup navigation pc with a fresh installation after X years of using a rolling release. New installation:
X86_64 (AMD64), Debian Sid
OpenCPN github (git pull today 5.0.522)
oesenc_pi (git pull today 4.2.4)

The first thing I was hoping for was testing the plugins from the plugin manager.
I cannot find any plugins other than those included in  OpenCPN sources (?) both with the update of master and beta.  Is the Debian installation still excluded or is there a general rule to only allow plugin downloads from known build sources?

Second point (perhaps GTK3) was that getting oesenc_pi to work was a real struggle. Not in the least because libsgllnx64-2.29.02.so was not copied with oeserverd. Do I need to create another package for libsgllnx64?

I was hoping to use the other packages using the PluginManager therefore those are not tested.

bdbcat, 2020-04-12:
1.  There should be a few managed plugins available for Debian Buster.  Not tested with Sid, AFAIK.
Will you show us the contents of "/etc/os-release" on Sid ?
2.  What steps did you use to install oeSENC_pi?

hreuver0183, 2020-04-12:
Contents of /etc/os-release:
`PRETTY_NAME="Debian GNU/Linux bullseye/sid"`
`NAME="Debian GNU/Linux"`
`ID=debian`
`HOME_URL="https://www.debian.org/"`
`SUPPORT_URL="https://www.debian.org/support"`
`BUG_REPORT_URL="https://bugs.debian.org`

Basically it is the same as testing (now: bullseye) with the addition of "Sid" (and newer packages).

For the oesenc_pi I used more or less the steps from the installation manual (developer part):
`mkdir build`
`cmake -DCMAKE_INSTALL_PREFIX=/usr ..`
`make -j 2 package`
and later as root
`dpkg -i oesenc_pi-4.2.4-0_debian-unstable.deb opencpn_5.0.522-1_amd64.deb`

bdbcat, 2020-04-13:
1.  There are no managed plugins supporting Debian 11 yet.  Sorry.
2.  There is no libsgllnx for Debian 11 yet.  Also sorry.
I'm afraid you are out on the bleeding edge....

hreuver0183, 2020-04-13:
Clear.

I can close this issue.

closed 2020-04-13 by hreuver0183