OpenCPN/OpenCPN #1843: wxWidget dll missing from package build

Rasbats, 2020-04-13:
Win8.1 Visual Studio Community 2017

Made own compilation of wxWidgets (new compilation today) as per the manual.
Built OpenCPN using the latest master code downloaded today. Made package.
Using this installer found that all(?) the 'wxmsw312u' dll were missing. They were present in my 'vc_dll'.
Copying all the missing dll into the program folder got OpenCPN working.

bdbcat, 2020-04-13:
Rasbats...
By design.  Since Windows does not have a mechanism for installing wxWidgets into the system directories, for OCPN the wxWidgets dlls are prebuilt, and are carried in the buildwin directory.  This directory is not part of the OCPN source tree, for size reasons.

Glad you found a workaround for your custom build of wxWidgets.

Rasbats, 2020-04-13:
Thanks Dave. Should have done more reading I think.

closed 2020-04-13 by Rasbats