OpenCPN/OpenCPN #1844: Move to Hosted Weblate translation platform

comradekingu, 2020-04-13:
Hello, I want to help translate this, as the archipelago is treacherous enough without having to make due with half a translation here in Norway.
Crowin is a very lackluster tool in terms of getting anywhere, both in terms of use and coverage.
There is a lot more activity over at Hosted [Weblate](https://weblate.org), which is a libre software tool that unlike Crowdin, doesn't sell profiling services. Click to ask for an account, and I'll get around to translating this. My grandfather taught me how to map-plotting manually when I was a kid, I'll show him when I go visit this summer how far the world has come :)

nohal, 2020-04-13:
What exactly is **blocking** you from translating on Crowdin, or if you personally have anything against Crowdin, sending a pull request with the PO files created any other way, that would account for all the work to both technically migrate and move all the people used to Crowdin to a different platform you personally prefer?

comradekingu, 2020-04-13:
The tracking and profiling it does, which is in turn sold to advertisers. The inability for anyone under 18 to participate, which I imagine for that reason. Don't know if OpenCPN is paying for hosting at Crowdin, but unless that is the case, the guidelines stipulate the project can't have a source of revenue to qualify. Either way that isn't optimal.

There are plenty OAuth 2.0 providers for logging it at Weblate, like GitHub/GitLab etc. logins, and the community is already a great deal bigger. I think my preference takes a backseat to the implication of spying, but that is a factor too. Just doing it the old way doesn't grant all the checks, overview and communication to be had.

Fairly sure Crowdin doesn't have a "git-wins" option either, so it would most likely create conflicts.
I am just seeing standard GNU gettext files and Android resource files, so smooth sailing with good integration. Staying on Crowdin for as long as the project has been on there hasn't garnered a lot of resulting translations, so speed and community might help in that regard?

I can promise fixing and finishing the nb_NO tranaslation and helping out in the other languages I know, and help administrate matters if added as an admin on there. Send the request and see how far it takes you :)

nohal, 2020-04-13:
Sending a request will pretty sure take us nowhere but to a place where we have more work to do and no free hands to do it, unfortunately.
All that is realistically feasible at the moment in case you are concerned about Crowding tracking and profiling you, is to take your fixed nb_NO (Or any other) PO file and import it to Crowdin keeping the current, working, workflow in place.
I can of course do it under my own identity, which is poisoned enough by being the owner of the OpenCPN project on Crowdin for it's whole existence.

And yes, I did read https://support.crowdin.com/privacy-policy/ but honestly don't see anything that would be out of today's "standards" for "free" hosted services.

And believe me, our problem is not whether the new service supports OAuth or not, although I would love that to be.

bdbcat, 2021-04-13:
We will continue to use crowdin for the reasonable future

closed 2021-04-13 by bdbcat

comradekingu, 2021-04-13:
Even if I did only upload there, I would be subject to https://support.crowdin.com/sub-processors/#crowdin-sub-processors. Also I would accept terms relating to my prior work, which is why I can't log in.

Whatever today's standards for "free" is, that isn't the comparison to make. Nor does it take into account the alternative. Furthermore, it doesn't qualify for "free".

>  If you're building awesome non-profit projects that could use the power of Crowdin, we're happy to help with this free option.

The privacy policy is bad enough, but it hides a lot in these documents:
https://support.crowdin.com/cookies/
https://support.crowdin.com/gdpr/
https://support.crowdin.com/terms/

It is possible to set up projects on https://hosted.weblate.org/hosting/ and then request confirmation once everything works. Happy to help.

bdbcat, 2021-04-13:
Give weblate a try if you like.  But this is very low priority, and as you know we are severely programmer-resource limited.