OpenCPN/OpenCPN #1845: Duotheistic project:  Support of cloudsmith api

rgleason, 2020-04-17:
I can see real some advantages to using the Cloudsmith API to allow browse, and installation of plugins.

Certainly for developers this would be a very direct and preferred way to check and test plugins that have build.

The main problem becomes maintaining a central source of plugins that are "vetted"  and "master" for users.

I for one, am not wedded to using the github.com/OpenCPN/plugins tool.  I find it somewhat clunky and discontinuous to be quite frank, and I wonder if we could not use the Cloudsmith repository and api to build a better, smoother mousetrap.

At this point it is probably too late, and this should be saved for a rainy day.

rgleason, 2020-04-17:
Look at this
https://api.cloudsmith.io/#/packages

leamas, 2020-04-17:
If we should enhance the UX, this is about the mockup at http://opencpn.duckdns.org/ocpn-download.  So far, I have understood that we are happy with this interface.

To create something like the proposed UI we need to parse the catalog, it's the only available source of the info required. Doing so, we also get the download URL.

In this context the cloudsmith API:s IMHO doesn't add anything.

leamas, 2020-04-17:
> Certainly for developers this would be a very direct and preferred way to check and test plugins that have build.

I would say, certainly not?  Devs can just import the plugin when they have built it. In general, a developer knows where the plugin she just built ends up and doesn't need to look for it. This is basically the #1836 use case.

rgleason, 2020-04-17:
For "Custom" why not have as an alternative, utilization of the Cloudsmith API for browsing the "OpenCPN Organization" repositories (Team member repositories, by invitation)?

This would be very handy and would fit within your MONOTHEISTIC main structure, but provide easier access for testers and developers.

leamas, 2020-04-17:
hm... do I get this straight?

The problem /use case you wan to address is developers and their need to locate the plugins they have built, before they are available in any catalog?

In order to fix this you propose:

  - That plugins are stored in a common cloudsmith directory
  - That we use the cloudsmith API to implement some browsing functionality in this directory.

Correct?

IMHO, this is a tall bill for this use case. In particular, having to store the plugin in a common directory is problematic, especially for new developers fir by definition don't have access to it.

rgleason, 2020-04-17:
Alec, No, you have misread this.

Provide a custom url option that 

- Uses the cloudsmith API to implement some browsing functionality
- Amoungst the Cloudsmith "OpenCPN Organization" Team Member repositoriies

leamas, 2020-04-17:
From the beginning: what is the actual problem you are trying to solve?

rgleason, 2020-04-18:
Users who are obliging by testing plugins won't deal with complex installation paths and developers are not going to be around to find and provide the path to the plugin. 

If you want an serious testing program, make it easy to load the plugin and get testing, don't put up barriers. A healthy Beta testing environment improves the quality of released plugins.  Most users are willing to try the plugin out and will actually "get into it" and test, but they get frustrated and will not dig into paths and complex installs.  I am talking about users from most environments here.

leamas, 2020-04-18:
OK. Progress. So we have three use cases handling import of separate plugins not immediately available in the install dialog:

  - The developer use case -- described and fixed in #1836 
  - The end-user use case -- described and fixed in #1839 
  - The testing use case -- yet to be described.

Can you describe the testing use case in short, bullet way like I did in #1839 and Dave in  #1836? 

Also, we need to understand this new use case compared to the beta and alpha catalogs which is the streamlined test environment.

rgleason, 2020-04-19:
Please see #1850

closed 2020-04-19 by rgleason