OpenCPN/OpenCPN #1849: distribution tarbals are not gnu-tar compatible

leamas, 2020-04-17:
As heading says: The tarballs created by cmake are not fully compatible with gnu tar. While they work for most users, this sometimes causes subtle errors downstream. Examples includes [1] which is triggered by the cmake tarball, but also recent problems in oesenc_pi when trying to add a metadata file to an archive.

tar is available on all platforms. Given this, it shouldn't be that complicated  to create a generic solution which is applied the main opencpn code which repacks that tarball using gnu tar. Hopefully, it can then also be used in the plugins to simplify the task to add metadata (and avoid possible glitches caused by the cmake tar format).

[1] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=831870

leamas, 2020-04-17:
hm... @bdbcat: How are the distribution source tarballs (those available on github) created?` make package` or directly from git?

bdbcat, 2020-04-17:
I assume directly from git, tar-ring the master source tree when requested.
Or maybe I don't understand the question?

leamas, 2020-04-18:
As I see it, there are two possible ways to generate a tarball:

  - Directly from git, using git-archive .
  - Using `make package_source`.

Other ways using git such as creating a tag and download that from github is basically the first option.

The reason I ask is that the only explanation I can find to the downstream Debian bug is that the tarball is created using `make package_source` (that is, by cmake).

We could of course close this bug by just deciding that next release will  use a source tarball directly from git. But then again, in my experience having a last step like make package_source is good since it can do some post-processing. In particular,  it can mark the tarball with the git hash and tag. This cannot be done (reasonably)  using git-archive

@nohal: have you any input on my question?

If I don't get any more input on this I'll proceed from the assumption that we just don't remember.

nohal, 2020-04-18:
Which tarballs are we talking about? We do not create any using cmake for OpenCPN. And from https://opencpn.org/OpenCPN/info/downloadopencpn.html we link to https://github.com/OpenCPN/OpenCPN/releases/tag/v5.0.0 for source, where the tarballs are generated by Github automatically when it receives a tag and there's nothing we can do about them short of generating some other tarballs ourselves.
In the referenced bug they say it is a bug in `tar` and I don't see any mention of OpenCPN or tarballs originating from the Github automatic system, so maybe try to show me the broken workflow relevant to us so that I understand better.

leamas, 2020-04-18:
We are talking exactly about the github tarballs. And since those are generated using git-archive,  we can drop this issue as for OpenCPN.

Given the downstream situation I suspected that they were generated using `make package_source`.  But obviously not. Thanks for info (and sorry for the noise). 

It is a known work-around to repackage using gnu tar to get around the problem, hence my suspicions (basically, I don't think Debian don't understands what's going on). There has been similar problems in the plugin buildchain -- here, the solution is to repack.

This could the end of the story, but it turns out there is another bug related to this: make package_source creates empty tarballs. This is a simple fix, there will be a PR for that part.

leamas, 2020-04-27:
New bug for the bad source tarballs. Remaining stuff closed, basically NOT-A-BUG.

closed 2020-04-27 by leamas