OpenCPN/OpenCPN #1850: RFE: Find and install a specific alpha/beta plugin on Cloudsmith Repos

rgleason, 2020-04-18:
USE CASE: Plugin Tester:**

- FIND new alpha/beta plugin on Cloudsmith from a list of "owners" + "repositories" + "environment" + "plugin name"  
- USE [Cloudsmith API](https://api.cloudsmith.io/#/packages)
- SOURCE list of "owners" from  Cloudsmith "OpenCPN Organization"
- USES "Download of specific plugins" technique  #1839  
- INSTALL  "Import Plugin" process.  #1836  

Example: 
In github we know the user, by knowing the user we can see their plugins and what then have done.
Deployment:
Cloudsmith is the choice for deployment. 12 environments get deployed there in various stages of completion  in different repositories "Prod" "Beta" "Alpha". Navigating this is not simple.

"Duotheistic project: Support of cloudsmith api"  #1845

rgleason, 2020-04-18:
Cloudsmith advises:
 An organisation endpoint is currently available to get a list of all the organisations the requesting user is associated with however there is no search facility at this time to find users that belong to a public repositories. 

We need a List of members of a particular organization, which would be useful to allow user/testers to navigate to appropriate repositories.

This may have to initially be an internal list.

rgleason, 2020-04-18:
Note: There may be more appropriate ways to accomplish the same thing. Please make suggestions.

rgleason, 2020-04-18:
Kyle from Cloudsmith led me to a partial solution that works.  It is more work for plugin devs.

1. Plugin Devs who are members of the OpenCPN Organization are permitted to upload their plugins needing testing,  to an common OpenCPN Organization repository.  This one would be appropriate:  https://cloudsmith.io/~opencpn/repos/plugins-beta/packages/
2. Which will  allow the existing Cloudsmith API to be used to fulfill our needs in find a specific plugin to test, and download it.  Use API to filter "plugin name",  "environment" etc, to find the right one.
3.  Then  install the plugin using "Import Plugin" process. #1836

leamas, 2020-04-18:
Before running into implementation: If I rephrase your use case like this

  - The developer generates a new build of a given plugin.
  - Developer informs tester that a new build exists for this plugin.
  - Tester locates the actual url for this build.
  - Tester downloads and imports the plugin using the new "Import plugin" functionality.

Can we agree that this description cover your use case? And that the problem to solve is the "locates the actual url" part?

rgleason, 2020-04-19:
Yes! precisely. Thanks.

rgleason, 2020-04-19:
https://cloudsmith.io/~opencpn/repos/plugins-beta/packages/

The discussion:

Kyle at Cloudsmith:
Yes,  if that group of developers were invited to the organization, they  could indeed upload to one or more common repositories within the  organization. Depending on your needs it could be as simple as one  common repository with tagged versions for alpha, beta, etc or you could  still split this into multiple repositories with a single type of  release. In either scenario, you could fetch everything using the  existing API.
Kyle from Cloudsmith

Sat, Apr 18, 2020 at 12:47 PM, "Rick Gleason" :
1. Can Plugin Devs who  are members of the Organization upload to several existing common  directories?  Would this allow the existing API's to be used to forfill  our needs enumerating the repositories using the existing API?

Sat, Apr 18, 2020 at 08:19 AM, "Kyle Harrison"  
Thanks for explaining  the use-case and linking to the Github issue. I can open a feature  request to add a members endpoint to support this however it will need  to be prioritized and it would likely be a while before it's picked up.  Currently the team is working hard on implementing improvements for all  customers globally (you can follow some of this over on https://changelog.cloudsmith.com/en). In  the short term, what I can suggest is that there may be an alternative  way to manage repositories by bringing users into an organisation, and  giving each user their own repository. This would allow the existing  API's to be used to fulfil your needs by enumerating the repositories  using the existing API.                        

Sat, Apr 18, 2020 at 08:07 AM, "Rick Gleason" 
Well actually what I am looking for is a list of members of a particular Org. The Org and members are open source. This would be useful to allow users to navigate appropriate repositories for testing plugins. 

Sat, Apr 18, 2020 at 08:06 AM, "Kyle Harrison"
Hi Rick, you are  correct. An organization endpoint is currently available to get a list  of all the organizations the requesting user is associated with however  there is no search facility at this time to find users that belong to a  public repositories.

You have "Org" but we would need "Members" wouldn't we?

leamas, 2020-04-19:
https://github.com/OpenCPN/OpenCPN/issues/1850#issuecomment-615913408
> Yes! precisely. Thanks. 

OK. Given that description I can see at three or four alternatives out of the top of my head. All of them gives IMHO more bang for the bucks. 

None uses the Cloudsmith API, so let's stop discussing this until we eventually decides to go that way. I'll post the alternatives one by one.

Stay tuned

leamas, 2020-04-19:
Option 1: Automagically updated catalog for all builds

Here, the idea is that  we have a fourth standard catalog which is updated automatically when new plugins are built.   This requires server-side code.

Implementation - flow
  - The plugin upload script informs the server about a new build (the url)
  - The server retrieves the tarball from the url and extracts the metadata
  - The server updates and publishes the new catalog more or less instantly.

leamas, 2020-04-19:
Option 2 Email notification

Here the idea is that upload scripts and tester uses a common mailing list.

Implementation - flow
  - We start a new mailing list somewhere.
  - The plugin upload script sends a message  to the mailing list including the download url + selected metadata
  - The tester subscribes to the mailing list
  - Tester uses standard email client tools to search for messages for a build, finds the url and uses it to download the plugin.

If I were tester, I would probably filter those messages so they went straight to a folder. But, that's just me.

leamas, 2020-04-19:
Option 3 IRC bot

Here, the idea is that tester and upload scripts uses a chat channel like IRC to communicate.

Implementation - flow
  - We start a new IRC channel on #freenode
  - Upload scripts automatically sends a message to that channel (url + selected metadata)
  - Tester listens to IRC and uses link therein to download plugin

IRC is probably the simplest, but there are other chats out there. Twitter comes to mind, would work the same way.

leamas, 2020-04-19:
So, we have four alternatives:
 - Arrange our plugin cloudsmith accounts and build a client supporting browsing - downloading of specific plugins https://github.com/OpenCPN/OpenCPN/issues/1850#issue-602443077
  - Use an automatic catalog https://github.com/OpenCPN/OpenCPN/issues/1850#issuecomment-616068456
  - Using email notifications https://github.com/OpenCPN/OpenCPN/issues/1850#issuecomment-616069152
  - Using a chat like IRC or perhaps Twitter https://github.com/OpenCPN/OpenCPN/issues/1850#issuecomment-616070160

Thoughts?

rgleason, 2020-04-19:
Lots of good ideas here. Which ones will promote a responsive and active Testing Program the best?
Email or IRC would be the easiest I think, is that the best way to handle this? A lot depends on how the details are handled.

We need to think about the average tester. Generally they are interested in one environment and will test for that environment. They would want notices for plugin changes in that environment.
Some users will have two environments that they use, Win + Mac and some will add Android.

Also having a catalog with all Beta Plugins utilizing the plugin Beta channel is a very good idea I think.  For quick check turnarounds plugin devs have friends and can send the url directly.

We need to think about this more and discuss it.

leamas, 2020-04-19:
> Also having a catalog with all Beta Plugins utilizing the plugin Beta channel is a very good idea

I certainly don't have that idea. The Beta program is a vetted program, distributed to a reasonably large user base. Nothing here should change that.

What I'm talking about is a fourth option, call it experimental or whatever. It is not vetted in any way, anything built would be part of it.

rgleason, 2020-04-19:
Yes, I think that is good and will be used.

leamas, 2020-04-19:
The use case here is testing a plugin which is *not* in any catalog (today).  The Beta program is unrelated, a completely other prograstory.

Given that the beta program exists, is the usecase "test last build" important enough to motivate the efforts required to implement any of the four alternatives?

rgleason, 2020-04-19:
Good point. I don't know. ...Maybe not?  

Also if a Dev wants it really tested out, and not just some first runs, they would want it up in the Beta Catalog anyway, and it would be there for awhile, maybe updating it as things crop up.

Also, we have the Alpha catalog which could be very useful using meta-urls, so Dave does not have to accept the PR's, perhaps that Catalog is more appropriate for initial testing with a close knit group of testers?

What do you think Alec? How should we look at this?  We don't want to build some system that is not used.

leamas, 2020-04-19:
you are the tester here... I rest my case.

rgleason, 2020-04-19:
Let's see how what we have is used. I think it is going to be fine. Closing. Thanks for considering this question.

closed 2020-04-19 by rgleason