OpenCPN/OpenCPN #1851: UI is huge

Wouterdek, 2020-04-19:
At least on Windows, this application is not properly DPI aware when using 2 monitors with different DPI settings.  As the [Microsoft documentation explains](https://docs.microsoft.com/en-us/windows/win32/hidpi/high-dpi-desktop-application-development-on-windows), applications should rescale based on the DPI of the monitor the window is located on. It seems wxWidgets supports this, as I was able to find [a resolved issue](https://trac.wxwidgets.org/ticket/17721) and [a pull request](https://github.com/wxWidgets/wxWidgets/pull/334) related to this, but OpenCPN does not.

The result is that certain parts of the UI are huge when dragged onto the second monitor. See [this screenshot](https://user-images.githubusercontent.com/913101/79688809-0ec56200-8251-11ea-969a-6b90d4eaca13.png) for an example.

Luckily, as you can see in [this small commit](https://github.com/mooction/MyNotePad/commit/cbf9a9814ad97107d4ac5864ce89ea9f29222dc6) from a similar wxWidgets application, this may be easy to resolve. The key here is to scale each element with the right DPI value, and tell Windows the application supports DPI changes with the PerMonitorV2 value in the manifest file.

rgleason, 2021-04-14:
This seems well documented with a partial solution, perhaps someone could make a PR? 
However is it still an issue, and is DPI awareness to be able to drag to another display a frequent necessity?

rgleason, 2023-03-26:
I think this can be closed now.  @Bdbcat   or @Wouterdek ?   Can you close it?

rgleason, 2023-04-28:
It is time to close this.

nohal, 2023-04-28:
@rgleason did you test the behavior on a multimonitor system? I don't think we did much to address this.

rgleason, 2023-04-29:
Sorry, no. I will try to test.

rgleason, 2023-05-03:
Where I have an additional moniter.  Dell Laptop is high res, monitore #2 is low res.
[image]

bdbcat, 2025-04-26:
Closing OBE

closed 2025-04-26 by bdbcat