OpenCPN/OpenCPN #1852: Some issues with a switch to flatpak

hreuver0183, 2020-04-19:
Since my system is not supported the pluginmanager and the documentation says all non-supported Linux systems should switch to flatpak, I tried it.

According to the manual on Debian you start with:
 `$ sudo apt update`
 `$ sudo apt install flatpak`
 `$ sudo flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo`
etc

When I follow that script I find myself with OpenCPN missing for the user when calling "flatpak list". As root or with sudo OpenCPN is included in the list. (I realize that is probably due to unfamiliarity with flatpak.)

I think the manual should prescribe:
 `$ sudo apt update`
 `$ sudo apt install flatpak`
 `$ flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo`
 `$ flatpak remote-add --user --no-gpg-verify ocpn-beta http://opencpn.duckdns.org/opencpn-beta/website/repo/`
 `$ flatpak install flathub org.freedesktop.Platform//18.08`
 `$ flatpak install ocpn-beta org.opencpn.OpenCPN`
After running 
 `$ flatpak run org.opencpn.OpenCPN`
I find that symlinks do not work (more an issue of flatpak probably).
I find the package manager installs most packages I want, but the first package fails in use.

After installing oesENC there is the usual dialog (website, preferences, reinstall, uninstall) but preferences is greyed out (4.2.7). The oesENC dialog under Charts does not appear.

Since my github version of OpenCPN uses a complete different set of plugins I findmyself wishing for a commandline config option. I would very much like a option to choose between to different configurations, something like:
`opencpn -c .opencpn-github`
`flatpak run opencpn -a .opencpn-flatpak`
Although logging in as a different user for testing flatpak is an option, however not the most userfriendly.

So far my conclusions on my first trial run with flatpak, basic system works, manual should be improved, add-ons is still work-in-progress (AMD64, Debian Unstable).

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/4499148/opencpn.log)

rgleason, 2020-04-19:
hreurver0183,  I can make you an editor of the wiki so you can make these changes, if you would like.
Private Message me  (rgleason) on the CForum with your email.

bdbcat, 2020-04-19:
hreurver0183...
You must enable the plugin to access "preferences".

leamas, 2020-04-19:
The manual was broken but has been updated. Please reset stuff using 'flatpak list' and 'flatpak uninstall' and then try the new instructions.

At a glance, I see nothing wrong w r t plugins in the logfile. As @rgleason says, you need to enable them in preferences.

It should be possible to add new .desktop files to start different configurations....

hreuver0183, 2020-04-19:
bdbcat...
I install and enable the oesenc plugin then I get a dialog like below.
[image]
(The dialog is not quite as I expected it to be. As far as I can see oesENC seems not to be properly installed.)

After closing the options dialog I close OpenCPN and restart Open CPN and I get the second dialog.
[image]
(From which I conclude that oesENC was never installed in the first place.)

leamas, 2020-04-19:
hm... I think this is about the shared configuration directory which means that your regular installations and the flatpak one collides.

Furthermore, just clearing ~/.opencpn also creates a unstable state with plugins still installed but without metadata.   

I have been able to reproduce problems similar to yours, and also to get rid of them after manually uninstalling oesenc and clearing ~/.opencpn. This is certainly not to say problem is solved...

We should try a patch which uses a separate  config directory for the flatpak instance.

@hreuver0183: are you able to apply a patch, rebuild and test?

hreuver0183, 2020-04-19:
I think I should be able to.
The only caveat here will probably be where to apply the patch. You will have to help me there since you know flatpak better than me.

rgleason, 2020-04-19:
If possible, I would like to see the following icons used, provided it is early enough not to confuse people:

- Managed Plugin = Checkbox with check 
- Unmanaged Managed Plugin = Checkbox without check.
- Unmanaged Plugin = Round Circle  with hole.

Alternatively,
- Managed Plugin = Round Circle (solid black) 
- Unmanaged Managed Plugin = Round Circle (white center)
- Unmanaged Plugin = Square Checkbox without check

Or whatever set of icons make the most sense.

leamas, 2020-04-19:
> he only caveat here will probably be where to apply the patch.

You mean where to locate the directory, right? It's more or less standard, set by $XDG_CONFIG_DIR in environment. Stay tuned for patch, need to test myself...

leamas, 2020-04-19:
Attaching a patch. To install it you need to first make sure that  the flatpak-builder package  is installed (i. e., that the command flatpak-builder is available).  Then:

  - Uninstall current installation:` flatpak uninstall --user org.opencpn.OpenCPN`
  - Download and uncompress patch.
  - Apply patch: `git am 0001-flatpak-Use-separate-configuration-dir-1852.patch`
  - Rebuild: `cd flatpak; make build` First time, this involves rebuilding wxWidgets  so it takes a while... The build will install org.opencpn.OpenCPN locally.
  - Start flatpak to make sure it's sound.

Then the hairy part: synchronizing state. This is what I did:

  - Remove all flatpak plugins libs: ` rm ~/.var/app/org.opencpn.OpenCPN/lib/opencpn/lib*`
  - Reinstall for example oesenc using the plugin manager. 
  - This creates a manifest., use it to remove all files, like` rm -f $(cat ~/.var/app/org.opencpn.OpenCPN/config/.opencpn/plugins/install_data/oesenc.files)`

Next time, the oesenc installation should work as expected.

Would appreciate if you could report back

[0001-flatpak-Use-separate-configuration-dir-1852.patch.gz](https://github.com/OpenCPN/OpenCPN/files/4499753/0001-flatpak-Use-separate-configuration-dir-1852.patch.gz)

hreuver0183, 2020-04-19:
OK, with some minor issues the building has started (indeed first wxWidgets), I will folowup later.

hreuver0183, 2020-04-19:
I checked if the patch was applied in OpenCPN/src (OK)
I checked if the binary in OpenCPN/flatpak/app/files/bin/opencpn was the same as under .local/share (OK)
However I could not find GTK_CONFIG in the binary and OpenCPN does not use $XDG_CONFIG_HOME for the .opencpn location.

As for the oeSENC-plugin, even after removing the plugin from the cache and reloading no success.
I tried the OCPN-draw-plugin which worked accordingly.
I tried downloading GSHHS (full) without success either.

I hoped it was a caching issue but...

leamas, 2020-04-19:
hm... did you apply the procedures  I mentioned to sycnhronize the state  in https://github.com/OpenCPN/OpenCPN/issues/1852#issuecomment-616180899 ?

leamas, 2020-04-19:
BTW: XDG_CONFIG_HOME is only defined inside the sandbox where org.o.O runs, it's not visible in a regular shell.

hreuver0183, 2020-04-19:
Mostly.

I used rm -rf ~/.var/app/org.opencpn.OpenCPN/ instead of the first rule of the hairy part and rm -rf .opencpn for the last part.

I understand that the XDG_CONFIG_HOME should be configured somewhere in /etc/xdg (currently it is not set). For testing I used an export command.

The other commands I have used almost literally (see txt file at the end).
When in doubt I can try the whole shebang again tomorrow with script running (for a literal transcript and results typing mistakes) but I don't think there will be another conclusion.

I did found one other thing:
- ocpn-draw works
- oesenc, weatherrouting and vdr fail in the same way 
My system seems to somehow fail to work with the way the last 3 are packaged.

[image]
[image]
[image]

[test-flatpak.opencpn.txt](https://github.com/OpenCPN/OpenCPN/files/4499977/test-flatpak.opencpn.txt)

leamas, 2020-04-19:
You really need to try to follow my commands. In particular,` rm ~/.opencpn` doesn't affect anything, the patch makes opencpn use `~/.var/app/org.opencpn.OpenCPN/config/.opencpn` instead.

leamas, 2020-04-19:
@bdbcat: 

Here is indeed some kind of installation problem, sorry for detour. Look in the attached file, excerpt from the stdout logging messages (which needs to be fixed!). Many files are just not installed.

My gut feeling is that this is a recent change. Guts are not that reliable, though.

@hreuver0183 : stay tuned...

[paths.txt.gz](https://github.com/OpenCPN/OpenCPN/files/4500051/paths.txt.gz)

leamas, 2020-04-19:
@hreuver0183: Here is yet another patch. It solves the problems for me. Same for you?
[0001-PluginHandler-flatpak-Handle-paths-with-a-.-prefix.patch.gz](https://github.com/OpenCPN/OpenCPN/files/4500122/0001-PluginHandler-flatpak-Handle-paths-with-a-.-prefix.patch.gz)

leamas, 2020-04-19:
@bdbcat: Seems to be about a ./ prefix to the paths.

bdbcat, 2020-04-19:
@leamas...
Yes, that is the trouble.  It appeared on your latest patch commonizing tarball creation.  Now the tarball extraction starts with "./"
I fixed it in github master for linux installs, but not yet for flatpak.
I'll do that now....
Wait, now I see you have done it in b24dc71456560199662507257f02382574d0568f
So, all well now?

bdbcat, 2020-04-19:
@rgleason 
"The idea is to have two distinctly different icons for "no catalog Managed Plugins and "Unmanaged Plugins"

There really is no difference, at run-time, between the two.  So this distinction is hard to detect.
Anyway, "no catalog Managed Plugins" are rare beasts, encountered in testing or system crash recovery only, right?

hreuver0183, 2020-04-20:
I will check the new patch tonight, since that involves the basic system.

I think I will ignore the not working plugins for now, since that might be something typical for my system.

leamas, 2020-04-20:
> The idea is to have two distinctly different icons [...]

...IMHO actually have some merit. But it definitely doesn't belong to this bug. Further discussion on this belongs to a (yet not filed) separate bug.

leamas, 2020-04-20:
@bdbcat:
> I fixed it in github master for linux installs, but not yet for flatpak. [...]
> Wait, now I see you have done it in

Is there any reason to not apply the same patch patch for windows and macOS, just in case and to be consistent?

leamas, 2020-04-20:
@hreuver0183 :
> I will check the new patch tonight, since that involves the basic system.

You can just reinstall, the patches (both) are applied in ocpn-beta

bdbcat, 2020-04-20:
@leamas ..
Checked and done for Win and Mac.
We need to continue to support the old style of tarball, for rPI builds not covered by your patch.  That is checked and OK now.

bdbcat, 2020-04-20:
@leamas 
Have the upstream OCPN patches made it into your flatpak repo yet?
I try "$flatpak update", and I don't seem to get the patched version,yet.
Is this the correct procedure?

leamas, 2020-04-20:
The nigthly build was failed,  not related to the code. I have restarted the build at https://app.circleci.com/pipelines/github/leamas/OpenCPN/700/workflows/725b0c90-ad28-4682-aed0-ae1216983c72/jobs/1464

I'm actually unsure about the update procedure, need to check (have we versions which makes update meaningful?). In the meantime, a uninstall/install makes it.

leamas, 2020-04-20:
The "old style" is really as documented. It's OK that we make a best effort to handle things not according to the spec, but in the long term  it might be better to get rid of the leading ./ , it shouldn't be there.

I understand I'm to blame...  ;)

bdbcat, 2020-04-20:
@leamas 
Lots of google juice on this topic. Many people run into this issue unexpectedly...

leamas, 2020-04-20:
@bdbcat: The flatpak fix I did was actually broken. Should be fixed by #1855. But let's wait and check the flatpak ci build before merging.

I'll try to rebuild ocpn-beta as soon as it is merged (unless I'm sleeping).

leamas, 2020-04-20:
hm... there is no flatpak ci build, right? Well, I have built it myself, and all changes are guarded by `#ifdef FLATPAK`. So I guess there is not much to wait for.

bdbcat, 2020-04-21:
So, how to test?
It is 00:53 GMT. Did the cron-bot run?

I uninstalled, and installed ocpn-beta.
No change in the behavior.  Private data dir was not changed....

leamas, 2020-04-21:
>It t is 00:53 GMT

... and I'm in bed.

> Did the cron-bot run?

It did, later: https://app.circleci.com/pipelines/github/leamas/OpenCPN/703/workflows/f4d800c0-2ef1-4e7b-bb98-900bb248ab78/jobs/1465. 

It is scheduled at 04.00 CET (GMT + 1)

Submitted #1857 so we can inspect flatpak version in runtime

hreuver0183, 2020-04-22:
@leamas:
Monday evening I did not get a working opencpn in flatpak (got stuck on what seemed local problems).
Tonight I tried again and it worked.

The oeSENC package worked (with the USB dongle).
OCPN-draw also works.
Haven't tested all plugins though.

I saw the plugins install under  ~/.var/app/org.opencpn.OpenCPN/config/opencpn
The logfile I found there and I also put the layers and UserIcons thers (OK). 
Meanwhile I still found the files opencpn.conf and navobj.xml under ~/.opencpn

For now it works 100% better than last weekend: It looks useable!

leamas, 2020-04-24:
@hreuver0183 : Great!

Now, about this bug: should we try to narrow it down if there are issues left, or just close it?

hreuver0183, 2020-04-25:
The manual has been updated.

I think there are no specific issues left so closing is most obvious now.

closed 2020-04-25 by hreuver0183

leamas, 2020-04-25:
Thanks for reporting!

Yes, the manual is updated, but the realities has changed once again. See #1866

hreuver0183, 2020-04-25:
Since it is WiP such changes are bound to happen.
It will be a challenge to keep the manual up-to-date with the current version.

However, it is not related to issue #1852 anymore.

I asked rgleason if I could help with the flatpak part of the manual.

leamas, 2020-04-25:
>  asked rgleason if I could help with the flatpak part of the manual.

Much appreciated!

If you could to these docs, here are the details. The cli is (copied)

    $ flatpak remote-add --user  --if-not-exists flathub https://dl.flathub.org/repo/flathub.flatpakrepo
    $ flatpak --user install http://opencpn.duckdns.org/opencpn/opencpn.flatpakref

From now, here is also a GUI  method using Gnome Software:
  - Point the browser to http://opencpn.duckdns.org/opencpn/
  - Download the flatpak repository link
  - Open it using Gnome Software. This adds a new software repository.
  - Check that is is enabled using Software Repositories in menu  top-right
  - Search for opencpn and install it.