OpenCPN/OpenCPN #1856: Latest tracks visible on screen but not in track manager - Win10 not Android

philma9, 2020-04-20:
For the last few weeks my tracks are not being recorded in the track manager although they show on the chart (even after stopping restarting opencpn).

I'm on 5.0.0 on Win 10 (up to date).

I've done about 100 nm since 21 March (last recorded track) and have the tracks visible on screen, but they are not showing in the track manager. I've been using opencpn on windows for over 10 yrs, and on various android platforms for 5 years so I'm doubtful that it's user error...especially as the tracks are all ok in the track manager on my Samsung tablet.

Got to assume it's a win 10 problem. Opencpn runs 24 hrs with occasional stops and I have tried reloading earlier navobj.xml's without any joy.

I loaded the Win10 navobj.xml onto my Samsung tablet and everything is ok - I can see all tracks on the screen and in the track manager.

So, this would seem to narrow the problem possibly to a presentation issue on win10.

By the way I am using the paid version of Opencpn on galaxy tab platform and on my phone. 
navobj.xml attached.
[navobj.zip](https://github.com/OpenCPN/OpenCPN/files/4506229/navobj.zip)

bdbcat, 2020-04-21:
I have imported the navobj.xml, and I see the following (pix).
Seems normal to me.  What are you seeing?
[image]

transmitterdan, 2020-04-21:
Dave,

With OpenGL turned on as I zoom in and out the tracks appear and disappear.  There doesn't seem to be any particular rhyme or reason.  With OpenGL turned off all the tracks display all the time.

bdbcat, 2020-04-21:
Dan...
IIUC, the OP problem is:
"my tracks are not being recorded in the track manager although they show on the chart "

transmitterdan, 2020-04-21:
I started a new track and it appears at the top of the list of tracks in the route manager dialog page.

@philma9, how do you know new tracks are not added to the list of tracks? When a track is building it should show in the list of tracks as a line highlighted in bold and the total length gradually increasing. Can you tell us exactly what you do to confirm no new track is created in the list?

philma9, 2020-04-21:
Yes, the new track appears at the top of the list, correct, but it does not change from 0 distance which si why I recorded the issue -  nothing was showing for over 100 nm - and a few 'start new track' although the tracks were recorded on screen. Also. if you wait a few seconds and hit 'start track again' the most recent track is replaced by the latest.

This problem is easily replicated (at least on my laptop) by 'unshowing' the last half dozen tracks. They disappear from the screen but you should still see several tracks in the west of the sound for which there appears to be no corresponding data visible in the Track Manager. IF there was a date issue with the data (and I currently assume there is not) then it is possible that the track data could be somewhere else in the table. I have not checkd for that.

transmitterdan, 2020-04-21:
So far I can't duplicate the zero length track.  My track continues to grow in length over time.

I have seen an issue related to sorting tracks based on name.  If the name includes a time value then it sorts differently than a track name that is a simple date string.  The tracks with a time string sort further down and I have to scroll down to find them.  This appears to be a wxWidgets bug.

transmitterdan, 2020-04-22:
Here is what I have learned:

New tracks are created without a m_TrackNameString field populated (Track: unnamed). When the track list control table is created for display, it manufactures a name based on date and time to be displayed in the track name column.  However, the track sort function does not use this manufactured name. So all tracks without a name sort "equally" because the sort function uses an empty string. Thus there is a probability that any new track will sort well down in a long list of tracks and not be visible at the top of the track list.  So that might explain why a user cannot see a new track under construction unless they scroll down. I consider this as a minor bug but there are multiple possible solutions. The simplest and logical could be to use the manufactured name in sorting.

Recommendation 1: Give all the older tracks a name and then the new track under creation will sort at one end of the list (empty name will sort at one end of the list).

Recommendation 2: Be sure to keep a backup of navobj.xml because there is higher probability of corruption for huge files like this (55MB is huge).  When O shuts down it completely rewrites this file and if it is interrupted the file will likely be corrupted.

philma9, 2020-04-22:
Thanks Dan. Agree with your conclusion. The data for the tracks is there but near the bottom of the long list. I guess I should have checked that, but it's not the 'sort' of behaviour I would have expected. I'll read the manual and try to reduce the size of my navobj.xml anyway. I do keep backups of navobj as a matter if course. Thanks again for taking the trouble to investigate.

transmitterdan, 2020-04-27:
@philma9 
Can we close this issue?

closed 2020-05-04 by bdbcat