OpenCPN/OpenCPN #1859: radar_pi master wont find requested URL returned error: 404 Not Found]

Hakansv, 2020-04-22:
Dave.., Alec.. or Kees..
For the moment I haven't investigated a lot.
If it's O code or radar_pi upload failure I don't know. I think you have more distinct knowledge about the URL and where to look first?

Håkan

```
09:00:30.676 MESSAGE pluginmanager.cpp:665 Installing managed plugin: Radar
09:00:30.791 MESSAGE Downloader.cpp:132 filesize https://dl.cloudsmith.io/public/kees-verruijt/ocpn-plugins-stable/raw/names/radar-msvc-10.0.14393-tarball/versions/v5.1.3/radar_pi-5.1.3.0_msvc-10.0.14393.tar.gz: 0 bytes

09:00:31.373 WARNING Downloader.cpp:84 Failed to get 'https://dl.cloudsmith.io/public/kees-verruijt/ocpn-plugins-stable/raw/names/radar-msvc-10.0.14393-tarball/versions/v5.1.3/radar_pi-5.1.3.0_msvc-10.0.14393.tar.gz' [The requested URL returned error: 404 Not Found]

09:00:36.565 MESSAGE PluginHandler.cpp:727 Cleaning up failed install of Radar
```

keesverruijt, 2020-04-22:
Yeah, I think this is my problem. I made a release and bungled some of the URLs, then moved them in CloudSmith. Note to self: not a good idea. Better to just release a new build (even if that takes 4 hours...)

douwefokkema, 2020-04-22:
Under windows the managed version is still at 5.0.4 beta 6. 
[image]

douwefokkema, 2020-04-22:
I am using O 5.0.522 (few days old) and attached the opencpn.log.
[opencpn.log.txt](https://github.com/OpenCPN/OpenCPN/files/4516184/opencpn.log.txt)

keesverruijt, 2020-04-22:
I didn't even know that O 5.0 was getting the managed plugins.

keesverruijt, 2020-04-22:
@douwefokkema that's just a build you made yourself right?
Which branch did you use?

douwefokkema, 2020-04-22:
The OCPN I am using was compiled from master 3 days ago. Pressing the update to managed version indeed updates to radar_pi 5.0.4-beta6. With oeSENC the update to 4.0 does not work, URL not found.

I used OpenCPN master branch.

bdbcat, 2020-04-22:
"With oeSENC the update to 4.0 does not work, URL not found."
What platform, please?

douwefokkema, 2020-04-22:
Sorry Works fine, I had to update plugin catalog (was at beta) to master. Now the managed plugin is radar_pi 5.1.3

douwefokkema, 2020-04-22:
That is to say, it sees version 5.1.3 but can not download as the URL is not found

douwefokkema, 2020-04-22:
[image]

keesverruijt, 2020-04-22:
I just created a PR with new updated 5.1.3 versions, clean build, once Dave pulls that in it should work again.

douwefokkema, 2020-04-22:
> 
> 
> "With oeSENC the update to 4.0 does not work, URL not found."
> What platform, please?

Dave, this is on Windows 10.

Hakansv, 2020-04-22:
Been out sailing , now buoyed.
I'm on win10. today's O master. oeSENC and OCPN_Draw works fine here.
Kees. Thanks I'll make new test once it's merged.
[image]

douwefokkema, 2020-04-22:
> 
> 
> > "With oeSENC the update to 4.0 does not work, URL not found."
> > What platform, please?
> 
> Dave, this is on Windows 10.

After update of plugin catalog oeSENC indeed updates fine to 4.2.7

bdbcat, 2020-04-22:
"I just created a PR with new updated 5.1.3 versions, clean build, once Dave pulls that in it should work again."

Done.
Thanks

Hakansv, 2020-04-23:
Managed radar_pi 5.1.3 now installs as it should. 
Thanks

closed 2020-04-23 by Hakansv