OpenCPN/OpenCPN #1860: beta PPA packages no longer want to install

keesverruijt, 2020-04-22:
`apt install opencpn` with the beta PPA selected gives:

    Unpacking opencpn (5.1.421.4-0~bionic1) ...
    dpkg: error processing archive /tmp/apt-dpkg-install-J3alKt/4-   opencpn_5.1.421.4-0~bionic1_amd64.deb (--unpack):
    trying to overwrite '/usr/share/opencpn/gshhs/wdb_borders_c.b', which is also in package opencpn-gshhs-crude 2.2.4-0~bionic1

bdbcat, 2020-04-22:
Kees...
The PPA is work-in-process right now.  If you need the latest master code, you can fetch the latest build artifacts from:
https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/

bdbcat, 2020-04-24:
Kees...
Ubuntu Beta PPA is now functional.
Thanks

closed 2020-04-24 by bdbcat