OpenCPN/OpenCPN #1864: MacOS plugin compatibility: too narrow defined.

leamas, 2020-04-23:
Current code in PluginHandler requires the plugin to be built with a version which is <= the main opencpn instance. However, discussions in #1839 has revealed that all MacOS plugins are compatible and could be used without respect to the os version.

The simple fix would be to relax the test in isCompatible() in PluginHandler.cpp. 

Other strategies could be that the plugin build chain declares the MacOS plugins to be lowest acceptable version -- perhaps cleaner, and could handle possible future ABI breaks better.

bdbcat, 2020-04-28:
I think the simple solution suffices for now.
If we have an ABI change with some new MacOS version, it will require a new OCPN base version for sure.  Probably manifested as a new wxWidgets build in the bundle.  And since this will require new plugins, too, then we can tighten up the requirement then.
So we shall "kick the can down the road" a bit more.

leamas, 2020-04-28:
Fixed in [9f14c0e927d].

Note that an ABI break  will fail:
  - Assume that v 6.0 introduces an ABI break for MacOS plugins
  - We will now have two set of plugins compiled for pre-6.0 and 6.0
  - The new version will be aware of this and only load correct 6.0 plugins
  - Old installations will load all plugins, even 6.0 which they can't handle.

closing

closed 2020-04-28 by leamas