OpenCPN/OpenCPN #1865: Plugin Downloader: deployment issues.

leamas, 2020-04-23:
The new plugin downloader in #1839 is ready for further testing. 

The deployment consists of two parts, frontend and backend. We need to decide how these should be deployed in a longer perspective.

For now, the frontend runs on a amazon ec2 instance owned by me. From a deployment perspective this is handled by serving static webpages (apache) and by building the runtime files from sources (npm). Both of these are readily available on all platforms, although very old versions might be a problem

The backend is written in ruby and runs on @mauroc's squiddio server. I don't know the deployment perspective on this. The backend is not that complicated and could probably be rewritten in for example PHP or python if this should easy the deployment.

rgleason, 2020-09-03:
"..Ready for further testing at http://opencpn.duckdns.org/ocpn-download. "
Not critical, but this link is a 404 Error.

leamas, 2020-09-04:
I know... Seems no point spending time on this until we have a deployment strategy.

leamas, 2021-01-30:
For now, available at http://mumin.crabdance.com/opencpn-dl/ and  http://gafsan.crabdance.com/opencpn-dl/

rgleason, 2021-01-30:
Maybe this should be migrated to opencpn.org   Plugin Downloads page?

leamas, 2021-08-02:
@bdbcat: This is not critical infrastructure, and perhaps we could close this issue after adding something like below to the plugins download page at https://opencpn.org/OpenCPN/info/downloadplugins.html

---

### Plugin tarballs downloader

The built-in plugin system ion OpenCPN downloads plugins in a special format called tarballs. Such tarballs can also be downloaded without using OpenCPN using the plugin downloader available [here]( http://mumin.crabdance.com/opencpn-dl/). An identical backup instance is available [here]( http://gafsan.crabdance.com/opencpn-dl/).

Once downloaded, such tarballs can be transferred to a computer running OpenCPN and then imported using the Import button available in the Options | Plugins tab [(Screenshot)](https://user-images.githubusercontent.com/405541/127825288-8ad8ab1f-6d7e-49e1-bf91-f6fb2e43ef79.png)

bdbcat, 2021-08-02:
Not sure why we even need this anymore.
What is the use-case?

rgleason, 2021-08-02:
I think it could help users to become testers. 

It also puts access to current plugins in the website "downloads" directory,  which will be a good substitute for the links K has for each plugin, and in addition this online db would eliminate any chores in updating the links.

I think it is pretty clever and would not take away from the normal Plugin Manager process (PIM), just augment it.

rgleason, 2021-08-02:
What is the use-case?
say a sailor is going to sea and wants updated tarballs for all his plugins. This is definitely the way I would suggest doing it! Rather than depending on tarballs in the cache!

Easiest way to do it.

leamas, 2021-08-02:
The design use-case is the sailor in a remote harbour without  network access and with a fixed computer (say, an rPI). He could then take a laptop (possibly a phone), go to some bar in the vicinity with network access, download plugins and bring them back to boat.

I also see Rick's "preload" use-case as a valid-one.

If one wants to refresh the memory there is #1839.

That said, I could live with a decision that it doesn't fill any purpose -- I can actually see the arguments also for this. But, IMHO we should now close this discussion and make a decision rather then just having #1865 lingering.

rgleason, 2021-08-02:
This will save K time updating links on the downloads page too.

bdbcat, 2021-08-03:
Thanks for the feedback.
Web site updated.
Ready to close this issue, and friends?

leamas, 2021-08-03:
OK. and thanks. Time to close.

closed 2021-08-03 by leamas