OpenCPN/OpenCPN #1870: macOS , object query window covers complete canvas

bcn2, 2020-04-27:
and will not remember a corrected size
This is around since 5.0
[image]

CarCode, 2020-04-27:
Resize the window, click OK at the window and then try again.
It works here with OCPN 5.0.1 and macOS Mojave/Catalina as always, no problems.

bcn2, 2020-04-27:
That's exactly what I tried. OCPN from today, macOS 10.15.4 Catalina.

CarCode, 2020-04-27:
If you are unable to resize the window by mouse dragging open opencpn.ini with TextEdit, search for S57QueryDialogSizeX and S57QueryDialogSizeY lines and change the values.
BTW there is no official OCPN from today. Use version 5.0.1.

bcn2, 2020-04-27:
Well, the parameters are getting written into the config after resizing with "ok", but are happily ignored after then on.
Resulting size is always filling the complete canvas but some margin on all sides
[image]

S57QueryDialogSizeX=133
S57QueryDialogSizeY=556

[image]

The version used is the one from the daily builds on Cloudsmith
OpenCPN_5.0.522-+1.95c9c41.pkg

CarCode, 2020-04-27:
I have checked this version now although it not even a beta but something as a delta or gamma version. It works here as always before and keeps its window sizes. So no problem for this.
But perhaps you take the wrong version?
Do not use OpenCPN_5.0.522-+1.95c9c41.pkg but OpenCPN_5.0.522-+1.95c9c41.dmg !!!
Nevertheless these versions have severe bugs. Look i.e. the contents of the folder Contents/Frameworks there are 2 different libpixman.dylibs and all libwx.dylibs have doubles with aliases. Never seen such a bad progamming for macOS. Terrible.

bcn2, 2020-04-27:
Gerhard, the pkg is the long claimed macOS installer version. No need to drag the app into the application folder...
Functional identical with the dmg

bdbcat, 2020-04-27:
@bcn2...
I think maybe you have "Settings->UserInterface->Enable scaled graphics interface" set?
This setting should default "ON" only for Android.

bcn2, 2020-04-27:
Correct, that's the trigger for the behavior. Interesting side effect...

closed 2020-04-27 by bcn2