OpenCPN/OpenCPN #1876: Question: wxWidgets version used in current builds

rgleason, 2020-04-28:
Windows: wxWidgets 3.1.2
Mingw: 3.1.2
MacOS:
Ubuntu generally, bionic, bionic-gtk3, xenial, buster:
Ubuntu Trusty:
Raspbian armhf stretch:
Raspbian armhf buster
Fedora
Flatpak

Thanks

jongough, 2020-05-02:
Hi All,
   I have just fallen through an issue with multiple versions of wxWidgets. In my development environment I have wxWidgets 3.0.4 & 3.1.2 installed. Whilst building I have been setting the forced version to 3.0.4 and that has been working fine whilst building OCPN and plugins locally as well as using Plugin Manager to install plugins from cloudsmith.

   I had a problem in one of the plugins and had to delete the build directory and all associated cmake files in it, effectively starting from scratch with the plugin. I forgot to force wxWidgets to use the 3.0 version so the plugin got built with the 3.1 version. I copied the 'lib' file (I am working on Linux) over the top of the one installed by Plugin Manger so that I could use debugging to sort out a non-related issue. I started to get crashes in parts of the code which have not changed for sometime. Even on my environment it took a few hours to figure out that it was the version of wxWidgets that was causing this. I was comparing the output of the circleci build to my build to see what was different. Changing the version of wxWidgets to 3.0.4 fixed the issue.

   Now, as I have been working on the plugin build process to try and make is 'easy' for developers I have noticed that the build environments subtly change over time (I think the macOS is the worst currently). It is quite possible that in the near future the version of wxWidgets could be changed and this would make the plugins incompatible on some environments and hence cause a crash. Perhaps we need to have the version of wxWidgets included in the metadata so that plugin manager can tell that they is a potential issue and not allow the plugin to be installed or run.

   I don't want to be debugging plugin issues with incompatible plugin software on some platforms for a user as it will be difficult to work out what is really going on.

rgleason, 2020-05-02:
> I don't want to be debugging plugin issues with incompatible plugin software on some platforms for a user as it will be difficult to work out what is really going on.

So actually setting the wxWidgets version would help?

jongough, 2020-05-02:
It 'may', but if OCPN is not generated on the same platforms at the same time things may change. You have seen how wxWidgets varies across environments, so, for instance, I am not sure we could bring the 'Trusty' environment up to a newer version, which is why I am raising the issue to get it solved before we have to deal with real crashes. We may be able to fix all environments at 3.0.4, but I am not sure. There are new capabilities being introduced with each version or widgets and care would have to be taken not to use these, or code guards around this.

bdbcat, 2020-05-03:
jon...
On linux, we do this already, after a fashion.  The pluginmanager will not load a plugin that uses a different version of wxWidgets than the core was built with.  See pluginmanager.cpp:2030.
Did this test not work in your recent experience?

On Mac and Windows, we bundle the wx libraries with the app, since those platforms have no useful package manager.  It should be sufficient to require plugin devs to use the same wx version.

I have no problem with adding a field to the metadata.  But I do not plan to parse that field in OCPN 5.2.  We gotta stop building and start shipping....

jongough, 2020-05-03:
Dave,
   So if it is checking how come it loaded and ran, until the crash, a plugin built with wxWidgets 3.1.2 on an OCPN built with wxWidgets 3.0.4? Surely this should have been picked up by the 'special' processing. I just ran the debug process on the code and it does not pick up the fact that the plugin is wx 3.1.2 whilst OCPN is wx 3.0.4 .

I have generated my testplugin using wx 3.1.2 and it is available to install via plugin manager using the 'Custom' install url:
```
https://dl.cloudsmith.io/public/jon-gough/testplugin_pi-alpha/raw/names/testplugin_pi-ubuntu-18.04-bionic-metadata/versions/1.0.82.2+2263.34ea4d5/testplugin_pi-1.0.82.2_ubuntu-18.04-bionic.xml
```
If you use this it will cause a sigsev (segmentation fault). If I generate the code using wx 3.1.3 it gets correctly rejected. Here is the log message:
```
16:45:42.885 MESSAGE update_mgr.cpp:194 Selected update: testplugin
16:45:42.887 MESSAGE pluginmanager.cpp:437 Installing testplugin
16:45:42.891 MESSAGE PluginHandler.cpp:965 Installing testplugin_pi-1.0.82.1_ubuntu-18.04-bionic.tar.gz from local cache
16:45:42.895 MESSAGE PluginHandler.cpp:552 archive finish write error: Can't restore time
16:45:42.895 MESSAGE PluginHandler.cpp:552 archive finish write error: Can't restore time
16:45:45.565 MESSAGE pluginmanager.cpp:1018 PlugInManager: loading plugins from /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib
16:45:45.566 MESSAGE pluginmanager.cpp:971 Using LD_LIBRARY_PATH: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib:/home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib:/home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib
16:45:45.566 MESSAGE pluginmanager.cpp:1007 Using PATH: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib:/home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib:/home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib:/home/jon/development/esp/esp-idf/components/esptool_py/esptool:/home/jon/development/esp/esp-idf/components/espcoredump:/home/jon/development/esp/esp-idf/components/partition_table/:/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/home/jon/.espressif/tools/xtensa-esp32s2-elf/esp-2019r2-8.2.0/xtensa-esp32s2-elf/bin:/home/jon/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/home/jon/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin:/home/jon/.espressif/python_env/idf4.1_py2.7_env/bin:/home/jon/development/esp/esp-idf/tools:/home/jon/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
16:45:45.566 MESSAGE pluginmanager.cpp:1024 Loading plugins from dir: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib
16:45:45.566 MESSAGE pluginmanager.cpp:1050 PlugInManager searching for PlugIns in location /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib
16:45:45.570 MESSAGE pluginmanager.cpp:1087 Found 3 candidates
16:45:45.570 MESSAGE pluginmanager.cpp:1094 Checking plugin candidate: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib/libsquiddio_pi.so
16:45:45.570 MESSAGE pluginmanager.cpp:1094 Checking plugin candidate: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib/libocpn_draw_pi.so
16:45:45.570 MESSAGE pluginmanager.cpp:1094 Checking plugin candidate: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib/libtestplugin_pi.so
16:45:45.577 MESSAGE pluginmanager.cpp:2059 Plugin is compatible: true
16:45:45.577 MESSAGE pluginmanager.cpp:2122 PlugInManager: Loading PlugIn: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib/libtestplugin_pi.so
16:45:45.622   ERROR dlunix.cpp:162 libwx_baseu_unofficial-3.1.so.3: cannot open shared object file: No such file or directory
16:45:45.622 MESSAGE pluginmanager.cpp:2181    PlugInManager: Cannot load library: /home/jon/opencpn/OpenCPN-5.0/build-kde181/plugins/lib/libtestplugin_pi.so 
```

jongough, 2020-05-03:
I think we need to start testing this beta to see what can or will go wrong for users. I am not sure much edge testing has been done and some of it will be very difficult to diagnose if users get the issue.

rgleason, 2020-05-03:
Jon, I was just reviewing some early .travis scripts and I believe I saw some MacOS scripts that downloaded wxWidgets directly from the navnux website.  Perhaps this is done where ever possible to limit the issue?  
For example look around in 
https://github.com/rgleason/weatherfax_pi/commit/c4132b584700838b631bca7abc6783691591ea0b

Found while investigating about portaudio and rtlsdr for weatherfax for mingw and macos

CarCode, 2020-05-03:
Some historical background is needed for understanding:
Things are going worse when a bug in wxWidgets and/or a new version of macOS suddenly results in black screens of OpenCPN when scrolling the window. Instead of using a fixed wxWidgets version David Register makes a complete desaster and introduced his own patched wxWidgets version. rgleason has pointed it out. (see above)
But longtime before that macOS does not compile OpenCPN with the old wxWidgets versions. Faulty usage of wxWidgets functions such as e.g. the attributs of Sizers seem to compile fine for Windows and Linux but not for macOS which is more restrictive. Obviously there have been reasons to switch to newer wxWidgets for Windows also. Maybe the Windows compiler got smarter now. But I guess most Linux versions comes with a buildin older wxWidgets version and nobody knows how to bundle the newer libraries with OpenCPN for Linux. So there is that mismatch now.

rgleason, 2020-05-03:
Thanks Carcode for your history.  We all do the best we can in a practical way. I guess this is a constant work in progress, that may get better in time.  With better installation tools, the market may focus on certain basic platforms, reducing the env and simplifying the issue Jon has raised.  Perhaps we keep what he has pointed out in mind when "things happen" so that we know the possible cause?

bdbcat, 2020-05-04:
Jon...
After all, it is a simple bug in OCPN, or maybe could be called a deficiency.
Your wxWidgets library for wx3.1 is called something like:
"libwx_baseu_unofficial-3.1.so.2"
That non-standard naming fooled the compatibility checker.  I daresay that this kind of problem would not be encountered very frequently in the wild...
Anyway, the check has been hardened in the current github master.  Your testplugin_pi build against wx3.1 will not load, and a message is reported to the user.

jongough, 2020-05-04:
I have just tried it with OCPN master (4eeeabb86) and with the plugin installed. Startup OCPN then goto Options. When Plugins tab is selected a message warning about the incompatibility is shown on the screen.
[image]
Select the plugin, it is greyed out at this point, and you get asked if you want to downgrade the plugin:
[image]
Click downgrade and you get told the plugin was successfully installed from the cache:
[image]
Click OK and the plugin now shows as available with the enable tick box presented:
[image]

Select checkbox and the plugin is shown as enabled:
[image]
Note that the 'Preferences' button is now available. Clicking this will have no effect.

So the plugin will not crash OCPN which is good. But from a user perspective they have changed the plugin (the downgrade), but it still does not work and there have been no messages about that. There is also no entry in the log to say that there was a problem with the plugin except for the instance the message box about compatibility was displayed to the user.

bdbcat, 2020-05-04:
I'm pretty sure there is a log message, something like:

MESSAGE pluginmanager.cpp:2059 Plugin is compatible: false

bdbcat, 2020-05-04:
I agree the offer to "Downgrade" to the same version is non-sense.  I'll look at that

jongough, 2020-05-04:
There is a message in the first instance, i.e. when the plugin is first going to be loaded, but after that you get taken back to the screen and offered the downgrade which appears to work, i.e. there is no message warning you again that you have an incompatible plugin. For the user this will be confusing, especially as the plugin is loaded from the cache and no error or warning is displayed and the plugin appears to offer the ability to enable it.

Hakansv, 2020-05-04:
Something connected here:  #1888

rgleason, 2020-05-05:
Let me know if we should close this now, please.

jongough, 2020-05-05:
I think there are still issues that need to be addressed when a plugin does not work. In this case it is still related to wxWidgets. A more general issue was raised: [https://github.com/OpenCPN/OpenCPN/issues/1893#issue-611521446](https://github.com/OpenCPN/OpenCPN/issues/1893#issue-611521446) but I am not sure it is being addressed.

bdbcat, 2020-05-05:
This particular issue, related to wxWidgets compatibility, is still open.  Screenshots, etc will be useful.

bdbcat, 2020-05-05:
The part of #1893 that addresses wxWidgets compatibility is a duplicate of this one, mostly.

jongough, 2020-05-05:
The screen shots are shown here: https://github.com/OpenCPN/OpenCPN/issues/1876#issuecomment-623246574

bdbcat, 2020-05-05:
Yes, thanks.  That's why I don't want to close it yet.

bdbcat, 2020-05-07:
Ready to close?

rgleason, 2020-05-07:
Jon G. Ok to close?

jongough, 2020-05-07:
OK to close as cannot install a plugin with incompatible versions of wxWidgets. Some of the messaging is not correct but the system does not crash.

rgleason, 2020-05-07:
Thanks bdbcat and JG!

bdbcat, 2020-05-07:
" Some of the messaging is not correct "

How so?

jongough, 2020-05-08:
[image]
Click instal lnothing shows up for ~10+secs you get
[image]
Click OK and you get
[image]
Notice OD is still running but in the plugins panel it is greyed out
Click on the greyed out OD and you get

jongough, 2020-05-08:
Further, add a custom URL and update the catalog to get a new version. The 'Update' for the button for the plugin shows, but clicking this gives 'install' rather than update on the dialog box. Cannot show this as the message box is over the top of the Plugins tab.
The plugin installs OK, but the plugin is greyed out. Click on it and it offers to install it again. Click on install, wait with nothing showing for ~10sec and get told it has been installed from the cache. Enable the plugin and it looks OK in the options panel, but the 'Preferences' button whilst showing does not work. The website button does.

jongough, 2020-05-08:
Selecting reinstall of Weather Routing The following is displayed:
[image]
Select 'more' and here is what is displayed:
[image]
Perhaps this should be a new issue

bdbcat, 2020-05-09:
Jon...
Leaving the custom URL comment aside, for a moment...
I cannot reproduce your screenshots on 18.04.  I do not know what your starting point is.  I do not know which catalog you are using.
But for sure, something is funny. Starting with:
1.  When you install a plugin from the "packaged" (i.e. gift box) icon state, that gift box should disappear.  And the plugin should then be shown as a normal plugin panel.
I do not see this happening in your screenshots or description.

jongough, 2020-05-09:
I am rather constrained at the moment to the versions available through Plugin Manager, so I am using the custom URL to be able to test different versions of the same plugin easily. Otherwise it will require updates to the master or beta lists for each test.

I just got this on Ubuntu 18.04 using the latest, I think, version of OCPN from your cloudsmith:
[image]
[image]
Notice the dialog does not resize.
Unfortunately Weather routing does not appear to be in the latest catalog so the plugin is now marked as not being under Plugin Manager control.
[image]
Updating the catalog using the 'custom' tab with:
```
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-prod/raw/names/weather_routing_pi-ubuntu-18.04-bionic-metadata/versions/v1.13.7.1/weather_routing_pi-1.13.7.1_ubuntu-18.04-bionic.xml
```
gives
[image]
Clicking update give:
[image]
clicking more gives:
[image]

bdbcat, 2020-05-09:
OK, I see that.
We need to refine the dialog layout, accommodating arbitrary text coming from the metadata.
I'll take a look at that.
Thanks

bdbcat, 2020-05-09:
Improvement in github now.
c07150bc9fa0c18a400edcd061e21e16ed8c740e

jongough, 2020-05-09:
The text is wrapped horizontally but the dialog is not changed vertically, so the 'Dismis' button is missing
[image]

rgleason, 2020-05-09:
Guys, in 1.13.5, Sean had these returns, I believe it was  "/n" or something like located to suit the previous Plugin Manager.  It was causing trouble in one of the releases and I noticed it, so I removed all the "/n" and made it just one continuous string of text.  

I am curious about this now. What is preferred for the descriptive text?  Just one continuous line with no returns?   Same for summary?

bdbcat, 2020-05-09:
Rick...
Preferred method is for the plugin author/sponsor to use "/n" (line feed) to layout the text as desired by the author.  Alternatively, if done as one continuous line, the code will split the line as necessary to fit the space.  But this can produce nonsense sometimes.
The current code will honor "/n" in the string.  Take a look at Radar_pi.  This one looks nice.

bdbcat, 2020-05-09:
Jon...
Scroll bar is present and works, dialog "Close" button works.
Enough....

Dave

closed 2020-05-09 by bdbcat

rgleason, 2020-05-09:
Thanks- will note that in cmake 
Preferred method is for the plugin author/sponsor to use "/n" (line feed) to layout the text

jongough, 2020-05-09:
This is about the user experience (UX in modern terms, HCI in ancient terms). The screen is quite big enough in most cases to display a few more lines, why put a scrollbar in and artificially constrain the dialog box? There has to be a maximum size, true, but most screens are bigger than 10 lines deep. 
Don't forget that the finish of the product matters. It doesn't matter how cleaver or complicated it is the way the user interacts and feels about the product is important. To borrow an old nautical saying "Don't spoil the ship for a Ha'p'orth of tar".

jongough, 2020-05-09:
> Thanks- will note that in cmake
> Preferred method is for the plugin author/sponsor to use "/n" (line feed) to layout the text

Except that this makes resizing the dialog problematic. For a plugin all the dialogs are under the control of the plugin, in this case it is under the control of OCPN, so putting "/n" in, unless to create a paragraph, makes no sense. The wrapping of words should be done on word boundary, i.e. space, and must be controlled by the owning process.

bdbcat, 2020-05-09:
Jon...
All about priorities.  Talk is cheap, but coding is not.  The simple changes you requested to resize the dialog took about 6 hours to complete satisfactorily for all platforms., since wxWidgets does not do this kind of thing well.
"The screen is quite big enough in most cases...".  And the other cases?  Again, requires coding and testing across our large platform matrix.
There are better, more generally critical, things to focus on.

bdbcat, 2020-05-09:
" so putting "/n" in, ..., makes no sense."
Unless you want to create a bullet list, like is done in Radar_pi.
This makes good sense to me.  The plugin author might like have some input into the way in which his information is presented, rather than have it be displayed by inscrutable rules.