OpenCPN/OpenCPN #1877: Plugins: Android Build

rgleason, 2020-04-28:
Sean asked about a script for android.

bdbcat, 2020-04-29:
Do you (or Sean) mean in the context of O5.2 managed plugins?  If so, nothing done on this just yet.  Soon on my radar screen.
I field lots of Android requests for Sean's plugins, which are now not working due to the Google requirement for 64bit packaging. But that is another topic....

rgleason, 2020-04-29:
Yes.  You're very busy, perhaps this needs to be done by others. Is it as simple as a ci/ script?
Somehow I think not, as adjustments will be needed on the plugin's code side too.
Perhaps Sean's plugins are best set up for an additional script.

bdbcat, 2020-04-29:
When I get to it, I will try to package a simple plugin (like VDR) as a managed variety for Android.  This code/script can become the "poster child" for adaptation of other plugins.
The main challenge will be probably finding a CI solution that understands Android...

rgleason, 2020-04-29:
I will look into this and post here some links which might help.

rgleason, 2020-04-29:
Looks like it is possible:
https://circleci.com/build-environments/linux/android/
https://circleci.com/docs/2.0/language-android/
https://circleci.com/orbs/registry/orb/circleci/android
https://medium.com/open-knowledge/android-continuous-integration-circleci-f4cb87a234f7

rgleason, 2020-04-29:
Dave, made you a collaborator https://github.com/rgleason/vdr_pi/tree/frontend1
Thanks.

bdbcat, 2020-04-29:
Rick...
Thanks, and thanks for the research.
But don't hold your breath. First priority for me is managing the Beta test.

rgleason, 2020-04-29:
I totally understand.

rgleason, 2020-10-17:
Android builds are implemented in frontend1 and frontend2.  so closing

closed 2020-10-17 by rgleason