OpenCPN/OpenCPN #1878: Plugin Manager: Installation of plugins on macOS Catalina puts plugin in wrong location

jongough, 2020-04-29:
Installed OpenCPN OpenCPN_5.1.428+1.e63cdd2.dmg (downloaded from cloudsmith) on macOS. It installed OK and ran OK. Trying to install new plugins using the Plugin Manager succeeds and the plugin shows up in the list of plugins. But enabling the plugin does nothing. The plugin is intalled into `./Library/Application Support/OpenCPN/Contents/unknown-paths`, this can be seen in the log
[macos opencpn log showing not finding new plugins.txt](https://github.com/OpenCPN/OpenCPN/files/4549939/macos.opencpn.log.showing.not.finding.new.plugins.txt).

rgleason, 2020-04-30:
Jon, your commits to testplugin_pi  fix_up are building MacOS and the deployed tarball appears to be configured properly.  I think this is fixed, but perhaps a Mac User could try loading this using "Import Plugin" to see if it works.
https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-alpha/raw/names/testplugin_pi-1.0.80-darwin-10.13.6-macos-tarball/versions/1.0.80.0+590.96fd9d4/testplugin_pi-1.0.80.0_darwin-10.13.6-macos.tar.gz

bdbcat, 2020-04-30:
If the plugin CI build has been corrected, this calls for a PR to the plugin master catalog.

jongough, 2020-04-30:
The installation process is now working and the master catalog will be updated soon.

closed 2020-04-30 by jongough

rgleason, 2020-04-30:
It is not corrected for weather_routing, weatherfax, watchdog or squiddio.
I am getting a permissions problem on all of them.
I have installed the changes from testplugin_pi for fixing the MacOS build.

1. .circleci/config.yml
2. cmake/in-files/cloudsmith-upload.sh.in
3. cmake/PluginLocalization.cmake
4. cmake/PluginInstall.cmake
5. cmake/PluginConfigure.cmake
6. ci/circleci-build-macos.sh
7. ci/control
8. buildosx//InstallOSX/plugin.pkgproj.in
9. Small changes in CMakeLists.txt

Note: Sometimes I've copied these files over from testplugin twice to be sure. MacOS still fails.

Regarding the new MacOS build, which is generally the only build not working for these plugins.  The error is a "permissions" error for all of them, so I must be missing something critical somewhere. 

- Squiddio -  https://github.com/rgleason/squiddio_pi/tree/frontend2_fix_up_MacOS  (Will make a PR to Mauro when working)
- Weather_routing -  https://github.com/rgleason/weather_routing_pi/commits/frontend2
- Watchdog - https://github.com/rgleason/watchdog_pi/tree/frontend2_fix_up_MacOS
- Weatherfax - https://github.com/rgleason/weatherfax_pi/tree/frontend2_fix_up_MacOS

Any guidance would be appreciated.  Testplugin_pi seems to work.
Perhaps Jon will be able to advise.

rgleason, 2020-04-30:
Recent Ocpn_Draw v1.6 has the same permissions problem when built by circleci

https://app.circleci.com/pipelines/github/jongough/ocpn_draw_pi/64/workflows/298b7c3b-0f5e-4e42-a81b-017e22dc7f80/jobs/628/steps

rgleason, 2020-04-30:
We could just bag MacOS and Mingw and just upload the other files to be tested and just forget the problem.

bdbcat, 2020-04-30:
Rick...
If Ocpn_Draw has the MacOS permission problem, let us wait until Jon has a handle on the solution there.  Whatever needs to be done there will then transfer to your plugins clearly.