OpenCPN/OpenCPN #1880: Plugin Manager: macOS catalina - difficult to find files for a user

jongough, 2020-04-29:
Installed OCPN (OpenCPN_5.1.428+1.e63cdd2.dmg) on macOS Catalina. It runs OK. Finding information about where Plugin Manager has place files is very difficult and not user friendly. The only method of finding the opencpn.log file is to issue a `find . -iname "*opencpn.ini*"` command from a terminal. This will then locate the file in `~/Library/Logs/` which is not a location available in Finder. Looking through the log file shows that most of the files are maintained in locations that a standard user cannot get to. This will make problem solving quite difficult.

bcn2, 2020-04-29:
Jon..

any specific reason you used the dmg and not the pkg?
pkg will be the future standard install method.

jongough, 2020-04-29:
No reason except I know dmg is/was the install format for macOS. I am not a mac person and really don't use one. Does it make a difference if I use the pkg file?

bdbcat, 2020-04-30:
Jon...
The "Help" icon on the toolbar has direct links to the logfile and config file.  Not so hard to find...

rgleason, 2020-04-30:
Does oesenc_pi   darwin install properly on MacOS?

rgleason, 2020-04-30:
https://cloudsmith.io/~david-register/repos/ocpn-plugins-stable/packages/detail/raw/oesenc-darwin-10.13.6-tarball/v4.2.0/

bdbcat, 2020-04-30:
Rick...
Wrong link.  The master catalog has the correct links.

rgleason, 2020-04-30:
Ok, but I don't have mac to try it. 

https://dl.cloudsmith.io/public/rick-gleason/opencpn-plugins-prod/raw/names/vdr-darwin-10.13.6-tarball/versions/v0.5.1.5/vdr_pi-0.5.1-5_darwin-10.13.6.tar.gz

bdbcat, 2020-04-30:
Rick...
oesenc from the catalog installs and runs fine.
vdr_pi from the catalog installs and runs fine.

No need to poke around the repos.  Use the curated versions offered by the master catalog.

jongough, 2020-04-30:
I thought I had added this screenshot:
[image]
As you can see there is nothing in the 'Help' that helps

bdbcat, 2020-04-30:
Toolbar "circle-?"

jongough, 2020-04-30:
I am not sure what you mean by "circle-?". I just clicked the 'Help' and this is what it shows.

bcn2, 2020-04-30:
This one:
[image]

jongough, 2020-04-30:
I never have that icon showing in linux or windows as it just takes up space. I have always used the 'Help' on the toolbar/ribbon at the top. On the mac this does not appear. Is this a 'mac' special way of dealing with users or is something missing that would put the 'about' item in the drop down list?

bdbcat, 2020-04-30:
"On the mac this does not appear"
What does not appear?
The toolbar "circle-?" icon appears on all platforms, unless actively disabled by the user.

tai-fun, 2020-04-30:
Look at the attached screenshot from Linux.
Compare with Jon's screenshot above from Mac.
It seems that "About OpenCPN" is not present in the Mac's Menu bars "Help"
entry.

Thomas

bdbcat, 2020-04-30:
Ahh....
We finally get to the bug.
"It seems that "About OpenCPN" is not present in the Mac's Menu bars "Help""

Thanks
Dave

bcn2, 2020-04-30:
The "about" is in the OpenCPN menu - this is pretty standard for macOS, I think...
[image]

bdbcat, 2020-04-30:
So, no bug?

bcn2, 2020-04-30:
For me not...

jongough, 2020-04-30:
Shows you I don't use macOS. Just tried it and it is there. I am too used to non-mac where this type of information is maintained under 'Help'.

closed 2020-04-30 by jongough

rgleason, 2020-04-30:
Jon, could you take a look at the MacOS builds for squiddio, wx-routing, wxfax and watchdog?
Everyting builds but MacOS and it is always a "permissions" error?  Thanks.

bdbcat, 2020-04-30:
Rick...
Please do not cross post.  This Issue has nothing to do with your specific plugin troubles.

rgleason, 2020-05-01:
OK, but it was closed. Sloppy I guess.