OpenCPN/OpenCPN #1881: Instalable plugins don't show on a home build.

RooieDirk, 2020-04-29:
If I do a home build (5.1.428) the list of instalable plugins doesn't show. Downloading the same version from Dave's CloudSmith rep however works fine.
Something simular as the earlier 'mint' problem?? I am using linux KDE neon.
uname -a gives:  "Linux rd 4.15.0-96-generic #97-Ubuntu SMP Wed Apr 1 03:25:46 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux"
[image]

bdbcat, 2020-04-30:
Please show the contents of:
/etc/os-release

Thanks
Dave

RooieDirk, 2020-04-30:
Dave,

`
dirk@rd:~/Development/OpenCPN-Dev/OpenCPN/build$ cat /etc/os-release 
NAME="KDE neon"
VERSION="5.18"
ID=neon
ID_LIKE="ubuntu debian"
PRETTY_NAME="KDE neon User Edition 5.18"
VARIANT="User Edition"
VERSION_ID="18.04"
HOME_URL="http://neon.kde.org/"
SUPPORT_URL="http://neon.kde.org/"
BUG_REPORT_URL="http://bugs.kde.org/"
LOGO=start-here-kde-neon
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

`

RooieDirk, 2020-04-30:
Work around.
If I use the following in the config, it workts OK.

CompatOS=Ubuntu
CompatOsVersion=18.04

(the previous entry's where empty)

bdbcat, 2020-04-30:
Dirk...
Try the latest github master, with config file entry overrides cleared.  Hard for me to test locally...

RooieDirk, 2020-04-30:
Dave,
Sorry doesn't help. 
I will try something tomorrow, for now it is bet time overhere.

bdbcat, 2020-04-30:
Dirk...
I was afraid of that.
Best if you can step through the code at

PluginHandler::isCompatible() at pluginhandler.cpp:170.

Without config file overrides, we should expect Ubuntu derivatives to eventually find themselves in the test loop at line 238.

RooieDirk, 2020-05-01:
Dave,
But it did work almost....
osd_name_like[i] was set to ' **"**ubuntu ', wich is not equal to ' ubuntu '

I hacked some lines to remove all nonalpha characters, but maybe you find a more elegant solution.
Anyhow for me this works.
`// running OS may be "like" some known OS
        for(unsigned int i=0 ; i < os_detail->osd_name_like.size(); i++){
            
            std::string a ;
            for( char c : os_detail->osd_name_like[i] ) if( std::isalpha(c) ) a += c ;
            os_detail->osd_name_like[i] = a;
            
            if( os_detail->osd_name_like[i]  == plugin_os){`

bdbcat, 2020-05-01:
The osd_name_like[] vector originally comes from:
OCPNPlatform.cpp:385

This is where the /etc/os-release is parsed.

Please check that, and see where the non-alpha chars are coming from.

RooieDirk, 2020-05-01:
The ' " ' is coming directly from the /etc/os-release file.
inseting an cout at line 384 OCPNPlatform.cpp,

                else if(str.StartsWith(_T("ID_LIKE"))){
                    val = str.AfterFirst('=');
                    std::cout << "wxString str, val =   " << str.c_str() << "   " << val.c_str() << std::endl ;

gives:  wxString str, val =   ID_LIKE="ubuntu debian"   "ubuntu debian"

where "ubuntu debian" after a split by " ", is divided in "ubuntu and debian"

If I check the www then Ubuntu has an line like:
   ID_LIKE=debian        note without ""
and debian doesn't have an ID_LIKE line.

bdbcat, 2020-05-02:
Dirk...
OK, I get it.
The double-quotes are actually in the spec, and I missed it:
https://www.freedesktop.org/software/systemd/man/os-release.html

I'll fix, in the os-release file parsing logic.

RooieDirk, 2020-05-02:
Dave,
You are spending to much hours behind the screen.;-)  Although not being able to test the code you are writing does't help.
I made a pull request.

bdbcat, 2020-05-02:
Dirk...
Should be fixed in current github master.

closed 2020-05-02 by bdbcat