OpenCPN/OpenCPN #1883: Ubuntu 20.04 runtime problem.

tai-fun, 2020-04-30:
Dave
O beta from the ppa runs fine.
Compiling my own, also works, but when trying to run it, I get a crash pretty quick.
This happens at the time when O is loading charts.

[image]
 Thomas

bdbcat, 2020-04-30:
Thomas...
Just to confirm, please edit config file and disable wmm_pi.

tai-fun, 2020-05-01:
Dave

Setting
[PlugIns/libwmm_pi.so]
bEnabled=0

makes no difference. Still Crashing.

I even have a crash with all plugins disabled. Using CM93 chart group I can
just see the charts loaded when the crash comes. Looks like this....
Thomas

tai-fun, 2020-05-04:
I cleaned out and renewed the "build" directory. Made no difference.
Updating from Git master about 0600 UTC 2020-05-03, stopped the crashes, for some reason.
Closing the issue.

closed 2020-05-04 by tai-fun