OpenCPN/OpenCPN #1886: Weather-routing uninstall

rgleason, 2020-05-01:
When 1.13.6 is uninstalled, 
**User directory: for win** 
c:/programdata/opencpn/plugins/weather_routing_pi/
"boats" and "polars" directories exist but are empty afterwards.
weatherroutingconfiguration.xml exists with the configurations intact.
What is the proper condition to leave this in?

**Installation directory for win**
C:\Users\Rick\AppData\Local\opencpn\plugins
there is no sign of weather_routing found after uninstall.

Questions: What happens if there is an older non-managed version of weather_routing and the user wants to retain his data but use an earlier version of the plugin?
Is the data left intact in the user directory? 
c:/programdata/opencpn/plugins/weather_routing_pi/?

leamas, 2020-05-04:
The empty directories left after uninstall are basically there by design. They do no affect anything.

The handling of plugin data directories are up to the plugins. However, in order for a plugin update to work data must be left as-is after uninstall (update is just a uninstall - install combination).

The plugin installer does not touch the plugin data dirs. In fact, it doesn't even know where they are.

rgleason, 2020-05-04:
This needs further discussion and understanding, and the mechanics should perhaps be detailed further in your wiki, for Plugin Dev's trying to adapt to the new "standard" Plugin Manager.

However I am closing this important topic.

closed 2020-05-04 by rgleason