OpenCPN/OpenCPN #1887: Weather-routing Install Path   (acceptable format?)

rgleason, 2020-05-01:
I don't know if these are acceptable formats for tarballs and need it confirmed.

**Ubuntu 18.04**
weather_routing_pi-1.13.6.5_ubuntu-18.04-bionic-gtk3.tar.gz\weather_routing_pi-1.13.6.5_ubuntu-18.04-bionic-gtk3.tar\
- metadata.xml
- weather_routing_pi-1.13.6.5_ubuntu-18.04-bionic-gtk3
      - usr
              - local
                       - lib   (everything below is ok)
                       - share  (everything below is ok)

**Windows**
weather_routing_pi-1.13.6.5_msvc-10.0.14393-MSVC.tar.gz\weather_routing_pi-1.13.6.5_msvc-10.0.14393-MSVC.tar\
 - metadata
 - weather_routing_pi-1.13.6.5_msvc-10.0.14393-MSVC
       - plugins
              - weather_routing_pi
                     - data
                            - boats (dir)
                            - polars (dir)
                            - weatherroutingconfiguration.xml
              - weather_routing_pi.dll
       - share
              - locale
                      -  (languages)

rgleason, 2020-05-02:
Leamas has this for determining tarball format.
https://github.com/leamas/OpenCPN/wiki/Tarballs

I note that the tarballs being created have the "metadata.xml"  file 
- up one directory from "plugins" and "share" for windows
- up one directory from  usr

This is NOT what is shown in the wiki here for Tarballs
https://github.com/leamas/OpenCPN/wiki/Tarballs

Exactly which is correct?

bdbcat, 2020-05-02:
Rick...
re: "the tarballs being created"
What tarball exactly?  I'll take a look.

rgleason, 2020-05-02:
Dave, right under "Windows" and "Ubuntu"  above.
https://github.com/OpenCPN/OpenCPN/issues/1887#issue-610996907

Thanks.

bdbcat, 2020-05-02:
Either will work.
The current (e.g.oesenc_pi) is like the comments in this thread.  I would say Leamas' wiki is not strictly correct.
But, as I say, either will work.
So it looks like weather routing tarball is OK.

rgleason, 2020-05-02:
Great, thanks. maybe Alec can show either location as an option.

closed 2020-05-02 by bdbcat

rgleason, 2020-05-03:
Dave, Windows installation steps through nicely, using the Import Plugin routine, the User Data Directory: C:/ProgramData/Opencpn/plugins/weather_routing/boats   is still empty, same with polars directory.  

The last of these screenshots shows the empty directory. 
In the WeatherRouting Configuration menu down towards the middle, under "Boat"
There is a path showing "C:\ProgramData\opencpn\plugins\weather_routing\boats\Boat.xml
If you hit "Edit" you can then change the Boat.xml to a different boat setup. 
The problem is that the standard xml files aren't there:

In the olden days these files would come from 
C:\Program Files (x86)\OpenCPN 5.1.428+c80e933\plugins\weather_routing_pi\data\boats  (and polars)

but now they must come from 
C:\Users\Rick\AppData\Local\opencpn\plugins\weather_routing_pi\data\boats  (and polars)

The files we need copied over are  (all of them *.* )  Boat.xml, Boat-Climatology.xml, Boat-Test.xml,  Boat-Test-Power.xml,  but primarily Boat.xml in order for the plugin to work "out of the box" so to speak.    The Polars files are many  (*.pol, *.txt, *.csv generally) about 137 files.

the file destination needed is
C:\ProgramData\opencpn\plugins\weather_routing\boats
C:\ProgramData\opencpn\plugins\weather_routing\polars
These directories already seem to exist.

[image]
[image]
[image]
[image]
[image]
[image]

bdbcat, 2020-05-04:
Rick...
I added logging info to show the files being copied, when necessary.  It works correctly for me on Win10.
e.g.

`MESSAGE WeatherRouting.cpp:428 WeatherRouting copy file: C:\Users\dsr\AppData\Local\opencpn\plugins\weather_routing_pi/data/polars\Cal2-46.pol to C:\ProgramData\opencpn\\plugins\weather_routing\\polars\Cal2-46.pol
`
Please check latest build.

rgleason, 2020-05-04:
Dave,  I appreciate your help. https://github.com/rgleason/weather_routing_pi/commits/master
I see the message above and [the .kdev4 files](https://github.com/rgleason/weather_routing_pi/commit/2f586ab79ccb25ef1097804ce514bf6c7d129609) 
[opencpn.log.txt](https://github.com/OpenCPN/OpenCPN/files/4575286/opencpn.log.txt)
Where do I find kdev4 output?

WeatherRouting.cpp:428 
WeatherRouting copy file: 
C:\Users\dsr\AppData\Local\opencpn\plugins\weather_routing_pi/data/polars\Cal2-46.pol 
to 
C:\ProgramData\opencpn\\plugins\weather_routing\\polars\Cal2-46.pol 

The source and destination paths appear to be correct, however when I install the plugin and check, I find no files located in:
C:\ProgramData\opencpn\\plugins\weather_routing\\polars
Why?   Is it the double backslash?

**Comment**
Normally we would have weather_routing_pi/data data files located here, just like the included plugins, simple and familiar:  
C:\Program Files (x86)\OpenCPN 5.1.428+c80e933\plugins\weather_routing_pi\data

Instead the new Plugin Manager installs them here:
C:\Users\Rick\AppData\Local\opencpn\plugins\weather_routing_pi\data

The plugin should copy the data files (polars\*.* and boats\*.*) to the User Read/Write Data directory located here:
C:\ProgramData\opencpn\plugins\weather_routing\polars\*.*  (all files)
C:\ProgramData\opencpn\plugins\weather_routing\boats\*.* (all files)
So that the user has access to them as the plugin expects, and so that user changes can be made and saved.

**Suggestion**
I believe the plugin should be programmed to copy the respective files from either the older source directories Program Files (x86)\OpenCPN 5.1.428+c80e933\plugins\weather_routing_pi\data
or the new Plugin Manager directory under %localappdata%\opencpn\plugins\weather_routing_pi\data  depending on weather the program has been built as a Managed Plugin or an older Packaged plugin.

rgleason, 2020-05-04:
That is entirely weird, the second time I tried to install it this morning, it seemed to work.
I know I had deleted the weather_routing data directory in ProgramData both times, but I believe the problem was clearly due to a previous installation and remainders left.

Sorry about the trouble. Will make a PR to master now.   How do I find the output of kdev4?  Very interested.

[image]