OpenCPN/OpenCPN #1889: Build errors on Debian sid/bullseye

leamas, 2020-05-02:
As heading says: as of the beta state, opencpn does not build without patches  on debian sid/bullseye.

I'm on it -- this is about the downstream packaging. Will take some time

leamas, 2020-05-02:
One of the problems: build and errors due to missing gtk3 headers and libs. The obvious (?) patch below.

@bdbcat : Since you commented out these lines: do you have any objections?

    index e236f5b..c3f51a7 100644
    --- a/CMakeLists.txt
    +++ b/CMakeLists.txt
    @@ -603,9 +603,9 @@ endif (OCPN_BUNDLE_DOCS MATCHES "ON")
     #
     # Linux: set up GTK (wxWidgets prerequisite).
     #
    -#if (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
    -#  include(OcpnFindGtk)
    -#endif ()
    +if (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
    +  include(OcpnFindGtk)
    +endif ()
 
     # set a build flag for arm architecture, to catch any rPI runtime changes
     # required
    @@ -1147,8 +1147,8 @@ target_link_libraries(${PACKAGE_NAME} PRIVATE ssl::sha1)
     # Linux: set up GTK
     #
     if (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
    -#  include(OcpnFindGtk)
    -#  target_link_libraries(${PACKAGE_NAME} PRIVATE ocpn::gtk)
    +  include(OcpnFindGtk)
    +  target_link_libraries(${PACKAGE_NAME} PRIVATE ocpn::gtk)
     endif (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
 
     # Search for opengles, short of running a program to test the speed of

bdbcat, 2020-05-02:
I think this will be OK.  The problem I observed was that when GTK3 is used, something in wxWidgets also defines the macro "__WXGTK20__".  That's odd.

Anyway, this confused the plugin elf compatibility check at pluginmanager.cpp:2034.  So I rearranged the macro check order there.  And all is well now, even if both "__WXGTK20__" and "__WXGTK3__" are defined.

To test, we need to try managed plugin detection on Ubuntu 20.04

leamas, 2020-05-02:
OK, thanks! There will be a PR, later. 

I have a preliminary builds of 0.5.1.148 using the official package toolchain available for Debian.  Nothing complicated, this patch was the most problamatic.

PR #1892 (2020-05-07): Debian downstream packaging fixes.
A bunch of patches from the downstream Debian packaging. Notably linking issues as discussed in #1889 and also gshhs bundling as discussed in #1875. EDIT: #1750.

The latter drops the configuration option completely, always bundling the crude data.
EDIT: the latter just sets the defaults to bundling gshhs and tcdata.

The rest should be trivial. some bugfix and a bunch of fixed warnings -- it now builds cleanly on sid.

Note: All of these patches have been applied to the downstream packaging branch. Accepted or not, they can still be simple downstream patches [... cut]
closed 2020-05-07 by bdbcat