OpenCPN/OpenCPN #1891: Latest stable update fails on RPI

andyrbarrow, 2020-05-02:
RPI 4, 4GB. Running headless with VNC. OpenGL enabled. 386K of video memory.

Ran "apt install opencpn" and upgraded to 5.0.0-20200428. OpenCPN fails to start.
Output of terminal:
`pi@raspberrypi:~ $ opencpn
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 934
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:5014:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
10:34:05: Device: 0: bcm2835 ALSA: IEC958/HDMI (hw:0,1)
10:34:05: Device: 1: bcm2835 ALSA: IEC958/HDMI1 (hw:0,2)
10:34:05: Device: 2: pulse
10:34:05: Device: 3: upmix
10:34:05: Device: 4: vdownmix
10:34:05: Device: 5: dmix
10:34:05: Device: 6: default
10:34:05: Device: 0: bcm2835 ALSA: IEC958/HDMI (hw:0,1)
10:34:05: Device: 1: bcm2835 ALSA: IEC958/HDMI1 (hw:0,2)
10:34:05: Device: 2: pulse
10:34:05: Device: 3: upmix
10:34:05: Device: 4: vdownmix
10:34:05: Device: 5: dmix
10:34:05: Device: 6: default
10:34:05: Device: 0: bcm2835 ALSA: IEC958/HDMI (hw:0,1)
10:34:05: Device: 1: bcm2835 ALSA: IEC958/HDMI1 (hw:0,2)
10:34:05: Device: 2: pulse
10:34:05: Device: 3: upmix
10:34:05: Device: 4: vdownmix
10:34:05: Device: 5: dmix
10:34:05: Device: 6: default
10:34:05: Device: 0: bcm2835 ALSA: IEC958/HDMI (hw:0,1)
10:34:05: Device: 1: bcm2835 ALSA: IEC958/HDMI1 (hw:0,2)
10:34:05: Device: 2: pulse
10:34:05: Device: 3: upmix
10:34:05: Device: 4: vdownmix
10:34:05: Device: 5: dmix
10:34:05: Device: 6: default

(opencpn:2598): Gdk-CRITICAL **: 10:34:05.798: IA__gdk_colormap_new: assertion 'GDK_IS_VISUAL (visual)' failed

(opencpn:2598): Gtk-CRITICAL **: 10:34:05.798: IA__gtk_widget_set_colormap: assertion 'GDK_IS_COLORMAP (colormap)' failed

(opencpn:2598): GLib-GObject-CRITICAL **: 10:34:05.799: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
libGL: built on Mar  4 2020 04:35:45
libGL: warning, gles is NULL
`
Downgrading with "sudo apt install opencpn=5.0.0-0~bionic1" gives a working version.
Starting from terminal after downgrade gives the same ALSA and JACK messages, but not the GL messages.

bdbcat, 2020-05-02:
From what repo did you get "5.0.0-20200428"?
I do not recognize this package version.

andyrbarrow, 2020-05-03:
Okay, I think I understand what happened. I'm using OpenPlotter. In order to have a repository for xygrib, free-x.de repository was added to /etc/apt/sources.d/openplotter.list as part of the installation. Apparently they just added version 5.0.0-20200428 of OpenCPN to that repository, which my system picked up as an upgraded version.

Now I need to figure out how to get apt to ignore that repository for OpenCPN, but use it for xygrib.

Sorry for the trouble.

bdbcat, 2020-05-03:
So, that version "5.0.0-20200428" looks like it trying to be something like 5.1.428, but broken in some way.
Have you tried the OpenCPN Beta PPA repo?  You should get the "official" 5.1.428 Beta article.

free-x, 2020-05-03:
@andyrbarrow 
actually OpenPlotter should use other repository
Why is the old repo still tied to you? I don't know. Here is the correct entry for openPlotter users
```
deb https://www.free-x.de/deb4op debian main
```
but thanks for the tip. I'll see what's wrong with the package.

andyrbarrow, 2020-05-03:
Looks like this is down to a bad repo entry in openplotter.list. Closing.

closed 2020-05-03 by andyrbarrow