OpenCPN/OpenCPN #1893: Plugin Manager: compatability checking of plugins not working

jongough, 2020-05-03:
On Ubuntu 18.04 with both wxWidgets 3.0.4 & 3.1.2 installed using current OCPN (0dc7c9f3) and testplugin_pi built using wxWidgets 3.1.2 installs OK by being downloaded from cloudsmith using the custom URL process. The plugin is initiated but during the process OCPN gets a sigfault. 

The code at pluginmanager.cpp 2030 (as pointed to by bdbcat) does not pick up this type of issue. It will also not pick up other issues around software that needs to be installed for the plugin to work successfully.

As the plugin knows at build time what its runtime requirements are these should be in the metadata xml file to allow pluginmanager to check that the prerequisits are available. There is currently no way for a plugin to require some non-OCPN provided software to be installed automatically, or for the user to be notified in a friendly way. Currently the only way is to crash OCPN. Current, production, installation mangers do this by default, OCPNs version of an installation manager should also do this by default.

bdbcat, 2020-05-04:
See #1876

bdbcat, 2020-05-04:
" It will also not pick up other issues around software that needs to be installed for the plugin to work successfully."

This is a completely separate issue, and not contemplated by OCPN as currently designed.

bdbcat, 2021-04-29:
Closing, inactive

closed 2021-04-29 by bdbcat