OpenCPN/OpenCPN #1894: Build: Dangerous  "bundle GSHHS data" option should be removed.

leamas, 2020-05-04:
After discussions in  #1750 and #1712  it has been settled that we should mirror the official package structure and also always bundle the crude GSHHS data.

In this perspective the OCPN_BUNDLE_GSHHS  config option is useless and even dangerous. The best would be to drop it entirely, thus ensuring that the GSHHS data always is present in some form.

To me, this is  a beta blocker since it is related to the upgrade paths between PPA and official Debian packages (that is, if such upgrade paths is a goal for next release).

rgleason, 2020-05-04:
Have you thought about perhaps guiding users towards a single package, whichever one should be the preference, so that there are not so many environments to build?  Would that help determine how GSHHS data is handled?

transmitterdan, 2020-05-04:
I agree we should always bundle CRUDE (and only CRUDE) GSHHS data with official releases.  I am unclear how this affects upgrades of existing distro users but think it is manageable based on earlier comments.

But I don't agree to remove the option from cmakelists.txt file.  The O project is used by more than just the PPA builders. This file is used by individual builders and they may have good reason to want to include or exclude GSHHS and TCDATA files in a single package.  For example, in the past some environmental groups used OpenCPN for research and they want to bundle hi-res GSHHS data because their users do not have (and cannot have) internet access.  There are probably other examples.  Some plugins also make good use of high-res GSHHS data.  So if someone wants to create a single package for individual use they may need to bundle large GSHHS datasets.

leamas, 2020-05-04:
@transmitterdan: 

Indeed. OTOH, larger resolutions don't conflict with the crude data, it's different filenames (please correct me if I'm wrong). 

So, the only reason to not bundle the crude data after adding higher resolutions is the wasted space. My thinking is that after adding higher resolutions (not to mention actual charts) the space consumed by the crude data could be neglected.

> So if someone wants to create a single package for individual use they may need to bundle large GSHHS datasets.

Right. But does this usecase justify adding/keeping complex cmake options, with obvious ptifalls?

Can  users like these environmental groups use the chartdownloader "in advance" to add whatever gshhs data they need, and distribute this like a chart set together with opencpn? After all, even such groups will need charts anyway?

I think we would benefit a lot by having one set of (crude) data always available, and one single mechanism (the chart downloader) to add higher resolutions.  

Here is also the implicit assumption that users installing opencpn have an internet connection.

Thoughts?

leamas, 2020-05-04:
Hm... my memory is quite good, but short.

Normally, I wouldn't  put an issue like this on the table in a beta cycle. However, the current gshhs option is broken (nothing bundled), so it needs to be fixed anyway. In this situation a complete overhaul IMHO is the best alternative. 

Will look further into a smaller bugfix as an alternative.

rgleason, 2020-05-04:
Dan, isn't the Chartdownloader option so dead simple and familiar that everyone should be able to use it? Why wouldn't that be a better alternative? 

I  understand the point about the environmental groups, but can't they change how they work to use these easier tools?

rgleason, 2020-05-04:
I would like to see a future plan to move users towards certain types of builds to simplify and reduce overhead which is directly proportional to the number of environments used.

leamas, 2020-05-04:
Commit above is a minimal fix which should work in beta bugfix context. 

The overall issue in this bug is still open. However, this could now be postponed to next release cycle.

transmitterdan, 2020-05-04:
> Dan, isn't the Chartdownloader option so dead simple and familiar that everyone should be able to use it? Why wouldn't that be a better alternative?
> 
> I understand the point about the environmental groups, but can't they change how they work to use these easier tools?

I already gave the reason.  These users don't have an internet connection neither are they able to even temporarily hook one up.  For them the "easiest" is to bundle the hi-res datasets into the install package.  Then they only need one file to hand out to all their users.  If something changes they can simply send out a new package file on a stick. Some of these users don't use charts at all.  They just use the GSHHS data to identify land/water boundaries.  There are many reasonable uses of O besides chart navigation.  I don't think it's proper to enumerate them here.

There are enough changes coming in 5.2 that I think we don't need even more draconian changes. Just my opinion...

bdbcat, 2020-05-04:
Alec...
On my large list for resolution before next Beta point.
I agree that we need to get to compatibility with the "official" debian package, to avoid serious upgrade troubles later.
Probably will adopt your patch, or something close to it.
I have a private script for building PPAs, and can build PPAs with CMAKE arguments as necessary, per-platform.
Please stand by.

leamas, 2020-05-05:
@bdbcat:
> However, the set it will not build PPAs directly upgradable from the "official" 5.0

It should be possible to build the PPA packages from the same toolchain as building the official package. The trick is basically some control fixes like Replaces: or so.

I could try to revise the official package so it can replace/upgrade  a PPA package. Don't know if the necessary Replaces: in control will survive a review, but it's worth a try. 

In any case, maintaining a branch of the official build chain which produces packages compatible with the current PPA package should be simple. The diff is likely to be very small. That is, one possible route is to use the official build chain also in the PPA. 

This would de-duplicate the packaging effort. And, important, it would also give us source packages, one of the missing features of current PPA solution.

leamas, 2020-05-05:
Made a proof-of-concept, a focal build based on the Debian/sid packaging. It's more or less out of the box. Available at https://gitlab.com/leamas/opencpn, in the focal/master branch.

Builds a package with the same structure as the official, but with (hopefully) correct Breaks/Replaces/Transition packages to cope with an upgrade from the previous PPA builds. 

Why? because it's possible... Note the debian/README.ubuntu.

leamas, 2020-05-06:
@transmitterdan : I hear you. Also noting the word *draconian* which carries a certain weight. 

> These users don't have an internet connection neither are they able to even temporarily hook one up

Isn't the easiest for such users to distribute a standard opencpn package + for example opencpn-gshhs-full  which still exists? Users then need to install two packages instead of of one, a drawback. But OTOH there is no need to generate and maintain a custom package, which is a advantage.

In the end, it seems that we still support this usecase quite well, at least on Debian/Ubuntu  (?)

leamas, 2020-05-06:
Perhaps a pandora's box to be kept closed, but: what was the problem with using the already existing gmt-gshhs-full et. al.  packages in Debian?

@nohal @bdbdcat: thoughts?

transmitterdan, 2020-05-06:
> In the end, it seems that we still support this usecase quite well, at least on Debian/Ubuntu  (?)

Sometimes I think we forget that 80+% of users are using Windows. There is no package manager for them.

I don’t see the controversy in keeping the options in the CMakeLists.txt file. Why is that “dangerous”? All PPAs are built with tested scripts. There is next to zero chance a PPA build will accidentally use the wrong option.

leamas, 2020-05-06:
> Sometimes I think we forget that 80+% of users are using Windows

IIRC,  it's actually about 50% according to Hubert's figures (sold chart licenses) . But you still have a point.

I have revised my [patch](https://github.com/OpenCPN/OpenCPN/pull/1892/commits/ff0f4f020d0) to just make it bundle the crude data by default. We could probably agree on that,  thus solving the issue that a default cmake  invocation without options doesn't bundle neither tcdata nor gshss. This should settle the short-time issue. OK?

The bigger issue is the upgrade paths between the PPA packages and the official Debian packages. This needs to be fixed in that context. 

BTW: The missing package manager on Windows is actually a strong reason to promote the platform-indepent chart downloader...

transmitterdan, 2020-05-06:
@leamas,

I believe this all started with https://github.com/OpenCPN/OpenCPN/issues/1712

I am not sure we needed to change anything from how it was in the past except fix a dependency bug which Dave has done.  I guess I thought you had felt the past practice needed to change which in turn has broken the upgrade path.  I fully agree we need to fix the upgrade path problem.  Perhaps there is nothing wrong with how it has always been.

FWIW, most users don't buy charts from Hubert.  The US accounts for a huge portion of the user base and they don't have to pay for charts.  They are mostly Windows based.

leamas, 2020-05-06:
@transmitterdan 

Here are two issues: 

The first is the short-term bundling of tcdata and gshhs. My first stab at this was to just drop the option completely (hence now misleading bug subject). However, I have seen the light and is now only proposing  this [patch](https://github.com/OpenCPN/OpenCPN/pull/1892/commits/ff0f4f020d0). Can we agree that this is acceptable?

The second issue is the package upgrade paths. This is now affected by the official package in Debian, which have another structure than the PPA packages (because of packaging rules). The need to fix these upgrade paths is another issue which I though we could discuss here.

I think we agree that these upgrades should work, and that a general strategy is to let the ppa packages mirror the official package structure. But the devil is in the details....

bdbcat, 2020-05-06:
@leamas 
But the devil is in the details....  indeed.
Working on that now, in background.  Build bots are slow....
Should have something later today, I hope.

bdbcat, 2020-05-07:
@leamas 
I manged to build a PPA for 20.04(focal) mirroring the "official" debian distribution, with packages "opencpn" and "opencpn-data".  These packages successfully upgrade the debian package offered by focal default.  And they can also recover from the initial Beta PPA, with its (now deprecated) dependence on GSHHS data sets.
This was done using your PR for cleaning up sid builds, which I have not yet committed to github.  But will very soon.

So I think we are all there.

...except...Couple of questions:
I used some parts of your gitlab focal/master branch to craft debian/control, and friends for PPA builds.  But I had trouble with a couple of files:
```
usr/share/metainfo/opencpn.metainfo.xml
usr/share/doc/opencpn/license.txt
usr/share/doc/opencpn/README.harmonics
usr/share/doc/opencpn/README.bells
```
Where does the opencpn.metainfo.xml come from, and what is it?

Finally, I see no real need to backport this mess to bionic or xenial.  Do you?

leamas, 2020-05-07:
> usr/share/metainfo/opencpn.metainfo.xml

This is former opencpn.appdata.xml (the old location is deprecated).  The relevant patch was supposed to be part of my PR, but somehow was lost. Hand-written patch: 
```

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 8d5cd8553..f79316660 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2178,7 +2178,7 @@ if (NOT APPLE)
     install(FILES data/opencpn.desktop DESTINATION ${PREFIX_DATA}/applications)
+    install(DIRECTORY DESTINATION ${PREFIX_DATA}/metainfo)
     install(
       FILES ${CMAKE_BINARY_DIR}/opencpn.appdata.xml
-      DESTINATION ${PREFIX_DATA}/appdata
+      DESTINATION ${PREFIX_DATA}/metainfo
+      RENAME opencpn.metainfo.xml
     )
     install(FILES opencpn.1 DESTINATION ${PREFIX_DATA}/man/man1)
   endif (UNIX)

```

> usr/share/doc/opencpn/license.txt
> usr/share/doc/opencpn/README.harmonics
> usr/share/doc/opencpn/README.bells

These are handled by the rules file in the packaging at https://gitlab.com/leamas/opencpn. The snippet:
```

override_dh_auto_install:
        dh_auto_install
        # Remove generated files without source.
        rm -rf debian/tmp/usr/share/opencpn/doc/opencpn
        # Move plugin documentation to proper location
        -mv debian/tmp/usr/share/opencpn/sounds/README.bells \
            debian/tmp/usr/share/opencpn/tcdata/README.harmonics \
            debian/tmp/usr/share/doc/opencpn
        # Use existing license files instead of bundled copy
        rm -f debian/tmp/usr/share/doc/opencpn/license.txt
        ln -s /usr/share/common-licenses/GPL-2 \
            debian/tmp/usr/share/doc/opencpn/license.txt

```

leamas, 2020-05-07:
> Finally, I see no real need to backport this mess to bionic or xenial. Do you?

Absolutely not.

leamas, 2020-05-10:
The OCPN_BUNDLE_GSHHS option is removed, presumably as part of effort to align PPA and Debian packages (?).

@bdbcat:  time to close?

leamas, 2020-05-10:
BTW: I note that as of current master 71656a263 the default `cmake - make - make install` does not install the tcdata HARMONICS.idx and friends, also when using -DOCPN_BUNDLE_TCDATA.  Is this on purpose?

bdbcat, 2020-05-10:
Hmmm..
Seems to be working for me.
TCData and friends install OK for me with defaults.
Dave

leamas, 2020-05-10:
Indeed it does, my bad (forgot to remove CMakeCache?). And, finally found the bundle_gshhs_opt() and bundle_tcdata_opt() which have lower case invocations of macros defined in upper case.

So, everything seems in place. Close?

leamas, 2020-05-10:
@transmitterdan : I was wrong, again. The OCPN_BUNDLE_GSHHS is still there (as well as OCPN_BUNDLE_TCDATA).

Sigh. Sorry for the noise.

transmitterdan, 2020-05-10:
@leamas : Not to worry. All seems ok now. Have built for 19.10 and Raspbian Buster and all seems ok.  Defaults do the right thing I think.

leamas, 2020-05-10:
Ok, time to close. Thanks to all of you!

closed 2020-05-10 by leamas