OpenCPN/OpenCPN #1895: Scaling bug - Real Scale Vector Ship

campbellhennessy, 2020-05-04:
Hello,

I believe I stumbled across an issue with the real scale vector ship icon.  When measured the icon did not agree (even at max zoom) with the vessel parameters.  Length of the icon appeared to be tied to the breadth, see attached images.

I believe I found the issue in the code (only the x scaling factor was being applied).  I have made the change on my fork of OpenCPN and it tests well on my machine (the icon now has the correct dimensions).  I will post a PR with the fix... I'm new to submitting, do I need an issue before a PR?
[image]
[image]
[image]
[image]

campbellhennessy, 2020-05-04:
I am trying to do a PR but somehow I seem to have changed a lot of the whitespace in the file I am working on.  Any suggestions?  I assume it has something to do with line endings.

transmitterdan, 2020-05-04:
The instructions on how to configure git should solve this problem.  Perhaps you did not precisely follow the instructions as to how to handle LF and CRLF in git.  You can submit the PR and we can see the problem.  You can always close the PR without any harm being done.

campbellhennessy, 2020-05-05:
Submitted PR #1896 to address this issue.  If anyone finds this later that has similar problems with whitespace, I fixed it by turning off automatic removal of trailing whitespace in my editor.

campbellhennessy, 2020-05-05:
#1896

campbellhennessy, 2020-06-03:
PR #1896 merged into master.

closed 2020-06-03 by campbellhennessy