OpenCPN/OpenCPN #1897: flatpak beta test need to be started

leamas, 2020-05-05:
I have set up a separate flatpak repo for the beta test. Preliminary dogfood testing looks OK using oesenc.

I think it's time to announce the flatpak beta. The bottleneck so far has been missing and/or wrong docs. Question is now how to get the message below ot our users.

------

The opencpn flatpak build is targeted to Linux users on non-debian distros includiing (but not limited to) Fedora, Arch and SuSE.

To test the beta, first install flatpak as described in https://flatpak.org/setup/

Install the flathub remote:

    $ sudo flatpak remote-add --if-not-exists \
        flathub https://dl.flathub.org/repo/flathub.flatpakrepo

Install the beta;

    $ flatpak install --user \
        http://opencpn.duckdns.org/opencpn-beta/opencpn.flatpakref

The initial installation is possibly huge, around 450Mb, This is a one-time setup download, later updates weighs in at about 50 Mb. Updates can be done using   `flatpak update org.opencpn.OpenCPN`

Running from command line:

    $ flatpak run org.opencpn.OpenCPN

Running from GUI: the flatpak installation is integrated in the desktop. Using for example gnome the flatpak'ed opencpn can be started as any other application.

leamas, 2020-05-13:
@bdbcat: Updated beta repository to 0e1b59584 + some local, flatpak-only patches:

```
git diff --stat 0e1b59584
 ci/generic-build-flatpak.sh      | 13 +++++++++++++
 data/opencpn.appdata.xml.in      | 28 ++++++++--------------------
 flatpak/Makefile                 |  3 ++-
 flatpak/org.opencpn.OpenCPN.yaml |  3 ++-
 4 files changed, 25 insertions(+), 22 deletions(-)
```

There will be a PR, for flatpak to work it needs to be before the release.

rgleason, 2020-05-13:
Now weatherfax flatpak build is failing since this change:
https://github.com/rgleason/weatherfax_pi/commit/a7289497e55a03c81c7a3bbb6b67afe838896104
Is this an unrelated problem?

bdbcat, 2020-05-14:
@leamas...
Not sure if this is a TODO for me or not....

bdbcat, 2020-05-14:
Rick...
Where is the build log?

leamas, 2020-05-14:
@bdbcat: I basically need your blessing so that users understands  that the flatpak beta is part of the official beta program.

One way could be that you repeated something like the message outlined in the initial comment in CF -- I'm sure there are other ways.

leamas, 2020-05-14:
@rgleason: In what way is the weatherfax plugin related to this issue?

bdbcat, 2020-05-14:
@leamas...
Consider Flatpak blessed.
I will write up something for CF, to teach, mainly, and also to try to get some testing going.

rgleason, 2020-05-14:
```
Now weatherfax flatpak build is failing since this change:
rgleason/weatherfax_pi@a728949
Is this an unrelated problem?
```
Jon figured out that somehow I did not have the github repository parameter in CMakeList.  So it is working now.  I don't always know causticity.
Thanks.

leamas, 2020-05-15:
@bdbcat :
> I will write up something for CF, to teach, mainly, and also to try to get some testing going.

Thanks, this is what I have envisioned, so to speak.

bdbcat, 2020-05-15:
@leamas...
Can you state, once more, the definitive command line instructions to install and run OCPN Flatpak?
Thanks
Dave

leamas, 2020-05-16:
For the beta: as described in https://github.com/OpenCPN/OpenCPN/issues/1897#issue-612887875

hreuver0183, 2020-05-16:
The OpenCPN documentation on flatpak is updated with above instructions and the comments from Dave in CF. Currently the page is shown in the regular Manual layout as "FlatPak (beta)".

leamas, 2020-05-17:
@hreuver0183, @bdbcat  : Thanks, great!

Assuming this is  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak.

Here is a minor error you might want to fix: For Fedora, the command   `sudo yum install flatpak is` described.. This is wrong for two reasons: flatpak is already pre-installed on Fedora, and furthermore yum has been replaced with dnf since long. I suggest that you drop this part; the https://flatpak.org/setup link is sufficient (and correct). 

That said, my goal with filing this bug was to get the flatpak beta ofiicially announced.  Since it's now is, let's close this bug.

closed 2020-05-17 by leamas

bdbcat, 2020-05-17:
"This is wrong for two reasons: flatpak is already pre-installed on Fedora, and furthermore yum has been replaced with dnf since long."
True for reasonably modern Fedora.  Maybe not so for older systems we are trying to support...

leamas, 2020-05-17:
@bdbcat: yum was replaced by dnf  in 2015. At this point, flatpak wasn't available so referring to yum instead of dnf is indeed plain wrong.

Flatpak has been included in Fedora since at least 2017. There is no way we could support such an old Fedora system, which has a very limited support window (one year).  Specifically, I seriously doubt the flatpak version used here could run the OpenCPN flatpak package.

Frankly: using flatpak requires a reasonably modern linux distro.  A several years old Fedora does not fall into this category. Fedora user who wants a  stable system uses for purposes like a navigation PC uses CentOS, not Fedora.

And: even if we wanted to support  for example F23 (which would be a major mistake, for sure), the correct info would be something like _"On Fedora, flatpak is pre-installed, just add the flathub remote using...,. Very old versions of Fedora might require adding flatpak using dnf install flatpak."_

All in all, this info soon get's complicated , hard to read and targeting a hypothetical user which more or less just not exist. The correct way to is to use the official docs by linking to https://flatpak.org/setup.

bdbcat, 2020-05-17:
@leamas...
So, you say, and I believe you.
However, I have found in over ten years of writing docs for OCPN users, that they want, nay need, to find all the basic information they require on a single page.  Chasing around multiple links to follow a multistep process, without a clear roadmap,  leads, often, to complete confusion.
So, maybe there is a correct way, and maybe there is a friendly way.  I find that O users read and respond better to the friendly way, even if sometimes incomplete, or even wrong.
So, this document stays.  And anyway, CF posts are incredibly transient.  Totally forgotten in 2 weeks.
We may be able to do better in the Wiki, which is IMHO today not "friendly" with respect to Flatpak.

Thanks for the facts.
Dave

leamas, 2020-05-18:
So, while we agree that the info should be factually correct, we disagree on the use of links.

Basically you prefer to have all information on a a single page. I understand your motivation, and you certainly have a point. And my phrasing that there is one "correct" way was clumsy;  sorry for that.

Personally, I have found that users avoids reading long text blocks. That is, while the use of links can lead to a mess, it's actually useful in other cases. It is IMHO not black or white but rather shades of grey. The Swedish adverb "lagom" meaning "not too much or too little" in this context comes to mind.

Given this the overall layout of the current wiki is perhaps not that bad with a short, terse summary for those who want that and the expanded text below for those who want that. Even when heading for the "complete" text the summary serves as an introduction.