OpenCPN/OpenCPN #1898: Problem mixing Gtk2 and Gtk3

tai-fun, 2020-05-08:
The last couple of days O is not running after compilation.
An instant crash with this message when starting from the CL:

(opencpn:7758): Gtk-ERROR **: 08:11:59.684: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
Trace/breakpoint trap (core dumped)

Using Ubuntu 20.04. Remember that this version of Ubuntu is Gtk3 only.

leamas, 2020-05-08:
See: https://github.com/OpenCPN/OpenCPN/commit/3d85a53aa3beef82068b2b7128aaa09870bb8c64#commitcomment-39026295

Basically, the configuration needs -DOCPN_FORCE_GTK3=ON to work. Copy from the commit comment:

---

First stab on this: resurrect some bits lost in bug hunting:

    diff --git a/CMakeLists.txt b/CMakeLists.txt
    index 7dc6a13c0..9137e353d 100644
    --- a/CMakeLists.txt
    +++ b/CMakeLists.txt
    @@ -712,13 +712,11 @@ if (NOT QT_ANDROID)
         set(wxWidgets_EXCLUDE_COMMON_LIBRARIES TRUE)
       endif (MSVC)

    -#  if (GTK2_FOUND)
    -#  set(wxWidgets_CONFIG_OPTIONS ${wxWidgets_CONFIG_OPTIONS} --toolkit=gtk2)
    -
    -# if (GTK3_FOUND)
    -#    set(wxWidgets_CONFIG_OPTIONS ${wxWidgets_CONFIG_OPTIONS} --toolkit=gtk3)
    -#  endif ()
    -    message(STATUS "Main: ${wxWidgets_FIND_COMPONENTS}")
    +  if (GTK2_FOUND)
    +    set(wxWidgets_CONFIG_OPTIONS ${wxWidgets_CONFIG_OPTIONS} --toolkit=gtk2)
    +  elseif (GTK3_FOUND)
    +    set(wxWidgets_CONFIG_OPTIONS ${wxWidgets_CONFIG_OPTIONS} --toolkit=gtk3)
    +  endif ()

       find_package(wxWidgets COMPONENTS ${wxWidgets_FIND_COMPONENTS})
       if (wxWidgets_FOUND AND OCPN_USE_WEBVIEW)
    @@ -728,7 +726,7 @@ if (NOT QT_ANDROID)
         add_definitions(-DwxUSE_WEBVIEW=0)
         list(REMOVE_ITEM wxWidgets_FIND_COMPONENTS webview)
         message(STATUS "Could not find wxWidgets webview add-on")
    -    find_package(wxWidgets COMPONENTS ${wxWidgets_FIND_COMPONENTS})
    +    find_package(wxWidgets COMPONENTS ${wxWidgets_FIND_COMPONENTS} REQUIRED)
       endif ()
       if (MSYS)
         # Convert msys to windows paths, and handle the missing /usr

This makes the cmake configuration just break if used without -DOCPN_FORCE_GTK3  on platforms not supporting gtk2. Good enough?

transmitterdan, 2020-05-08:
Generally, we have always required this  "cmake .." without any options to work on all Linux platforms.

leamas, 2020-05-08:
Basically, I agree.

However, the other part is that we have since long been having gtk2 as the default. It is perfectly possible to detect the situation the there is gtk3 but not gtk2, and in this case force the use of gtk3. But if there is any kind of mistake in that code, there is no escape for someone trying to build gtk2.

It's not that I'm impossible to convince to set gtk3 implicitly. But it feels somewhat scary, especially in the beta cycle. 

If we still let it fail, but with a better error message mentioning -DOCPN_FORCE_GTK3? Would that be OK?

transmitterdan, 2020-05-08:
Hmmmm.. Would have to think about it.

I think both gtk2 and gtk3 are available on 19.10 but the new CMakeLists.txt assumes one flavor of gtk before finding wxWidgets which for whatever reason prefers another version.  I think we should sort that out.  Why is it important to decide on a gtk version before finding the wxWidgets libraries?  Doesn't the process of finding wxWidgets necessarily also drag in the right gtk version?

leamas, 2020-05-08:
In any case I was wrong. It should be simple to setup the default value for OCPN_FORCE_GTK3. This carries no substantial risks, since it can be overridden on command line. I'm on it, will take some time (heading for a bottle of wine, company and dinner right now).

rgleason, 2020-05-08:
Should I continue trying to install Opencpn  Ubuntu 18.04 or 18.04 gtk3 seem to work? 

I have a git clone of Opencpn beta and have built that and sudo make install that and have been using that.

Doesn't Ubuntu support parallel installs?
I have been trying to install the beta tarball for [Ubuntu 18.04 and gtk3](https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/detail/opencpn-bionic-gtk3_5142810ef89d2-1_amd64deb/) with no success.

https://www.cruisersforum.com/forums/f134/opencpn-version-5-1-428-beta-test-233515-4.html#post3134100  Download from here [Ubuntu 18.04 bionic gtk3
](https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/detail/opencpn-bionic-gtk3_5142810ef89d2-1_amd64deb/)

https://www.cruisersforum.com/forums/f134/opencpn-version-5-1-428-beta-test-233515-3.html#post3134207

https://www.cruisersforum.com/forums/f134/opencpn-version-5-1-428-beta-test-233515-3.html#post3134265

https://www.cruisersforum.com/forums/f134/opencpn-version-5-1-428-beta-test-233515-3.html#post3134490

Now I just saw this post from Dave
https://www.cruisersforum.com/forums/f134/opencpn-version-5-1-428-beta-test-233515-2.html#post3134581

Perhaps the problem with installation is not caused by mixing gtk3 with non, but more related to the fact that I have already installed opencpn and Ubuntu does not support parallel installations?

rgleason, 2020-05-08:
Is this needed in plugins cmake too?

bdbcat, 2020-05-09:
I have adopted Leamas' patch in OpenCPN github master.
Test builds on bionic and focal look OK on quick test.
@tai-fun ...  Please close after your test.

Thanks
Dave

rgleason, 2020-05-09:
I would be interested in any conclusions or summaries with regard to the use of 
Opencpn with gtk3/without  and plugins with gtk3/without.
Guidelines for install and use.

leamas, 2020-05-10:
From my perspective, this could now be closed

tai-fun, 2020-05-10:
I was waiting you. Closing now.

closed 2020-05-10 by tai-fun

leamas, 2020-05-10:
Right. Thanks for reporting!