OpenCPN/OpenCPN #1899: Weatherfax Install Path & User Data Directory need fix

rgleason, 2020-05-08:
Current repository and branch:  https://github.com/rgleason/weatherfax_pi/tree/frontend2

These files are now loaded into (for Plugin Manager)

    C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi\data\WeatherFaxInternetRetrieval.xml
    C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi\data\WeatherFaxSchedules\WeatherFaxInternetRetrieval.xml

They used to be loaded into (for Plugin Packaged installer, earlier Opencpn Program)

    C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins\weatherfax_pi\data\WeatherFaxInternetRetrieval.xml
    C:\Program Files (x86)\OpenCPN 5.0.0+9065270\plugins\weatherfax_pi\data\WeatherFaxSchedules.xml

The User Data Directory location for both conditions should be here

    C:\ProgramData\opencpn\plugins\weatherfax\UserCoordinatesSets.xml
    C:\ProgramData\opencpn\plugins\weatherfax\CoordinatesSets.xml
    C:\ProgramData\opencpn\plugins\weatherfax\   Weatherfax PNG,JPG, GIF files

**TEST:**  Load Beta  Weatherfax into Beta Opencp.  Pick from one of these tarball files.  Download and use Import Plugin. Install and Enable. Then weatherfax. At the first menu Pick "Retrieve" > "Internet"  and the path to weatherfaxInterneteRetrieval.xml is not found.

Currently the plugin is looking for the vital  weatherfaxInterneteRetrieval.xml  file in  
    C:\Program Files (x86)\OpenCPN 5.1.428+c80e933\plugins\weatherfax_pi\data\weatherfaxInterneteRetrieval.xml

Note that when I copy the weatherfax_pi\data\*.* file to this location, the plugin does load the weatherfaxinternetretrieval.xml file and I can select weatherfax files to download.
However when I select to download, the destination file is not found, and the message says there is no working connection  when it should be looking in this folder!

    C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi\data 
    or %localappdata%\opencpn\plugins\weatherfax_pi\data

and it should be saving the downloaded weatherfax files to the  User Data Directory that has read/write:

    C:\ProgramData\opencpn\plugins\weatherfax\

Additionally, it should copy two other files to this directory:
        CoordinateSets.xml
        UserCoordinateSets.xml
`---------------------------------------`
**WeatherFax.cpp**
Line 137

    #define FAIL(X) do { error = X; goto failed; } while(0)
    static void LoadCoordinatesFromXml(WeatherFaxImageCoordinateList &coords, wxString coordinatesets)
    {
        TiXmlDocument doc;

        wxString error;
        wxString coordinatesets_path = weatherfax_pi::StandardPath();
        wxString s = wxFileName::GetPathSeparator();
        wxString default_coordinatesets_path = *GetpSharedDataLocation() + _T("plugins")
            + s + _T("weatherfax_pi") + s + _T("data") + s;

Line 476

    #ifdef WIN32
            // attempt to convert using PVW32Con.exe
            wxString s = wxFileName::GetPathSeparator();
            wxString pvw32con = *GetpSharedDataLocation() + _T("plugins")
                + s + _T("weatherfax_pi") + s + _T("PVW32Con.exe");
            wxString cmd = pvw32con + _T(" \"") + filename + _T("\" -t --o \"" + filename + _T("\""));

            int ret = ::wxExecute(cmd, wxEXEC_SYNC);

            wxLogMessage(_("try to execute :") + cmd + _T("\nreturn code ") + wxString::Format(_T("%d"), ret));
            if(!wimg.LoadFile(filename))
    #endif

Line 791

    bool WeatherFax::DownloadFile( wxString filename )
    {
        const wxString url = m_weatherfax_pi.m_UpdateDataBaseUrl + filename;
        const wxString target_filename = m_weatherfax_pi.StandardPath() + filename;
        const wxString tmp_filename = wxFileName::CreateTempFileName( _T("weatherfax_") );
        _OCPN_DLStatus res = OCPN_downloadFile( url, tmp_filename, _("WeatherFax Data update"),
                                _("Reading Headers: ") + url, wxNullBitmap, this,

Line 830

    void WeatherFax::OnUpdateData( wxCommandEvent& event )
    {
        if( DownloadFile( _T("WeatherFaxInternetRetrieval.xml") ) &&
           DownloadFile( _T("CoordinateSets.xml") ) ) {
            m_InternetRetrievalDialog.Load(true);
        }
        if( DownloadFile( _T("WeatherFaxSchedules.xml") ) ) {
            m_SchedulesDialog.Load(true);
        }

`===========`

Example of what was done with weather_routing

weather-routing.cpp

https://github.com/rgleason/weather_routing_pi/commit/97dd0fb350e7a3b7504b5dbea9ff1fcf2fbebfb9#diff-dbe7dd150a3359638d0b109f149909a2L165

   /* if the boats or polars directories did not previously exist, populate them */
    if (forceCopyBoats)
        CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/boats"), boatsdir);
        CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("/data/boats"), boatsdir);
    if (forceCopyPolars)
        CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/polars"), polarsdir);
        CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("/data/polars"), polarsdir);

    int confVersion;
    pConf->Read ( _T ( "ConfigVersion" ), &confVersion, 0);
@@ -205,11 +205,11 @@ WeatherRouting::WeatherRouting(wxWindow *parent, weather_routing_pi &plugin)
                     result.GetCount();
                 i++) {
            if (result[i] == 0) {
                CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/boats"), boatsdir);
                CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/polars"), polarsdir);
                CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("/data/boats"), boatsdir);
                CopyDataFiles(GetPluginDataDir("weather_routing_pi") + _T("/data/polars"), polarsdir);
        }
        else if (result[i] == 1) {
            wxString cfg = GetPluginDataDir("weather_routing_pi") + _T("plugins/weather_routing_pi/data/") + _T("WeatherRoutingConfiguration.xml");
            wxString cfg = GetPluginDataDir("weather_routing_pi") + _T("/data/") + _T("WeatherRoutingConfiguration.xml");
            if (wxFileName::FileExists(cfg))
            wxCopyFile(cfg, m_default_configuration_path);
        }
@@ -1576,7 +1576,7 @@ void WeatherRouting::OnManual ( wxCommandEvent& event )
void WeatherRouting::OnInformation ( wxCommandEvent& event )
{
    wxString infolocation = GetPluginDataDir("weather_routing_pi")
        + _T("plugins/weather_routing_pi/data/") + _("WeatherRoutingInformation.html");
        + _T("/data/") + _("WeatherRoutingInformation.html");
    wxLaunchDefaultBrowser(_T("file://") + infolocation);

leamas, 2020-05-10:
Shouldn't this be filed against the Weatherfax plugin?

rgleason, 2020-05-10:
Please leave in place. This is a path problem caused by Plugin Manager changes.  Hopefully Jon will fix it.

jongough, 2020-05-12:
This is one of the joys of using the new Plugin Manager, it has invalidated some older API calls without telling anyone. I think Weatherfax is using GetpSharedDataLocation but you now have to use GetPluginDataDir or it all goes wrong. I did raise this as an issue, but.....

I will see if I can update weatherfax to use the new call.

leamas, 2020-05-12:
Just in case someone else is reading this: _invalidated some older API calls_ statement. 

The only call (function) change is indeed the  GetpSharedDataLocation() -> GetPluginDataDir() update. as described in https://github.com/leamas/opencpn/wiki/Plugin-adaptation. 

That said, there are also other changes to do in order to work with the plugin manager, described in link above.

jongough, 2020-05-12:
That is still invalidating an API call without telling anyone. The behaviour trend has been set. Why would anyone think to start to look elsewhere than OCPN? This should have been in the README for OCPN and broadcast to those building and trying to maintain plugins, as well as a helpful return message or build failure when trying to use the invalidated API call.

leamas, 2020-05-12:
For others to judge...

rgleason, 2020-05-12:
> I will see if I can update weatherfax to use the new call.

Thanks Jon, much appreciated.  I hope I can get the hang of it.

Perhaps this would be good  "a helpful return message"  with

"The call (function) GetpSharedDataLocation() must be changed to  GetPluginDataDir() update. as described in https://github.com/leamas/opencpn/wiki/Plugin-adaptation." 

to help get other plugin devs on the right track?

Thanks.

jongough, 2020-05-12:
The old API has to remain in place for those plugins that have not been upgraded and will be delivered not using the Plugin Manger. I think there may be a build time change needed that will 'invalidate' the use of the older API call such that if you try to use API 1.16 AND the old data location call you will get a build failure that is informative. I think this should be done by the proponents of the Plugin Manager to avoid issues around 'inclusion' of these changes.

bdbcat, 2020-05-13:
Jon/Rick...
There is nothing new here.
I think it is fair to hold plugin devs to a fairly high standard, such that they can follow the issues discussion, read available pubs, debug their own code, and deliver working plugins.  Without fuss, even in a fast-moving dynamic environment.

rgleason, 2020-05-13:
Jon has made the change and I think I should be able to do this in the future. 
It is new to me, even though I saw how you did it Dave.  So now closing.  

Maybe some Plugin Dev will find this.
https://github.com/jongough/weatherfax_pi/commit/8ce3e3cfdbb1c51e232c6b7282ced1f89673dbe6
Thank you Jon.

bdbcat, 2020-05-13:
Rick...
I am looking at the code, and Jon's fix will not work.

This:
```

OpenXML(GetPluginDataDir("weatherfax_pi") + _T("plugins")
            + s + _T("weatherfax_pi") + s + _T("data") + s
            + _T("WeatherFaxInternetRetrieval.xml"));

```Should probably be:

```
OpenXML(GetPluginDataDir("weatherfax_pi") + s + _T("data") + s
            + _T("WeatherFaxInternetRetrieval.xml"));

```
That is to say, the parameter to GetPluginDataDir() is already included in the return value.
That will need fixing in several locations, wherever GetPluginDataDir() is called.

jongough, 2020-05-13:
True, I forgot that the new call comes back with the full path rather than the shared path. Will fix it up.

rgleason, 2020-05-13:
Also, I think I should mention here for others, that the Jon needed to change the API to API 116 in two places, but I guess that should be understood since this is a new parameter in the API.

rgleason, 2020-05-13:
//	GetPluginDataDir("weatherfax_pi") + _T("plugins")
//        + s + _T("weatherfax_pi") + s + _T("data") + s;
   
Is this right? 
	GetPluginDataDir("weatherfax_pi") + s + _T("data") + s;

rgleason, 2020-05-13:
Testing the [tarballs  ](https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-alpha/packages/)in msvc Windows and Ubuntu Bionic  using internetFaxRetrieval, the files download and are overlayed correctly.  After Jon confirms the getplugindatadir  lines are right, I will upload to master plugins.  

Thank you for all your help (Jon and Dave!)  
Closing now.

closed 2020-05-13 by rgleason