OpenCPN/OpenCPN #1913: SENCThreadManager uses NULL pointer in corner case

transmitterdan, 2020-05-19:
I have found a potential for O to crash at shutdown due to use of NULL pointer in SENCThreadManager::FinishJob().  The call to MyFrame::GetPrimaryCanvas() can return NULL if O is shutting down.  When that happens then if ENC threads are still running, as they finish they try to use this null pointer and O will crash with a seg-fault.  I think all uses of GetPrimaryCanvas() outside the class itself should be guarded by NULL testing before use.

Hakansv, 2020-05-19:
I've surprisingly seen the Error message about O has crashed at last run rather often although I didn't noticed else than a normal shut down. This could be the possible cause?

bdbcat, 2021-05-03:
Closed by commit

closed 2021-05-03 by bdbcat