OpenCPN/OpenCPN #1914: ChartDownloader is corruping url

RooieDirk, 2020-05-19:
In the netherlands we have a site that is providing all kinds of info in the form of layers and or charts.
It used to work with the chartdownloader.
If you add a custom chart source  https://www.nautin.nl/opencpn/nautin.xml  , you can download extra usericons layers and some mbtiles charts. (All in the dutch language) 
But the download url is now somehowe corrupted.
[image]
[image]
[nautin.xml.txt](https://github.com/OpenCPN/OpenCPN/files/4652695/nautin.xml.txt)

rgleason, 2020-05-19:
Was this Klaus Spithost's website   https://openwaterkaart.nl?  I guess not.
He has just become an editor to improve the layers section of the manual but he has layers for Netherlands waterways which should be in Chartdownloader.

transmitterdan, 2020-05-20:
The XML file is mal-formed.  For example:

```
<zipfile_location><cr><lf>http://www.nautin.nl/opencpn/opencpn/nautin.usericons.zip<cr><lf></zipfile_location>
```

Within the zipfile_location there are embedded <cr><lf> ascii codes.  These are not allowed characters in a URL.

RooieDirk, 2020-05-20:
Ah, thanks. I'll forward it to the author of the script.

closed 2020-05-20 by RooieDirk