OpenCPN/OpenCPN #1915: AIS realtime prediction when target speed unavailable

davegardner, 2020-05-20:
AIS Class A targets with status **Unknown**, or where speed is unavailable are still 'predicted'. Is this right?

Hakansv, 2020-05-20:
@davegardner 
All A or B targets are included in the prediction. Status is not checked. So yes, that's correct.
A target without defined SOG should not be predicted and that would be corrected in Beta 5.1.518. From what O version is your shot?

davegardner, 2020-05-20:
@hakansv I am running 5.1.518 compiled locally from Master.

davegardner, 2020-05-20:
Accidentally closed it so just reopened.

Hakansv, 2020-05-20:
@davegardner 
Hmm..
I protected a undefined SOG as of below. A undefined SOG has a value of 555 so 103 would be enough. But now I see other has use a slightly decreased value for that. The reason may be a lost target is given a SOG at 103 Kn and if this is the case in your example, the target has been lost but received a new report before deletion, then my 103 limit is not enough but would read e.g. 102.2.
In case you still have the same sittuation around you could test to change "td->SOG < 103.0" to "td->SOG < 102.2" in below code snippet. 
I'll PR a change for this anyhow.
Thanks for the report.
Håkan

ais.cpp row ~1483
```
if (g_bDrawAISRealtime && 
            (td->Class == AIS_CLASS_A || td->Class == AIS_CLASS_B ) &&
             td->SOG > g_AIS_RealtPred_Kts &&
             td->SOG < 103.0) {
```

davegardner, 2020-05-20:
Hakan, That fixed it.
Wow! the power of magic numbers:-)
The situation with that ship is stable to I was able to test it.
I changed the SOG test to 102.2 and the target is now stationary, as it should be.
This is a very nice addition to AIS target tracking.
Thanks.

Hakansv, 2020-05-20:
Very good !
That change is now merged by Dave to master as well.
Time to close this?
Thanks
Håkan

davegardner, 2020-05-20:
Thanks again Hakan.

closed 2020-05-20 by davegardner