OpenCPN/OpenCPN #1917: Beta: Crash on delete of all tracks

andyrbarrow, 2020-05-20:
About 50% of the time, when I select "Delete All" for tracks, OpenCPN crashes.

My system is stationary right now at my home, and I leave track recording turned on. I usually get around a dozen track recordings in 24 hours (why isn't it a single track recording?). 

If I go to the track log and select Delete All, I usually see the list disappear, then OpenCPN crashes. Next start is a safe mode suggestion. Again, this happens about 50% of the time.

RPI4 4GB

transmitterdan, 2020-05-20:
Can you post the offending ~/.opencpn/navobj.xml file here?  I cannot reproduce this using my track list which has hundreds of tracks some covering hundreds of miles.  You obviously have to make a copy of the file before deleting the tracks.

rgleason, 2021-04-23:
I think this can be closed now. Can the original person close it? @andyrbarrow  or perhaps @transmitterdan

bdbcat, 2021-04-27:
Closing, inactive

closed 2021-04-27 by bdbcat