OpenCPN/OpenCPN #1918: Plugin Manager: Two Instances of weatherfax

rgleason, 2020-05-20:
Jon has fixed the builds and several path problems. Now I am trying to confirm that the plugin works properly in Windows.

After first cleaning out all instances of weatherfax in the 

- cache, 
- programdata/opencpn
- %localappdata%/opencpn/plugins
- program files (x86)\opencpn\plugiins

Then using  "Import Plugin "  to install the most current tarball [here](https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/?q=weatherfax*msvc)

I get two instances of weatherfax.  Why?

[image]
[image]
 
Also it is very difficult for me to be sure that I am testing the correct plugin because I cannot see the entire version number. I have asked for this ability a dozen times and it always gets cut off!!!

The version I want to test is weatherfax_pi-1.9.5.8_msvc-10.0.14393-MSVC.tar.gz
or  "1.9.5.8"    

I am not at all sure I have all previous versions cleaned out, because there are two instances in the plugin list, and furthermore, I am not sure that version 1.9.5 is actually 1.9.5.8!!!!   Must I resort to simply changing the version to 1.9.6!!!!

Isn't there some way that I can see the full version number?  IE 1.9.5.8?  That would make this much quicker and easier.

bdbcat, 2020-05-20:
Rick....
1. Semantic versioning for OCPN plugins has, at most, 3 digits.  That is the way it is. "Must I resort to simply changing the version to 1.9.6!!!!"   Well, yes.  OCPN has no way to determine that 1.9.5.8 is different from 1.9.5.7
2.  If the version shown has 2 digits, it means that this is most likely a legacy installation.  3 digits means the entry must have come from a catalog.
3.  If you enable a plugin, restart OCPN, and then inspect the logfile, you will see exactly where the plugin is coming from as it is loaded.

Dave

rgleason, 2020-05-20:
Thanks,
I hadn't thought of this

"If you enable a plugin, restart OCPN, and then inspect the logfile, you will see exactly where the plugin is coming from as it is loaded."
 
I am going to see if we can get the full version shown in the About tab or something as Leamas said I shouldn't change the 3rd digit, (that is reserved for the dev author).  <-- Really?

rgleason, 2020-05-20:
I find just one version listed in log   Weatherfax "1.9", this is not "1.9.5.8"  and it is using the older api 1.13,  so perhaps there is an older weatherfax version floating around somewhere that I have not removed?

```
17:46:08.811 MESSAGE pluginmanager.cpp:1134 Checking plugin candidate: C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi.dll
17:46:08.827 MESSAGE pluginmanager.cpp:2082 Plugin is compatible: true
17:46:08.827 MESSAGE pluginmanager.cpp:2145 PlugInManager: Loading PlugIn: C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi.dll
17:46:08.858    INFO pluginmanager.cpp:2316 PlugInManager:  C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi.dll
17:46:08.858    INFO pluginmanager.cpp:2317         Plugin common name: WeatherFax
17:46:08.858    INFO pluginmanager.cpp:2319         API Version detected: 113
17:46:08.858    INFO pluginmanager.cpp:2320         PlugIn Version detected: 1.9
```

I find weatherfax.dll 833 kb, dated 5/20/2020 in %localappdata%\opencpn\plugins
with a weatherfax_pi\data directory.  Opencpn has the  weatherfax 1.9 version enabled.

Disable weatherfax 1.9 , and I find
```
17:52:55.789 MESSAGE pluginmanager.cpp:554 PlugInManager: using data dir: C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi
17:52:55.789 MESSAGE WeatherFax.cpp:200 Weather Fax : Failed to load coordinate sets
```  

Then try to enable  Weatherfax 1.9.5 and find this in the log

```
18:19:18.762 WARNING pluginmanager.cpp:7953 Cannot load sound file: C:\Program Files (x86)\OpenCPN 5.1.428+c80e933\sounds/2bells.wav
18:19:40.573 MESSAGE update_mgr.cpp:196 Selected update: Weatherfax
18:19:40.589 MESSAGE pluginmanager.cpp:437 Installing Weatherfax
18:19:40.589 MESSAGE PluginHandler.cpp:965 Installing weatherfax_pi-1.9.5.8_msvc-10.0.14393-MSVC.tar.gz from local cache
18:19:40.673 MESSAGE PluginHandler.cpp:558 archive write install header error: Can't unlink already-existing object
18:19:40.673 MESSAGE PluginHandler.cpp:558 archive write install header error: Can't unlink already-existing object
18:19:44.206 MESSAGE pluginmanager.cpp:2082 Plugin is compatible: true

``

The plugin does not show the Weatherfax icon when enabled.

`

rgleason, 2020-05-20:
Version 1.9 is not unloading

```
18:34:39.496 MESSAGE update_mgr.cpp:196 Selected update: Weatherfax
18:34:39.496 MESSAGE pluginmanager.cpp:1623 PlugInManager: Deactivating PlugIn: 
18:34:39.496 MESSAGE pluginmanager.cpp:432 Uninstalling Weatherfax
18:34:39.496 MESSAGE pluginmanager.cpp:1623 PlugInManager: Deactivating PlugIn: 
18:34:39.512 WARNING PluginHandler.cpp:912 Cannot remove file C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi.dll: Unknown error
18:34:39.527 WARNING PluginHandler.cpp:912 Cannot remove file C:\Users\Rick\AppData\Local\opencpn\plugins\portaudio_x86.dll: Unknown error
18:34:41.444 MESSAGE pluginmanager.cpp:437 Installing Weatherfax
18:34:41.444 MESSAGE PluginHandler.cpp:965 Installing weatherfax_pi-1.9.5.8_msvc-10.0.14393-MSVC.tar.gz from local cache
18:34:41.474 MESSAGE PluginHandler.cpp:558 archive write install header error: Can't unlink already-existing object
18:34:41.474 MESSAGE PluginHandler.cpp:558 archive write install header error: Can't unlink already-existing object
18:34:56.477 MESSAGE pluginmanager.cpp:2082 Plugin is compatible: true
```

bdbcat, 2020-05-20:
Rick...
1." PlugInManager: Loading PlugIn: C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi.dll"
Delete this one manually.
2. "Installing weatherfax_pi-1.9.5.8_msvc-10.0.14393-MSVC.tar.gz from local cache"
Clear the cache.
3.  Start O, make sure that no WeatherFax from any location is loaded.  Look at the log.
4.  Then try your import.

Dave

leamas, 2020-05-21:
@bdbcat @rgleason 

> Semantic versioning for OCPN plugins has, at most, 3 digits. 

hm... The version we use actually supports four digits as documented in the [header file](https://github.com/OpenCPN/OpenCPN/blob/master/include/semantic_vers.h). 

The fourth digit corresponds to a "release" i. e., a change by for example Rick on a version supplied from a plugin project like weatherfax.When printed, it's only visible when defined.

rgleason, 2020-05-21:
Ok alec, good point. Thanks. Perhaps I have to change Jon's "version tweak" in CMakeLists.txt

set(VERSION_MAJOR "1")
set(VERSION_MINOR "9")
set(VERSION_PATCH "5")
set(VERSION_TWEAK "8")

to 
set(VERSION_MAJOR "1")
set(VERSION_MINOR "9")
set(VERSION_PATCH "5")
set(VERSION_REVISION "8")

>  Versions uses a modified semantic versioning scheme:
>  * major.minor.revision.post-tag+build.
>  * for example 1.2.6.1-alfa+deadbee. The values major, minor, revision
>  * and post should be integers. The tag is a pre-release marker, the build

Or is it not that simple?

bdbcat, 2020-05-21:
Rick...
What is the objective here, please?  Simply to get WeatherFax working?  Or something else?

rgleason, 2020-05-21:
Dave,  I followed your 
Rick...
1." PlugInManager: Loading PlugIn: C:\Users\Rick\AppData\Local\opencpn\plugins\weatherfax_pi.dll"
Delete this one manually.

```
2. "Installing weatherfax_pi-1.9.5.8_msvc-10.0.14393-MSVC.tar.gz from local cache"
Clear the cache.
3. Start O, make sure that no WeatherFax from any location is loaded. Look at the log.
4. Then try your import.
```
I ended up adding an important thing to the list. 

1. Reloading the Master Catalog to remove the weatherfax metadata in that catalog.
1. After clearing the cache of all weatherfax tarballs: C:\ProgramData\opencpn\plugins\cache\tarballs
1. Checking the cache metadata file:  ocpn-plugins.xml 
1. C:\Users\Rick\AppData\Local\opencpn\plugins  or %localappdata%\opencpn\plugins remove weeatherfax files
1. Check Standard internal plugins only: C:\Program Files (x86)\OpenCPN 5.1.518+1e2918d\plugins

I then checked the log file and ini file and found that the program was clear of weatherfax.
Then I loaded the program again, and Imported the most current tarball [here](https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-beta/packages/?q=weatherfax*msvc)

I found that there were two icons in the list again. One for Weatherfax 1.9 (not managed)
and another one for Weatherfax 1.9.5   When I enabled the 1.9.5 version no toolbar icon appeared and it simply did not work.  When I enabled the 1.9 (not managed) version, the toolbar icon shows and it works downloading faxes.

Therefore I am still very confused about what is happening here. It does not make sense to me. Perhaps I a being a bit thick, but why are there two plugins listed, when I only loaded one plugin?  Additionally the version numbers don't seem to be right.

Is an "imported plugin" treated as "not managed"?  But even then, why not show the full version number?  and why do I see two instances in the list and can select "enable" from either one?  It just does not make sense to me.

[image]

[image]

[image]

[image]

[image]

Also attached is the weatherfax entries from the log
[weatherfax-log-file.txt](https://github.com/OpenCPN/OpenCPN/files/4663134/weatherfax-log-file.txt)

bdbcat, 2020-05-21:
OK, I'll take a look.
Dave

rgleason, 2020-05-21:
Been working on this this morning.
The objectives are:
1. Understand the way Plugin Installer works with regard to installing tarballs, from catalogs and Import.
2. Get a handle on how to test plugins, particularly with regard to version number confusion, --Am I testing the exact right plugin?
3. Fix weatherfax if it needs fixing  ---I still don't know if it is fixed!
4. Answer the question why two lines are shown for one plugin, if indeed that is happening.  And if that is happening, see if it can be fixed!

rgleason, 2020-05-21:
Dave. Look for the "here" link 4 minutes ago

bdbcat, 2020-05-21:
OK, I'll look at (4).

bdbcat, 2020-05-21:
Rick...
The simple answer to (4).
There is a problem with CommonName.  One plugin is called "Weatherfax" and the other is called "WeatherFax".
Not sure which is "right", but OCPN, as always, considers them to be separate plugins.
I think that the root problem is that the imported Metadata file  calls it "Weatherfax", but the plugin reports CommonName as "WeatherFax".  or vice versa...

And so relates to (2):  You are effectively testing two different plugins, with different names.

Another related effect, and this is IMPORTANT.
If you import a plugin, and then later update the catalog by normal method, then the imported plugin becomes "unmanaged", since the metadata for that plugin is no longer available.   To say another way:  The metadata for an imported plugin only exists as long as the current catalog is not overwritten by a catalog download. But the plugin still is loaded, and it works as an unmanaged plugin, with 2 digit version reporting.

leamas, 2020-05-21:
> but the plugin still is loaded, and it works as an unmanaged plugin, with 2 digit version reporting.

Hm... IMHO here is (should be?) two cases: 
   - Plugins using API  <=116 will/must indeed report two digits version without metadata since no other info is available.
   - Plugins  using API >= 117 reports a full semantic version in the plugin interface. Fort these it should be possible to display complete version also without metadata.

rgleason, 2020-05-21:
Dave and Alec,  Thank you both for the information and help.  If you don't mind, I will leave this up until I get it sorted out, and will report, ...or if you'd like, I can just close it now and report anyway.

rgleason, 2020-05-21:
Dave

Weatherfax_pi.cpp 
Line 129
```
wxString weatherfax_pi::GetCommonName()
{
    return _("WeatherFax");
}
```

CMakeLists.txt 
Line 42
```
SET(VERBOSE_NAME weatherfax)
# Search "getcommonname"in weather_routing_pi.cpp
set(COMMON_NAME "Weatherfax")

```

So you are right, it is incorrect.

Weather_routing uses this in weather_routing_pi.cpp
Jon's idea I think.  Which I am going to start using in all Sean's plugins.
```
wxString weather_routing_pi::GetCommonName()
{
  //   return _T("WeatherRouting");
    return _T(PLUGIN_COMMON_NAME);
}

```

rgleason, 2020-05-22:
**Weatherfax_pi** 

Have changed the Common Name, now would like to be sure we
have the right API version and setup.
1. Is there a conflict between  weatherfax_pi.h   MY_API_VERSION_MAJOR etc and Cmakelists?
2. In weatherfax_pi.h  Line 56  "public opencpn_plugin_116"   what is this for?

CMakeLists.txt
----------------------
set(VERSION_MAJOR "1")
set(VERSION_MINOR "9")
set(VERSION_PATCH "5")
set(VERSION_TWEAK "8")
set(VERSION_DATE "05/19/2020")
set(OCPN_MIN_VERSION "ov50")
set(OCPN_API_VERSION_MAJOR "1")
set(OCPN_API_VERSION_MINOR "16")

weatherfax_pi.h
-----------------------
#define     MY_API_VERSION_MAJOR    1
#define     MY_API_VERSION_MINOR    13
line 56
class weatherfax_pi : public opencpn_plugin_116

bdbcat, 2020-05-22:
Rick...
1."In weatherfax_pi.h Line 56 "public opencpn_plugin_116" what is this for?"
The selected class name must match the declared API version.  Your entries look inconsistent.  You probably want API 116.  There is truly no reason to move to 117 if your needs are met with 116.
So, change:
#define MY_API_VERSION_MINOR 13
to
#define MY_API_VERSION_MINOR 16
Dave

rgleason, 2020-05-22:
Thanks Dave. Will make those changes.
I guess I will need to ask Jon what part  Ocpn_api_version_major  and minor plays.

I have tested the repaired weatherfax using the most recent  msvc tarball and it works as expected, without dual icons, and much simpler actions. Thank you.
https://cloudsmith.io/~rick-gleason/repos/opencpn-plugins-alpha/packages/

rgleason, 2021-06-06:
Fixed

closed 2021-06-06 by rgleason