OpenCPN/OpenCPN #1921: Plugin metadata needs $ARCH field

bdbcat, 2020-05-21:
Gentlemen...
In the process of testing the plugin manager and plugin CI workflow, especially of ARMXX systems, it becomes clear that we need to add the plugin build-time $ARCH parameter to the managed plugin metadata file.  Initial attempts to accommodate , e.g., oeSENC on generic Ubuntu bionic, for ARM64 systems, by trying to "fudge" the metadata "target" name, quickly leads to spaghetti.  We might make it work, but it would be fragile going forward.  Let us fix it now.
I propose to add a field:  TARGET_ARCH to the metadata, to be populated during the plugin CI build step.  This will be polled as necessary during OCPN runtime to determine basic plugin compatibility. 
I will modify what we call "Frontend 1" on oesenc_pi as an example.
Of course, this means that all current managed plugins will need to be modified, rebuilt, and PRs posted to the master plugin catalog branch.
Sorry for the last minute hitch.
Dave

leamas, 2020-05-22:
Perhaps this is the right solution, not sure, havn't looked into the actual arm cases. However:
  - The gtk2/gtk3 mess in bionic cannot be ba handled with an arch parameter either. Same for upcoming wxWidgets 3.0 -> 3.2 ABI break.
  - Bionic is sort of worst case: when moving forward we will drop gtk2, simplifying things again.

I  don't see the complete picture, but: In what way simplifies a TARGET_ARCH field compared to using for example _debian-armhf_ and _debian-aarch64_  as TARGET_OS? The omplexity is still the same, or?

That is, could we work around this by making the  TARGET_OS by definition defined as os-arch? Ubuntu -> ubunto-x86_64, debian -> debian-aarch64 etc. This means a smaller change, and adheres to  schemes used  by for example Debian and Fedora, so it  is sort of tested.

Still means changes in all plugins, but they should be smaller...

Just my 5 öre

bdbcat, 2020-05-22:
An example:
1.  Today, raspbian (armhf) reports (via /etc/os-release) that it is "like" debian.  So, we get ambiguity trouble between raspian version 10, and debian version 10.  Both are declared "compatible" with buster on rPI, but clearly not.  Indeed, today we are left assuming (guessing?) that a declared debian 10 plugin build is X86_64.  Adding an explicit $ARCH filter will fix this. 

So, if we have to rebuild all the plugins to expand the name field, might as well make it explicit. And its only one line of harmless code...

And, of course, the code simplifies immediately if the $ARCH does not match the host. No trouble there.  

Anyway, for insurance, now is the time to add this, before production users see it.

leamas, 2020-05-22:
I see the need to include the architecture in the matcing, no problem.

It's just that the this basically is about giving a unique name to each ABI. From this perspective, for example the difference between linking against gtk2 or gtk3 is roughly the same as the difference between for example aarch64 and armhf. It's all about having a unique name, and it could be done using a scheme for  TARGET_OS like described above.

This keeps the basic model simple: an plugin is compatible if it's using the same TARGET_OS

bdbcat, 2020-05-22:
The best use of the ARCH is "dis-matching", immediately eliminating candidates without name parsing.

So, we do both.  
This means that all the linux names need changing, AFAICT?

leamas, 2020-05-22:
> This means that all the linux names need changing, AFAICT?

Yes, unless we add some heuristics for missing -arch suffix.

bdbcat, 2020-05-22:
I'll adapt oesenc as a model, and share the results.  Prefer no heuristics.  That way lies spaghetti.

leamas, 2020-05-22:
>  Prefer no heuristics. That way lies spaghetti.

Agreed

leamas, 2020-05-22:
Saw your letter. Done is done.  That said:

  - Why  use the os-arch naming scheme only on Linux? Fltatpak definitiely also needs  an -arch suffix the same way. And since 64-bit builds are looming for both Windows and macOS it would make a lot of sense ot use it everywhere in  an uniform way.
  - If we use this naming scheme, what is the new  TARGET_ARCH  needed for?

leamas, 2020-05-22:
BTW: If you still go for using (and requiring) TARGET_ARCH¸ please update the xsd file in plugins project.

bdbcat, 2020-05-22:
TARGET_ARCH is a fallback, an insurance, a confirmation.  Not least, it eliminates incompatible candidate plugins that may cause unpredictable system response, long before that could happen.
And it is cheap.

bdbcat, 2020-05-22:
"please update the xsd file in plugins project."
Done.

rgleason, 2020-05-27:
Jon has done Testplugin, Ocpn_draw and I have moved this change to weatherfax.

rgleason, 2020-05-27:
Do we need an updated version of Opencpn to run these?

leamas, 2020-07-03:
Time to close?

leamas, 2020-07-24:
@bdbcat: ^

closed 2020-07-24 by bdbcat