OpenCPN/OpenCPN #1922: Hard to find Github project from website

maxekman, 2020-05-23:
Really cool project, but I had a hard time finding the source code here on Github. If you call a project open-something I would suggest that you provide a really easy way for users to get to the source, be it for reading, contribution or other involvement. A suggestion could be the “Fork me on Github” corner stripe or the Github star button.

Hakansv, 2020-05-23:
@maxekman 
There are plenty of ways for this.
The "code" page on this site has the fork link.
You could also look here: 
https://opencpn.org/OpenCPN/info/downloadopencpn.html
or: 
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_windows

maxekman, 2020-05-25:
Sure, I did find my way here. I just thought adding a Github corner on the front page would give it a bit more modern feel regarding the use of the open-term.

Here is a modern take on the classic Github corner ribbon. http://tholman.com/github-corners/

Feel free to close this if you feel that nothing can be improved in this area. These were my observations when seeing the project for the first time and visiting the webpage, and could be considered “valuable” first-use feedback.

rgleason, 2021-04-22:
Perhaps this issue can be closed now? We will have a link to Opencpn github in the newly revised Developer's Manual.
Will the original person please close this.  Thanks

maxekman, 2021-04-22:
Sure, I can close this.

closed 2021-04-22 by maxekman