OpenCPN/OpenCPN #1924: Debug build hangs in gtk3

RooieDirk, 2020-05-24:
A local build (debug) hangs on an assert in wxwidgets. Building a release works fine.
[image]
For the backtrace see file
[backtrace.txt](https://github.com/OpenCPN/OpenCPN/files/4674123/backtrace.txt)

bdbcat, 2020-05-25:
What distro and version, please?

RooieDirk, 2020-05-25:
Dave,
Kubuntu 2020.04

leamas, 2020-05-25:
Basically a duplicate of #1167. See also #1166

transmitterdan, 2020-05-26:
There are a lot of debug asserts in OpenCPN.  Some of them I think are actually bugs in non-OpenCPN code.  Removing all of them will be a pretty big task.

leamas, 2020-05-26:
@transmitterdan :
Indeed. In this case  http://trac.wxwidgets.org/ticket/17702, fixed in master -- the fix is supposed to be included in 3.2

leamas, 2020-05-26:
Not a single reference was ok. The duplicate is https://github.com/OpenCPN/OpenCPN/issues/1500, and the upstream bug https://trac.wxwidgets.org/ticket/18688.

Sorry for the mess...

RooieDirk, 2020-07-24:
Seems to be fixed in wxwidgets 3.2

closed 2020-07-24 by RooieDirk