OpenCPN/OpenCPN #1926: Plugin Manager: Import Plugin and uninstall and reinstall Plugin

rgleason, 2020-05-25:
If there are multiple tarball versions of a plugin in the cache when a plugin is Uninstalled using the PI Manager, and a new tarball is Imported, the version that is installed seems to be the older one, at least the data files being used are the older version. This has been real disconcerting, because I am trying to update weatherFaxInternetRetrieval.xml  and each time I have an update I get into this struggle to get the correct version loaded. 

Until Jon gave me full access to the version number I did not even know which version of plugin was being used. Now I am finding that there seems to be something wrong with the way multiple cached versions are selected.

rgleason, 2020-05-25:
PS: All the weatherfax files in the cache had the same version number but some were more recent builds.

leamas, 2020-05-25:
> All the weatherfax files in the cache had the same version number but some were more recent builds.

This is basically a broken behaviour. Two different tarballs with the same version indicates problems in the build chain. You could include the git commit,  the ci build number and/or a manually updated release number in the version to get around this.

Bottom line: there can't be two different tarballs with the same version in a sane echosystem.

bdbcat, 2020-05-25:
Rick...
As Alec says.
In development, as you are, if you do not (or cannot) bump the plugin version number for each build, then you will probably need to add a manual step to your workflow to clear the tarball cache before attempting to import a newly built plugin.  When ready to publish to end-users, bump the version number, and let it fly as a new release.
This scenario will not happen with end-users, because presumably the devs (you) will not publish multiple plugins with the same version number.
Development is never easy, and not without petty annoyances.  But we learn to live with them.
Dave

rgleason, 2020-05-26:
I guess things change, when I started plugins,  I was told not change the version, now I need to increment each time there is a change. Ok, if these are the rules, I will follow them.

I've also been noticing that JG increments each time. This is not too hard to do. Thanks.

I am going to assume if I increment and have unique version numbers I will not have this problem anymore.

Is there someplace where these rules are written down, because I am not writing anything.

closed 2020-05-26 by rgleason

bdbcat, 2020-05-26:
Rick...
If you are working for yourself, not sharing with anyone else, you may want to NOT change the version, but just remember to clear the cache between iterations.  Might be faster...

rgleason, 2020-05-26:
Thanks, will consider.

leamas, 2020-05-26:
> I guess things change, when I started plugins, I was told not change the version, now I need to increment each time there is a change. Ok, if these are the rules, I will follow them.

Remember; we added a fourth digiit to the version for exactly this purpose. The contract is that  the three first parts (major-minor-patch) belongs to the plugin developer, don't change those if your are not the primary developer.

But the fourth digit, the release, is yours to change. Just bump it for every build or so. It doesn't matter if it gets big.

rgleason, 2020-05-26:
Ok, thank you Alec

rgleason, 2020-05-26:
Do you know if cache gets confused if you have the same plugin version metadata in the master catalog and the beta catalog?

bdbcat, 2020-05-26:
Rick...
There is only one cache, used for all catalogs.

rgleason, 2020-05-26:
Ok, so the backup file for both alpha and master is stored in the cache and will overwrite one another, unless different versions have unique version numbers.

rgleason, 2020-05-26:
While we are discussing pi manager, do you think the changes in format have stopped so we mice can come out and play?

There is a lot of plugins still to fix.
Weather_routing, Watchdog, Vdr, to begin with.

Jon has been a huge help getting weathfax working and then working under this new regiem and that is ready for release hopefully tonight to master.