OpenCPN/OpenCPN #1928: Serial interface for NMEA0183 input -AIS data & navigation instruments-  switch without reason

SYLinamar, 2020-05-26:
>setting>connections>Data input

In our setup we get two data input feeds:

NMEA0183 navigation data via a serial interface (into USB 2.0) at BAUD rate of 4.800

AIS input (EASY TRX Nav transceiver) at a BAUD rate of 38400

Both inputs are in NMEA0183 format. Upon re-starting OpenCPN the two data connections change and either the Baud rates for USB0 and USB1 have to be chenged or the USB connectors have to be swapped.

transmitterdan, 2020-05-26:
What OS/version?  Sometimes the program itself has no say in the assignment of USB serial port identifiers.

SYLinamar, 2020-07-04:
The OS version "Raspian Buster" (OS: Raspian, Version: Buster). The OpenCPN version is the latest one that was available for Raspian Buster via the package manaager at the moment, we sent the first mail. 
Sorry, that it took some time to answer. I hope nevertheless that helps and that you can pick this up again.

RooieDirk, 2020-07-07:
This is an OS problem. your OS starts numbering with the first device coming up, and this is not always happen in the same sequence.
The good news is you can tell your OS to give an fixed name to a fixed device, and you can use this fixed name in OpenCPN.
see:
https://www.domoticz.com/wiki/Assign_fixed_device_name_to_USB_port
https://www.domoticz.com/wiki/PersistentUSBDevices

Hakansv, 2020-07-07:
Or use Openplotter and its "serial" for the same purpose :)

so6power, 2020-09-03:
> This is an OS problem. your OS starts numbering with the first device coming up, and this is not always happen in the same sequence.
The good news is you can tell your OS to give an fixed name to a fixed device, and you can use this fixed name in OpenCPN.

This can be a solution. However, it can be a problem if several serial-usb adapters are sharing the same IDs.
You can also opt for another solution, using network connections via tools like gpsd , kplex, muplex...

rgleason, 2021-04-27:
@so6power
RooieDirk and  HakanSV have given two solutions. This issue can now be closed.  Will the Origin Poster please close this?

closed 2021-04-28 by rgleason